bdring/FluidNC #570: Errors uploading configuration and changing configuration values

codingishard404, 2022-08-16:
### Controller Board

FluidNC Pen Laser Controller (SPI) with TMC2130 drivers, running FluidNC v3.4.4

### Help From Board Vendor

- [ ] Yes
- [ ] No
- [X] Not Applicable

### Machine Description

The machine is a corexy sand table (ZENXYv2).

### Configuration file

```yaml
**config_01** - this file works and loads without errors.

board: FluidNC Pen/Laser 2130
name: TMC2130 ZenXY
meta:
stepping:
  engine: RMT
  idle_ms: 255
  pulse_us: 2
  dir_delay_us: 1
  disable_delay_us: 0
  segments: 6

axes:
  shared_stepper_disable_pin: gpio.13
  shared_stepper_reset_pin: NO_PIN
  x:
    steps_per_mm: 80.000
    max_rate_mm_per_min: 1000.000
    acceleration_mm_per_sec2: 25.000
    max_travel_mm: 450.000
    soft_limits: false
    homing:
      cycle: 2
      allow_single_axis: true
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.39
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      tmc_2130:
        step_pin: gpio.14
        direction_pin: gpio.12
        disable_pin: NO_PIN
        r_sense_ohms: 0.110
        run_amps: 0.250
        hold_amps: 0.250
        microsteps: 32
        toff_disable: 0
        toff_stealthchop: 5
        use_enable: false
        cs_pin: gpio.16
        spi_index: -1
        run_mode: StealthChop
        homing_mode: StealthChop
        stallguard: 0
        stallguard_debug: false
        toff_coolstep: 3

    motor1:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      null_motor:

  y:
    steps_per_mm: 80.000
    max_rate_mm_per_min: 1000.000
    acceleration_mm_per_sec2: 25.000
    max_travel_mm: 885.000
    soft_limits: false
    homing:
      cycle: 1
      allow_single_axis: true
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.36
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      tmc_2130:
        step_pin: gpio.25
        direction_pin: gpio.26
        disable_pin: NO_PIN
        r_sense_ohms: 0.110
        run_amps: 0.250
        hold_amps: 0.250
        microsteps: 32
        toff_disable: 0
        toff_stealthchop: 5
        use_enable: false
        cs_pin: gpio.17
        spi_index: -1
        run_mode: StealthChop
        homing_mode: StealthChop
        stallguard: 0
        stallguard_debug: false
        toff_coolstep: 3

    motor1:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      null_motor:

  z:
    steps_per_mm: 80.000
    max_rate_mm_per_min: 1000.000
    acceleration_mm_per_sec2: 25.000
    max_travel_mm: 1000.000
    soft_limits: false
    homing:
      cycle: 0
      allow_single_axis: true
      positive_direction: true
      mpos_mm: 0.000
      feed_mm_per_min: 50.000
      seek_mm_per_min: 200.000
      settle_ms: 250
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      null_motor:

    motor1:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      null_motor:

kinematics:
  CoreXY:

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  cs_pin: gpio.5
  card_detect_pin: NO_PIN

control:
  safety_door_pin: NO_PIN
  reset_pin: NO_PIN
  feed_hold_pin: NO_PIN
  cycle_start_pin: NO_PIN
  macro0_pin: NO_PIN
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN

coolant:
  flood_pin: NO_PIN
  mist_pin: NO_PIN
  delay_ms: 0

probe:
  pin: NO_PIN
  check_mode_start: true

macros:
  startup_line0:
  startup_line1:
  macro0:
  macro1:
  macro2:
  macro3:

start:
  must_home: true
  deactivate_parking: false
  check_limits: false

user_outputs:
  analog0_pin: NO_PIN
  analog1_pin: NO_PIN
  analog2_pin: NO_PIN
  analog3_pin: NO_PIN
  analog0_hz: 5000
  analog1_hz: 5000
  analog2_hz: 5000
  analog3_hz: 5000
  digital0_pin: NO_PIN
  digital1_pin: NO_PIN
  digital2_pin: NO_PIN
  digital3_pin: NO_PIN

arc_tolerance_mm: 0.002
junction_deviation_mm: 0.010
verbose_errors: false
report_inches: false
enable_parking_override_control: false
use_line_numbers: false
planner_blocks: 16
NoSpindle:
```

### Startup Messages

```text
[MSG:INFO: FluidNC v3.4.4]
[MSG:INFO: Compiled with ESP32 SDK:v3.3.5-1-g85c43024c]
[MSG:INFO: Configuration file:config_01.yaml]
[MSG:INFO: Machine TMC2130 ZenXY]
[MSG:INFO: Board FluidNC Pen/Laser 2130]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN]
[MSG:INFO: Stepping:RMT Pulse:2us Dsbl Delay:0us Dir Delay:1us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Shared stepper disable gpio.13]
[MSG:INFO: Axis X (0.000,450.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_2130 Step:gpio.14 Dir:gpio.12 CS:gpio.16 Disable:NO_PIN Index:-1 R:0.110]
[MSG:INFO:     Neg Limit gpio.39]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Y (0.000,885.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_2130 Step:gpio.25 Dir:gpio.26 CS:gpio.17 Disable:NO_PIN Index:-1 R:0.110]
[MSG:INFO:     Neg Limit gpio.36]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Z (-1000.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:   Motor1]
[MSG:INFO: X Axis driver test passed]
[MSG:INFO: Y Axis driver test passed]
[MSG:INFO: Kinematic system: CoreXY]
[MSG:INFO: Using spindle NoSpindle]
[MSG:INFO: WiFi is disabled]
```

### User Interface Software

FluidTerm

### What happened?

I deleted all existing files except for config_01.  I set the configuration to config_01, above. I reloaded without errors. I ran $CD and copied the output into a new file, called config_02. I ran a text comparison to doublecheck that config_01 and config_02 are identical.

[config_02.txt](https://github.com/bdring/FluidNC/files/9354699/config_02.txt)

I set the configuration to config_02, hit ctrl+R, and got the following errors:

### Other Information

```
[MSG:INFO: FluidNC v3.4.4]
[MSG:INFO: Compiled with ESP32 SDK:v3.3.5-1-g85c43024c]
[MSG:INFO: Configuration file:config_02.yaml]
[MSG:WARN: Ignored key caler]
[MSG:ERR: Skipping key feed_scaler indent 6 thisIndent 0]
[MSG:ERR: Skipping key motor0 indent 4 thisIndent 0]
[MSG:ERR: Skipping key limit_neg_pin indent 6 thisIndent 0]
[MSG:ERR: Skipping key limit_pos_pin indent 6 thisIndent 0]
[MSG:ERR: Skipping key limit_all_pin indent 6 thisIndent 0]
[MSG:ERR: Skipping key hard_limits indent 6 thisIndent 0]
[MSG:ERR: Skipping key pulloff_mm indent 6 thisIndent 0]
[MSG:ERR: Skipping key null_motor indent 6 thisIndent 0]
[MSG:ERR: Skipping key motor1 indent 4 thisIndent 0]
[MSG:ERR: Skipping key limit_neg_pin indent 6 thisIndent 0]
[MSG:ERR: Skipping key limit_pos_pin indent 6 thisIndent 0]
[MSG:ERR: Skipping key limit_all_pin indent 6 thisIndent 0]
[MSG:ERR: Skipping key hard_limits indent 6 thisIndent 0]
[MSG:ERR: Skipping key pulloff_mm indent 6 thisIndent 0]
[MSG:ERR: Skipping key null_motor indent 6 thisIndent 0]
[MSG:INFO: Axes: using defaults]
[MSG:INFO: Machine None]
[MSG:INFO: Board None]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN]
[MSG:INFO: Stepping:RMT Pulse:4us Dsbl Delay:0us Dir Delay:0us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (-1000.000,0.000)]
[MSG:INFO: Axis Y (-1000.000,0.000)]
[MSG:INFO: Axis Z (-1000.000,0.000)]
[MSG:INFO: Kinematic system: CoreXY]
[MSG:INFO: Using spindle NoSpindle]
[MSG:INFO: WiFi is disabled]
```

MitchBradley, 2022-08-16:
What method did you you use to upload config_02 ?

What is the result of $localfs/show=config_02 ?  Check to see if it is garbled compared to the copy of the file on your host computer.

bdring, 2022-08-16:
Yes, please send `$localfs/show=config_02.yaml` to show the exact contents of the file.

What text editor are you using. It is best to use something that can validate and colorize yaml files. I use Notepadd++ and VSCode for yaml editing.

codingishard404, 2022-08-17:
Well this explains why it's erroring, half the file is gone. I've been uploading via Ctrl+U on fluidterm.

I made config_01 in notepad, when I got the first errors I switched to notepad++ to more easily find the line numbers. I've uploaded (or at least tried to) files from each.

```ok
$localfs/show=config_02.yaml
caler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      null_motor:

    motor1:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      null_motor:

kinematics:
  CoreXY:

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  cs_pin: gpio.5
  card_detect_pin: NO_PIN

control:
  safety_door_pin: NO_PIN
  reset_pin: NO_PIN
  feed_hold_pin: NO_PIN
  cycle_start_pin: NO_PIN
  macro0_pin: NO_PIN
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN

coolant:
  flood_pin: NO_PIN
  mist_pin: NO_PIN
  delay_ms: 0

probe:
  pin: NO_PIN
  check_mode_start: true

macros:
  startup_line0:
  startup_line1:
  macro0:
  macro1:
  macro2:
  macro3:

start:
  must_home: true
  deactivate_parking: false
  check_limits: fals
ok
```

bdring, 2022-08-17:
Try uploading again and immediately checking it.

MitchBradley, 2022-08-17:
I have seen problems with SPIFFS losing data, especially when it is nearly full. I want to switch to LittleFS which is reputed to be much better than SPIFFS - but we are overwhelmed with other stuff including every new user begging for some strange feature.  One of the primary motivations for the switch to the new framework is because it includes LittleFS support.  We are still reeling from that switchover.  Once we do adopt LittleFS, that will come with its own set of migration issues.

bdring, 2022-08-17:
Agreed about the nearly full thing. Make sure there is more than enough room.

codingishard404, 2022-08-17:
```$Xmodem/Receive=config_02.yaml
[MSG:INFO: Received 4581 bytes to file /localfs/config_02.yaml]
ok
$localfs/show=config_02.yaml
ok
$localfs/show=config_02.yaml
ok
$localfs/show=config_01.yaml
board: FluidNC Pen/Laser 2130
name: TMC2130 ZenXY
meta:
stepping:
```
cut the rest of config_01 out, just tried it to make sure it was working.

```
$localfs/list

[FILE:/favicon.ico|SIZE:1150]
[FILE:/index.html.gz|SIZE:126651]
[FILE:/preferences.json|SIZE:1156]
[FILE:/config_01.yaml|SIZE:4583]
[FILE:/config_02.yaml|SIZE:0]
[LocalFS Free:36.28 KB Used:133.10 KB Total:169.38 KB]
ok
```
Should I try reformatting? I'm wondering if it hasn't been emptying the memory  when I delete files or something.

bdring, 2022-08-17:
Rerun, install-fs.bat.

I once got a "stuck" corrupted file that would not behave. Install-fs fixed it. Cleans out that partition.

codingishard404, 2022-08-17:
That did the trick! Thank you both for your time.

closed 2022-08-17 by codingishard404