bdring/FluidNC #571: WiFi will not connect to my router

maxcel2, 2022-08-17:
### Controller Board

Not relevant

### Help From Board Vendor

- [ ] Yes
- [ ] No
- [X] Not Applicable

### Machine Description

Not relevant

### Configuration file

```yaml
Not relevant
```

### Startup Messages

```text
Resetting MCU
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:12812
load:0x40080400,len:3032
entry 0x400805e4

[MSG:INFO: FluidNC v3.5.1]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
[MSG:INFO: Local filesystem type is SPIFFS]
[MSG:INFO: Configuration file:config.yaml]
[MSG:INFO: Machine ESP32-WROOM-32]
[MSG:INFO: Board WARCO WM16 MILL]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.21 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN]
[MSG:INFO: Stepping:RMT Pulse:3us Dsbl Delay:5us Dir Delay:5us Idle Delay:250ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (0.000,480.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:gpio.13:low Dir:gpio.12 Disable:gpio.15:low]
[MSG:INFO:     Neg Limit gpio.33:pu]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Y (0.000,160.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:gpio.14:low Dir:gpio.27 Disable:gpio.2:low]
[MSG:INFO:     Neg Limit gpio.32:pu]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Z (0.000,270.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:gpio.26:low Dir:gpio.25 Disable:gpio.4:low]
[MSG:INFO:     Neg Limit gpio.35]
[MSG:INFO:   Motor1]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: PWM Spindle Ena:gpio.23 Out:gpio.22:low Dir:NO_PIN Freq:2500Hz Period:16383]
[MSG:INFO: Using spindle PWM]
[MSG:INFO: Probe Pin: gpio.34:low]
[MSG:INFO: Connecting to STA SSID:HUAWEI-B535-49C4]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connecting...]
[MSG:INFO: Connecting....]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connecting...]
[MSG:INFO: Connecting....]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: AP SSID FluidNC IP 192.168.0.1 mask 255.255.255.0 channel 1]
[MSG:INFO: AP started]
[MSG:INFO: WiFi on]
[MSG:INFO: Captive Portal Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]

Grbl 3.5 [FluidNC v3.5.1 (wifi) '$' for help]
[MSG:INFO: '$H'|'$X' to unlock]
```

### User Interface Software

Fluidterm

### What happened?

I set up the WiFi connection to my router as follows:
$Sta/SSID=HUAWEI-B535-49C4
$Sta/Password=******
$Sta/IPMode=DHCP
$Sta/IP=0.0.0.0
$Sta/Gateway=192.168.8.1
$Sta/Netmask=255.255.255.0

The startup messages show that it does not connect, but falls back to AP mode.

### Other Information

Erased the ESP32 using erase.sh and loaded a sample wifi test program through Arduino IDE with the following function: (Courtesy of Random Nerd Tutorials)

`void initWiFi() {
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  Serial.print("Connecting to WiFi ..");
  while (WiFi.status() != WL_CONNECTED) {
    Serial.print('.');
    delay(1000);
  }
  Serial.println(Wi-Fi.localIP());
}`

That also failed to connect until the delay was changed to 5000, when it connected every time.

The conclusion is that some routers require more time to connect than is being allowed by the delay interval between retries.
Is there a way to increase the delay between retries to connect?

bdring, 2022-08-17:
There is no way to change the wait delay without changing the firmware. It is probably not something we would want to do for everyone as some people would rather have it change to AP mode sooner. Adding a setting is a bit of work we would not see as a high priority.

I would also be concerned that something else is wrong and you have a poor connection that might fail. You are the first to raise this issue and the example you found also has a shorter delay.

maxcel2, 2022-08-17:
Thank you for your response. 
1. Regarding poor connection, it is working in the same room as the router and recording better than 60% signal. 
2. Tried a delay of 2 secs, that would sometimes connect, but not always.
3. Presumably, changing firmware means changing a value in the source code. I need to be able to connect in station mode, so perhaps you could point me in the right direction, so that I can make the change and recompile it.
4. Cannot see what difference a few extra seconds to connect would make, but perhaps you might consider increasing the retry connect delay and reducing the number of retries, which would keep the time to fall back to AP the same.

bdring, 2022-08-17:
3. I would just search for `"Connecting`

MitchBradley, 2022-08-17:
FluidNC does not "retry".  The intermediate "Connecting..." messages do not change the outstanding connection attempt.  They simply represent the result of polling the status to see if connection has succeeded.

MitchBradley, 2022-10-03:
This is stale and we don't have an obvious way to change it.

closed 2022-10-03 by MitchBradley