bdring/FluidNC #572: Feature:

ashokavarthanan, 2022-08-17:
### Machine Context

My CNC machine. ESP32-GRBL
x100mm-y200mm-z75mm
1.5kw Spindle.

### Feature Description

Hi, 
I am using  FluidNC from last month, its nice.
I want to save "Machine position (x y z) " for last 10 history 
when "SET COORDINATE ZERO" into ESP32 memory.
(with time if possible)

Thanks.

### Other Approaches

nothing.

### How I Can Help

nothing.

bdring, 2022-08-17:
That is not supported by standard gcode.

In my opinion a PC based gcode sender would be a better place. It would have access to time and more storage.

ashokavarthanan, 2022-08-17:
10 lines of CODE takes 525-bytes (around 45 kilobytes free in ESP32 ).
I think its enough.
and 
when SET COORDINATE G-code execute  "G10 L20 P1"
if g-code "G10 L20 P1" execute  can save line? (Current Machine position)
Thanks.

bdring, 2022-08-17:
Then what?

How is that value returned to the user? 
What does the user do with it? 
How is it applied to the machine?
Where does time come from?
How do existing senders deal with it?

A PC sender would do a better job of this.

ashokavarthanan, 2022-08-17:
its More helpful for re-position when power cuts.

bdring, 2022-08-17:
The value is already stored in G54-G59

ashokavarthanan, 2022-08-17:
I understand...  The value is stored.
But when is test Manual Power cut and re-connect ESP 
the Machine value all 0
???

bdring, 2022-08-17:
Regardless of what we do, you must rehome, then the G54 offset is correct.

ashokavarthanan, 2022-08-17:
I am not familiar with G54 G-code.
I start machine every time with rehome.
But when power failure( or Hang or any Error and issue ) more time I loss  Machine position.
This way more Helpful.
Anyway Thanks for Answer.

MitchBradley, 2022-08-17:
Loss of machine position on power fail is unavoidable since the motors lose their lock and and mechanical stresses that have accumulated are released.

closed 2022-08-17 by MitchBradley

bdring, 2022-08-17:
You have a lot to learn about machine positions. Watch this video. It will likely solve your problem.

https://www.youtube.com/watch?v=fGtbkVJBXyE&