bdring/FluidNC #573: TMC2209: Microsteps config not working

Michaelcombs, 2022-08-17:
### Controller Board

ESP-32S DEVKIT DOIT

BigTreeTech TMC2209 motor drivers 
[docs](https://github.com/bigtreetech/BIGTREETECH-TMC2209-V1.2)

Custom made control board
[image]

### Help From Board Vendor

- [ ] Yes
- [ ] No
- [X] Not Applicable

### Machine Description

Coil winder with 1 axis for bobbin rotation and another axis for wire guide which drives T8 screw. I've isolated the problem to a single axis in the config below.

### Configuration file

```yaml
name: "ESP32 Dev Controller V4"
board: "ESP32 Dev Controller V4"

stepping:
  engine: RMT
  idle_ms: 250
  dir_delay_us: 1
  pulse_us: 2
  disable_delay_us: 0

axes:
  
  x:
    steps_per_mm: 200
    max_rate_mm_per_min: 500
    acceleration_mm_per_sec2: 25
    max_travel_mm: 200
    homing:
      cycle: 2
      mpos_mm: 10
      positive_direction: true
    motor0:
       tmc_2209:
        uart:
          txd_pin: gpio.10
          rxd_pin: gpio.9
          rts_pin: NO_PIN
          cts_pin: NO_PIN
          baud: 115200
          mode: 8N1
        addr: 3
        r_sense_ohms: 0.110
        run_amps: 0.700
        hold_amps: 0.700
        microsteps: 2
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: StealthChop
        homing_mode: StealthChop
        use_enable: false
        step_pin: gpio.22
        direction_pin: gpio.23
        disable_pin: gpio.21
    motor1:
      null_motor:
```

### Startup Messages

```text
[MSG:INFO: FluidNC v3.5.1]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
[MSG:INFO: Local filesystem type is SPIFFS]
[MSG:INFO: Configuration file:config2.yaml]
[MSG:WARN: microsteps value 0 constrained to range (1,256)]
[MSG:INFO: Machine ESP32 Dev Controller V4]
[MSG:INFO: Board ESP32 Dev Controller V4]
[MSG:INFO: SPI not defined]
[MSG:INFO: Stepping:RMT Pulse:2us Dsbl Delay:0us Dir Delay:1us Idle Delay:250ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (-190.000,10.000)]
[MSG:INFO:   Motor0]
[MSG:INFO: Trinamic Stepper Uart Tx:gpio.10 Rx:gpio.9 RTS:NO_PIN Baud:115200]
[MSG:INFO:     tmc_2209 Step:gpio.22 Dir:gpio.23 Disable:gpio.21 Addr:3 R:0.110]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Y (-1000.000,0.000)]
[MSG:INFO: Axis Z (-1000.000,0.000)]
[MSG:ERR: X Axis driver test failed. Check motor power]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Using spindle NoSpindle]
[MSG:INFO: Connecting to STA SSID:XXXXXXX]
[MSG:INFO: Connecting.]
[MSG:INFO: Connected - IP is XXX.XXX.X.XX]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://Grbl2.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]

Grbl 3.5 [FluidNC v3.5.1 (wifi) '$' for help]
[MSG:INFO: '$H'|'$X' to unlock]
[MSG:INFO: Caution: Unlocked]
```

### User Interface Software

ESP3D

### What happened?

The microstep setting is either not sent to the TMC2209 correctly OR the FluidNC step or microstep calculations or pulses are not correct.  I am using a 200 steps per revolution motor and counting bobbin rotations. 

Upload config and reboot through interface:
steps_per_mm: 200
microsteps: 0
Move X access 1mm with ESP3d interface => 1 rotation ( .13A idle current) This is what I expected.

Upload changed config and reboot through interface:
steps_per_mm: 200
microsteps: 2
Move X access 1mm with ESP3d interface => 1 rotation ( .13A idle current). I expected 1/2 rotation. 
2 microsteps/Step * 200 Steps/rotation * = 400 steps/rotation.  
1mm * 200 steps/mm = 200 steps. 200 steps/400 steps/rotation = 1/2 rotation.

Cycle controller power AND 12V motor supply voltage
steps_per_mm: 200
microsteps: 2
Move X access 1mm with ESP3d interface => 1/8 rotation ( .75A idle current). I expected 1/2 rotation.

I can replicate this problem with various microstep settings. 

I can then take the same controller, upload a small program using an arduino framework and the TMC2209Stepper library and set the microsteps reliably. I can then reload FluidNC onto the controller and the config microsteps has no impact until I power cycle the driver.

MitchBradley, 2022-08-17:
When you see a line like this, `[MSG:ERR: X Axis driver test failed. Check motor power]` nothing else about the communication with the TMC device will work.

bdring, 2022-08-17:
Can you attach the schematic and the arduino test code you used?

Michaelcombs, 2022-08-18:
Here is the schematic. IC2 is the driver in question.
[coil_winder.pdf](https://github.com/bdring/FluidNC/files/9366990/coil_winder.pdf)

Here is the code. It repeatedly moves the motor 200 steps with 1000ms delay between. _driver1.microsteps(0);   moves 1 revolution, _driver1.microsteps(2);  moves half revolution, etc....

`#include <ESP_FlexyStepper.h>
#include "TMCStepper.h"

//TMC2209 
#define DIR1_PIN          23 // Direction
#define STEP1_PIN         22 // Step
#define SERIAL1_PORT      Serial2 // UART
#define ENABLE1_PIN       18 // Enable pin
#define DRIVER_ADDRESS   0b00 // address
#define R_SENSE 0.11f 

TMC2209Stepper _driver1(&SERIAL1_PORT, R_SENSE, DRIVER_ADDRESS);
ESP_FlexyStepper _stepper1;

void setup() {
  
   
    pinMode(ENABLE1_PIN, OUTPUT);
    digitalWrite(ENABLE1_PIN, LOW); //enable the motor
    SERIAL1_PORT.begin(115200);

    _stepper1.connectToPins(STEP1_PIN, DIR1_PIN);
    
    _driver1.begin();               
    delay(50);
    _driver1.toff(5);               
    _driver1.rms_current(1000);      
    _driver1.microsteps(0);           // Change microsteps here
    _driver1.pwm_autoscale(true); 
    _driver1.en_spreadCycle(true);   
    _driver1.internal_Rsense(false);
    
}

void loop() {

  _stepper1.setSpeedInRevolutionsPerSecond(1); 
  _stepper1.setAccelerationInRevolutionsPerSecondPerSecond(1);
  _stepper1.setTargetPositionRelativeInSteps(200);
  
  while (!_stepper1.motionComplete())
  {
      _stepper1.processMovement();
  }

  vTaskDelay(1000);
}`

MitchBradley, 2022-08-18:
addr: 3
vs.
#define DRIVER_ADDRESS   0b00 // address

Hmmmm.

Michaelcombs, 2022-08-18:
Sorry, the addr: 3 was iteration #57 of "let's try anything". I was definitely using addr: 0 when I had the problem. I'll update the config above to eliminate confusion.

MitchBradley, 2022-08-18:
Since the two TMC2209s are strapped to the same address with MS0 MS1, they will collide on reads.

Michaelcombs, 2022-08-18:
The second TMC2209 was removed for the test.

Michaelcombs, 2022-08-18:
Also, notice in the schematic that the second TMC2209 is on a separate serial port.

Michaelcombs, 2022-08-18:
The config limits microsteps to a range of 1 to 256. The full step setting (i.e. no microsteps) is 0.

bdring, 2022-08-18:
1. In the config...txd_pin: gpio.10 and rxd_pin: gpio.9 are on the [do not use list](http://wiki.fluidnc.com/en/hardware/esp32_pin_reference#dont-use-or-not-recommended).
2. Why do you have separate UARTs for each chip?
3. RX2, TX2, etc are poor choices for pin names as ESP32 does not care. Use the GPIO numbers.
4. What brand and rev are the drivers? do you have a schematic?
5. What are the actual voltages of MS1 and MS2. It appears they are floating.

ithinkido, 2022-08-18:
That broad does not even have GPIO  9 and 10 broken out.
[image]

bdring, 2022-08-18:
[image]

Michaelcombs, 2022-08-18:
MS1 and MS2 are floating. Correcting this eliminated the 'X Axis driver test failed.' msg.  I suppose this may have prevented FluidNC from attempting to transmit settings. I was transmitting settings (in custom code) always which apparently worked even with MS1 and MS2 floating.

1. I apologize. I am using pin 17 and 18 (UART 2). I was changing hardware, configs, etc while attempting a solution. 9 and 10 are UART 1 which I had reassigned from flash and using to testing on additional hardware.
2. The board was designed for custom code. I'm making a change to FluidNC.
3. I was using the PCB labels on the ESP32 module.
4. BigTreeTech v1.2
5. MS0 and MS1 are floating. 

Thank you and I apologize for posting a PCB design issue.

closed 2022-08-18 by Michaelcombs

bdring, 2022-08-18:
Thanks in advance for your generous donation to the project.

MitchBradley, 2022-08-18:
The driver test failed message results from a preliminary test to see if it is possible to read back the DRV_STATUS register.  If that fails, FluidNC will not even try to transmit the settings.  Your test code does not do any reads, but simply transmits without any confirmation that the connection is intact.  In some cases, TMC2209s will work in transmit-only mode.  FluidNC requires a bidirectional communication path where reading is possible.