bdring/FluidNC #578: Test-Pattern for testing Hardware step by step

ClausElectronics, 2022-08-20:
### Controller Board

JackLab

### Help From Board Vendor

- [ ] Yes
- [X] No
- [ ] Not Applicable

### Machine Description

The machine will run, hopefully. It is not finishes - milling machine - 80 x 120 x 25 cm

### Configuration file

```yaml
not available
```

### Startup Messages

```text
Hello, I'm a new member here. So it can happen that I m wrong here.

I want to control my CNC with ESP32 and WEBUI.
So my short questions are:

On witch voltage levels (0V, 3.3V) the inputs pins of the ESP32 must be, that I can move for example the x-axis motor.

Because  there are inputs like x-Limits,y, z,...Door ; probe, Hold resume...

Thank you very much for your help!
```

### User Interface Software

WebUI

### What happened?

no movement of the motor

MitchBradley, 2022-08-20:
You must have a configuration file to enable the features that you want.  The configuration file tells which pins are used for which functions, the active level of those pins, and many other things.  See http://wiki.fluidnc.com/en/config/overview for complete information.

ClausElectronics, 2022-08-22:
Thank you for the link. In the meantime I read  and watched a lot of articles and Videos... Now I have this result:...with the config-File.

It is hard to get so much information at once... ;)

**Output:**
No SD Card
Commands
$/axes/x/max_travel_mm
$/axes/x/homing/mpos_mm
$/axes/x/homing/positive_direction
$/axes/y/max_travel_mm
$/axes/y/homing/mpos_mm
$/axes/y/homing/positive_direction
$G
$/axes/x/max_travel_mm=1000.000
ok
[MSG:DBG: Cannot open file /spiffs/favicon.ico]
[MSG:DBG: Cannot open file /spiffs/favicon.ico.gz]
$/axes/x/homing/mpos_mm=0.000
ok
$/axes/x/homing/positive_direction=true
ok
$/axes/y/max_travel_mm=1000.000
ok
$/axes/y/homing/mpos_mm=0.000
ok
$/axes/y/homing/positive_direction=true
ok
[GC:G0 G54 G17 G21 G90 G94 M5 M9 T0 F0 S0]
ok
[MSG:DBG: Cannot open file /spiffs/macrocfg.json]
[MSG:DBG: Cannot open file /spiffs/macrocfg.json.gz]
[MSG:DBG: Cannot open file /spiffs/preferences.json]
[MSG:DBG: Cannot open file /spiffs/preferences.json.gz]
<Alarm|MPos:0.000,0.000,0.000|FS:0,0|WCO:0.000,0.000,0.000>
<Alarm|MPos:0.000,0.000,0.000|FS:0,0|Ov:100,100,100|A:>
<Alarm|MPos:0.000,0.000,0.000|FS:0,0>
$X
ERROR:152
error:152

<Alarm|MPos:0.000,0.000,0.000|FS:0,0>
<Alarm|MPos:0.000,0.000,0.000|FS:0,0|WCO:0.000,0.000,0.000>
<Alarm|MPos:0.000,0.000,0.000|FS:0,0|Ov:100,100,100|A:>
<Alarm|MPos:0.000,0.000,0.000|FS:0,0>
<Alarm|MPos:0.000,0.000,0.000|FS:0,0>

**config-file:**

name: "ESP32 Dev Controller V4"
board: "JackLab_GRLB_ESP32_CNC_Controller_V10"

start:
  must_home: true
  deactivate_parking: false
  check_limits: false
  
stepping:
  engine: RMT
  idle_ms: 250
  dir_delay_us: 1
  pulse_us: 5
  disable_delay_us: 1

axes:
  shared_stepper_disable_pin: gpio.13:high
  x:
    homing:
      cycle: 1
      mpos_mm: 100
      positive_direction: false
    steps_per_mm: 800
    max_rate_mm_per_min: 2000
    acceleration_mm_per_sec2: 25
    max_travel_mm: 1000
     
    motor0:
      limit_all_pin: NO_PIN
      stepstick:
        direction_pin: gpio.26
        step_pin: gpio.12
    motor1:
      null_motor:

  y:
    steps_per_mm: 800
    max_rate_mm_per_min: 2000
    acceleration_mm_per_sec2: 25
    max_travel_mm: 1000
    homing:
      cycle: 2
      mpos_mm: 10
      positive_direction: false

    motor0:
      limit_all_pin: gpio.4:low:pu
      stepstick:
        direction_pin: gpio.25
        step_pin: gpio.14
    motor1:
      null_motor:

  z:
    steps_per_mm: 800
    max_rate_mm_per_min: 2000
    acceleration_mm_per_sec2: 25
    max_travel_mm: 1000
    homing:
      cycle: 1
      mpos_mm: 10
      positive_direction: true

    motor0:
      limit_all_pin: gpio.15:low:pu
      stepstick:
        direction_pin: gpio.33
        step_pin: gpio.27
    motor1:
      null_motor:

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  cs_pin: gpio.5
  card_detect_pin: NO_PIN

coolant:
  flood_pin: gpio.31
  mist_pin:  gpio.21

        
probe:
  pin: gpio.32:low:pu

PWM:
  pwm_hz: 5000
  output_pin: gpio.17
  enable_pin: gpio.22
  direction_pin: NO_PIN
  disable_with_s0: false
  s0_with_disable: true
  spinup_ms: 0
  spindown_ms: 0
  tool_num: 0
  speed_map: 0=0% 10000=100%
Footer

ipsod, 2022-08-22:
Hi.  Can you provide some more details so we can understand what you're trying to do and where it's going wrong?  Photos of the hardware you're using, and how it's wired, for example.

ipsod, 2022-08-22:
Also, have you looked at the documentation?  It's up-to-date, and I find it much more useful than videos.

You can find it here: http://wiki.fluidnc.com/en/installation

ClausElectronics, 2022-08-23:
Thank you for your answer...

I wanted to open fluidterm2...I got this massage:..so I can't see the error massages of WEBIO

[image]

I use the board of JackLAB. I found it here in Github - and I tested it by programming the outputs and reading the inputs - and it works..
[Grbl_ESP32.pdf](https://github.com/bdring/FluidNC/files/9400612/Grbl_ESP32.pdf)

I only changed on the board the input resistors R1..R8 (330 Ohm) to use is for 5V-Inputs and not for 24 V. I calculated it...
With my easy Arduino-program I was able to control my stepper motors x,y,z and also the input-signals:

//Test IO Pins ESP32 (38 Pins, PZ Delivery, ES32_WROOM-32) Board for CNC JachLab 

 //Outputs
#define X_STEP_OUT 12
#define X_DIR_OUT 26

#define Y_STEP_OUT 14
#define Y_DIR_OUT 25

#define Z_STEP_OUT 27
#define Z_DIR_OUT 33

#define Step_Enable 13

#define Flood 16
#define Mist 21

#define Spindle_Enable 22
#define Spindle_PWM 17
//Inputs
#define Hold 36 
#define Resume 39
#define Abort 34
#define Door 35
#define Z_Probe 32
#define Z_Limit 15
#define Y_Limit 4
#define X_Limit 2

void setup() {
  pinMode(X_STEP_OUT, OUTPUT);
  pinMode(X_DIR_OUT, OUTPUT);
  pinMode(Y_STEP_OUT, OUTPUT);
  pinMode(Y_DIR_OUT, OUTPUT);
  pinMode(Z_STEP_OUT, OUTPUT);
  pinMode(Z_DIR_OUT, OUTPUT);
  pinMode(Flood, OUTPUT);
  pinMode(Mist, OUTPUT);
  pinMode(Spindle_Enable, OUTPUT);
  pinMode(Spindle_PWM, OUTPUT);
  pinMode(Step_Enable, OUTPUT);

  pinMode(Hold, INPUT);
  pinMode(Resume, INPUT);
  pinMode(Abort, INPUT);
  pinMode(Door , INPUT);
  pinMode(Z_Probe, INPUT);
  pinMode(Z_Limit, INPUT);
  pinMode(Y_Limit , INPUT);
  pinMode(X_Limit , INPUT);
  
  digitalWrite(Step_Enable, HIGH);   
  digitalWrite(X_DIR_OUT, HIGH); 
  digitalWrite(Y_DIR_OUT, HIGH);   
  digitalWrite(Z_DIR_OUT, HIGH);   
  digitalWrite(Flood, HIGH);  
  digitalWrite(Mist, HIGH);   
  digitalWrite(Spindle_PWM, HIGH);  
  Serial.begin(9600); 
}
void loop() {
  
  Serial.print("Hold: ");   Serial.println(digitalRead(Hold));
  Serial.print("Resume: ");  Serial.println(digitalRead(Resume));
  Serial.print("Abort: ");  Serial.println(digitalRead(Abort));
  Serial.print("Door: ");  Serial.println(digitalRead(Door));
  Serial.print("Z_Probe: ");  Serial.println(digitalRead(Z_Probe));
  Serial.print("Z_Limit: ");  Serial.println(digitalRead(Z_Limit));
  Serial.print("Y_Limit: ");  Serial.println(digitalRead(Y_Limit));
  Serial.print("X_Limit: ");  Serial.println(digitalRead(X_Limit));
  
  digitalWrite(X_STEP_OUT, HIGH);   
  digitalWrite(Y_STEP_OUT, HIGH);   
  digitalWrite(Flood, HIGH);   
  digitalWrite(Mist, HIGH);   
  digitalWrite(Spindle_PWM, HIGH);   
  delay(1000);                       
  digitalWrite(X_STEP_OUT, LOW);
  digitalWrite(Y_STEP_OUT, LOW);
  digitalWrite(Z_STEP_OUT, LOW);   
  digitalWrite(Flood, LOW);  
  digitalWrite(Mist, LOW);   
  digitalWrite(Spindle_PWM, LOW);   
  delay(1000);                      
}

.So, and now...?

bdring, 2022-08-23:
What is WebIO.
What is JackLAB

Links please.

ClausElectronics, 2022-08-23:
Sorry, I meant:  ESP3D for FluidNC

Here are the Links:
https://github.com/bdring/Grbl_Esp32/issues/171#issuecomment-534652692
https://github.com/bdring/Grbl_Esp32/issues/171#issuecomment-552127326

ClausElectronics, 2022-08-23:
<Idle|MPos:10.000,0.000,0.000|FS:0,0>
$J=G91 G21 F1000 X100
<Idle|MPos:10.000,0.000,0.000|FS:0,0>
ok
<Jog|MPos:10.004,0.000,0.000|FS:90,0>
<Jog|MPos:48.382,0.000,0.000|FS:1000,0>
<Jog|MPos:96.545,0.000,0.000|FS:1000,0>
<Idle|MPos:110.000,0.000,0.000|FS:0,0>
<Idle|MPos:110.000,0.000,0.000|FS:0,0|WCO:0.000,0.000,0.000>
<Idle|MPos:110.000,0.000,0.000|FS:0,0|Ov:100,100,100>
...
It seems to work, I have to connect the motors ....

thank you - it was the wrong version...

But do you have an answer about the FluidTerm. I started it directly in the folder of the exe file...
And how it will work? USB and Wifi during the same time?

ClausElectronics, 2022-08-23:
Win 8.1

bdring, 2022-08-23:
Can you try it on a more modern computer?

bdring, 2022-08-23:
You can upload config files via WebUI, but it is more difficult to debug problems. you must also reboot after uploading, so that is a slow process when using the WebUI.

ClausElectronics, 2022-08-23:
Yes, I know, I already uploaded the config-file. And it works :)  First I will try another Serial-Interface-program until I have Win 10 or 11.

closed 2022-08-23 by bdring