bdring/FluidNC #579: Problem:

ModuloFS, 2022-08-21:
### Controller Board

Selfmade

### Help From Board Vendor

- [ ] Yes
- [ ] No
- [X] Not Applicable

### Machine Description

Gantry router Ratrig Killerbee with DM556T stepper driver and dual y motors and limit switches.

### Configuration file

```yaml
name: "ModuloController"
board: "ModuloControl V1"

stepping:
  engine: RMT
  idle_ms: 255
  dir_delay_us: 1
  pulse_us: 5
  disable_delay_us: 5

axes:
  shared_stepper_disable_pin: gpio.23
  
  x:
    steps_per_mm: 400.000
    max_rate_mm_per_min: 4000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 550.000
    soft_limits: true
    hard_limits: true
    homing:
      cycle: 2
      mpos_mm: 550.000
      positive_direction: true
    
    motor0:
      limit_pos_pin: gpio.35:low
      stepstick:
        direction_pin: gpio.27
        step_pin: gpio.26
    motor1:
      null_motor:

  y:
    steps_per_mm: 400.000
    max_rate_mm_per_min: 4000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 750.000
    soft_limits: true
    hard_limits: true
    homing:
      cycle: 3
      mpos_mm: 750.000
      positive_direction: true

    motor0:
      limit_pos_pin: gpio.39:low
      stepstick:
        direction_pin: gpio.32:low
        step_pin: gpio.25
    motor1:
      limit_pos_pin: gpio.36:low
      stepstick:
        direction_pin: gpio.22:low
        step_pin: gpio.21

  z:
    steps_per_mm: 400.000
    max_rate_mm_per_min: 4000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 150.000
    soft_limits: true
    hard_limits: true
    homing:
      cycle: 1
      mpos_mm: 150.000
      positive_direction: true

    motor0:
      limit_pos_pin: gpio.34:low
      stepstick:
        direction_pin: gpio.18
        step_pin: gpio.19
    motor1:
      null_motor:

       
probe:
  pin: gpio.4:high
  
control:
  safety_door_pin: gpio.33:low
  reset_pin: NO_PIN
  feed_hold_pin: NO_PIN

relay:
  direction_pin: NO_PIN
  output_pin: gpio.5
  enable_pin: NO_PIN
  disable_with_s0: false
  s0_with_disable: true
  spinup_ms: 0
  spindown_ms: 0
  tool_num: 0
  speed_map: 0=0.000% 0=100.000% 1=100.000%
```

### Startup Messages

```text
[MSG:INFO: FluidNC v3.5.1]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
[MSG:INFO: Local filesystem type is SPIFFS]
[MSG:INFO: Configuration file:config_ratrig_killerbee.yaml]
[MSG:WARN: Ignored key hard_limits]
[MSG:WARN: Ignored key hard_limits]
[MSG:WARN: Ignored key hard_limits]
[MSG:INFO: Machine ModuloController]
[MSG:INFO: Board ModuloControl V1]
[MSG:INFO: SPI not defined]
[MSG:INFO: Stepping:RMT Pulse:5us Dsbl Delay:5us Dir Delay:1us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Shared stepper disable gpio.23]
[MSG:INFO: Axis X (0.000,550.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:gpio.26 Dir:gpio.27 Disable:NO_PIN]
[MSG:INFO:     Pos Limit gpio.35:low]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Y (0.000,750.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:gpio.25 Dir:gpio.32:low Disable:NO_PIN]
[MSG:INFO:     Pos Limit gpio.39:low]
[MSG:INFO:   Motor1]
[MSG:INFO:     stepstick Step:gpio.21 Dir:gpio.22:low Disable:NO_PIN]
[MSG:INFO:     Pos Limit gpio.36:low]
[MSG:INFO: Axis Z (0.000,150.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:gpio.19 Dir:gpio.18 Disable:NO_PIN]
[MSG:INFO:     Pos Limit gpio.34:low]
[MSG:INFO:   Motor1]
[MSG:INFO: safety_door_pin gpio.33:low]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Relay Spindle Ena:NO_PIN Out:gpio.5 Dir:NO_PIN]
[MSG:INFO: Using spindle Relay]
[MSG:INFO: Probe Pin: gpio.4]
[MSG:INFO: AP SSID FluidNC IP 192.168.178.1 mask 255.255.255.0 channel 1]
[MSG:INFO: AP started]
[MSG:INFO: WiFi on]
[MSG:INFO: Captive Portal Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
```

### User Interface Software

FluidTerm or gSender

### What happened?

If i activate (pull low) the safety_door_pin i geht the Message in the Console "[MSG:INFO: Check door]". Then i reset the hardware state of the pin back to high. Up to now i only get answers from the controller with the "?" character (this is the answer in this state: "<Door:0|MPos:0.000,0.000,0.000|FS:0,0|Pn:XYZ>"). All "$" commands are not responded by the controller. If i want to come back to a Idle state i want to use the "~" character, because up to now the software gSender is not able to send ctrl + X. But the "~" is not working, i have to send this character multible times to geht the controller back to a Idle state and to responding to my inputs. This is booth happening at fluidterm and at the console of gSender (https://github.com/Sienci-Labs/gsender).
Is this a normal behaviour?

bdring, 2022-08-21:
I can confirm some of your issue. I tested versions 3.4.4, 3.5.1 and branch EventQueue2. Everything works perfectly for me, except that I need to send 2 `~` commands.

```
[MSG:INFO: '$H'|'$X' to unlock]
$H
ok
g28
ok
g1 f200 x50
ok
; activated latching door switch
[MSG:INFO: Check door]
; machine runs the parking sequence
<Door:1|WPos:12.999,140.000,-5.000|Bf:14,128|FS:0,0|WCO:0.000,-150.000,0.000>
; deactivated door switch
<Door:0|WPos:12.999,140.000,-5.000|Bf:14,128|FS:0,0|WCO:0.000,-150.000,0.000>
; sent ~
[CycleStart]
; machine returns to location prior to parking
; sent ~
[CycleStart]
; machine completes interrupted g1 f200 x50
```

We can put this on our list of things to investigate. It does not seem like a show stopping problem at this time. (almost a desirable feature)

bdring, 2022-08-21:
The problem has been fixed in the EventQueue2 branch (not the release). That is a working branch for a couple of open issues, so it is a future release candidate.

closed 2022-08-21 by bdring