bdring/FluidNC #580: Problem: Configuration for MKS DLC32 v2.1 & A4988 steppers

molekm, 2022-08-21:
### Controller Board

MKS DLC32 v2.1

### Help From Board Vendor

- [ ] Yes
- [ ] No
- [X] Not Applicable

### Machine Description

Default China laser K40 

### Configuration file

```yaml
board: MKS-DLC32 V2.1
name: K40 Laser
meta: (01.01.2022) by Skorpi

kinematics:
  Cartesian:

stepping:
  engine: I2S_STATIC
  idle_ms: 0
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0
axes:
  shared_stepper_disable_pin: I2SO.0
  x:
    steps_per_mm: 157.750
    max_rate_mm_per_min: 18000.000
    acceleration_mm_per_sec2: 1500.000
    max_travel_mm: 325.000
    soft_limits: true
    homing:
      cycle: 1
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 300.000
      seek_mm_per_min: 5000.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.36
      hard_limits: false
      pulloff_mm: 2.000
      stepstick:
        step_pin: I2SO.1
        direction_pin: I2SO.2

  y:
    steps_per_mm: 157.750
    max_rate_mm_per_min: 12000.000
    acceleration_mm_per_sec2: 300.000
    max_travel_mm: 220.000
    soft_limits: true
    homing:
      cycle: 1
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 300.000
      seek_mm_per_min: 5000.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.35
      hard_limits: false
      pulloff_mm: 2.000
      stepstick:
        step_pin: I2SO.5
        direction_pin: I2SO.6:low

  z:
    steps_per_mm: 157.750
    max_rate_mm_per_min: 12000.000
    acceleration_mm_per_sec2: 500.000
    max_travel_mm: 80.000
    soft_limits: true
    homing:
      cycle: 0
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 300.000
      seek_mm_per_min: 1000.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.34
      hard_limits: false
      pulloff_mm: 1.000
      stepstick:
        step_pin: I2SO.3
        direction_pin: I2SO.4

i2so:
  bck_pin: gpio.16
  data_pin: gpio.21
  ws_pin: gpio.17

spi:
  miso_pin: gpio.12
  mosi_pin: gpio.13
  sck_pin: gpio.14

sdcard:
  cs_pin: gpio.15
  card_detect_pin: NO_PIN

control:
  safety_door_pin: NO_PIN
  reset_pin: NO_PIN
  feed_hold_pin: NO_PIN
  cycle_start_pin: NO_PIN
  macro0_pin: gpio.33:low:pu
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN

macros:
  startup_line0:
  startup_line1:
  macro0: $SD/Run=lasertest.gcode
  macro1: $SD/Run=home.gcode
  macro2:
  macro3:

coolant:
  flood_pin: NO_PIN
  mist_pin: NO_PIN
  delay_ms: 0

probe:
  pin: gpio.22
  check_mode_start: true

Laser:
  pwm_hz: 5000
  #L on Beeper / IN on TTL
  output_pin: gpio.32
  enable_pin: I2SO.7
  disable_with_s0: false
  s0_with_disable: false
  tool_num: 0
  speed_map: 0=0.000% 0=12.500% 1700=100.000%
# 135=0mA 270=5mA 400=10mA 700=16mA
user_outputs:
  analog0_pin: NO_PIN
  analog1_pin: NO_PIN
  analog2_pin: NO_PIN
  analog3_pin: NO_PIN
  analog0_hz: 5000
  analog1_hz: 5000
  analog2_hz: 5000
  analog3_hz: 5000
  digital0_pin: NO_PIN
  digital1_pin: NO_PIN
  digital2_pin: NO_PIN
  digital3_pin: NO_PIN

start:
  must_home: false

# 5,18,19,22,23,25,26,27,32,33,39,I2SO.7
# SDA 0 / SCL 4
```

### Startup Messages

```text
$/axes/x/max_travel_mm
$/axes/x/homing/mpos_mm
$/axes/x/homing/positive_direction
$/axes/y/max_travel_mm
$/axes/y/homing/mpos_mm
$/axes/y/homing/positive_direction
$G
$/axes/x/max_travel_mm=325.000
ok
$/axes/x/homing/mpos_mm=0.000
ok
$/axes/x/homing/positive_direction=false
ok
$/axes/y/max_travel_mm=220.000
ok
$/axes/y/homing/mpos_mm=0.000
ok
$/axes/y/homing/positive_direction=false
ok
[GC:G0 G54 G17 G21 G90 G94 M5 M9 T0 F0 S0]
ok
<Idle|MPos:0.000,0.000,0.000|FS:0,0|Pn:PZ|WCO:0.000,0.000,0.000>
<Idle|MPos:0.000,0.000,0.000|FS:0,0|Pn:PZ|Ov:100,100,100>
```

### User Interface Software

WebUI

### What happened?

The motors won't move / home. Endstops seems to work when manually home it.

$/axes/x/max_travel_mm
$/axes/x/homing/mpos_mm
$/axes/x/homing/positive_direction
$/axes/y/max_travel_mm
$/axes/y/homing/mpos_mm
$/axes/y/homing/positive_direction
$G
<Alarm|MPos:-357.502,0.000,0.000|FS:0,0|Pn:PZ>
<Alarm|MPos:-357.502,0.000,0.000|FS:0,0|Pn:PZ>
<Alarm|MPos:-357.502,0.000,0.000|FS:0,0|Pn:PZ>
<Alarm|MPos:-357.502,0.000,0.000|FS:0,0|Pn:PZ>
$HX
<Home|MPos:-0.228,0.000,0.000|FS:5000,0|Pn:PZ>
<Home|MPos:-148.127,0.000,0.000|FS:5000,0|Pn:PZ>
[MSG:ERR: Homing fail]

### Other Information

For sure it is wrong config for A4988, however cannot locate right setting for this stepper & MSK v2.1

bdring, 2022-08-21:
Please post your startup messages.

http://wiki.fluidnc.com/en/support/requesting_help#fluidnc-startup-messages

molekm, 2022-08-21:
> Please post your startup messages.
> 
> http://wiki.fluidnc.com/en/support/requesting_help#fluidnc-startup-messages

Sorry took the wrong message ;), now message is from the startup :)

bdring, 2022-08-21:
Try using version 3.4.4. It is the [version we recommend](http://wiki.fluidnc.com/en/installation#using-pre-compiled-files) right now.

Before you do that though, try sending `$message/level=debug` and paste the response. It should show you more about why it failed.

Here is the report from a successful home.

```
[MSG:DBG: Homing X]
[MSG:DBG: PrePulloff]
[MSG:DBG: Fast approach]
[MSG:DBG: X target -198.000 rate 600.000]
[MSG:DBG: Pulloff0]
[MSG:DBG: X target 3.000 rate 300.000]
[MSG:DBG: Slow approach]
[MSG:DBG: X target -3.300 rate 300.000]
[MSG:DBG: Pulloff1]
[MSG:DBG: X target 3.000 rate 300.000]
```

JemandAnders, 2022-09-03:
I am using the DRV8825 and the motors are moving and homing. I used the same config as you as a basis. The motors never move, right?
I am quite new to FluidNC. I can see, the MC is "thinking" the motors are turning. The alarm goes active at -357.502 which is max travel x * 1.1.  The alarm goes active some seconds after activating homing? 
I am guessing on a hardware problem. Drivers plugged in correctly? ;-)

closed 2024-02-03 by bdring