bdring/FluidNC #581: Problem: X Axis driver test failed

lagnat, 2022-08-21:
### Controller Board

6-pack

### Help From Board Vendor

- [ ] Yes
- [ ] No
- [X] Not Applicable

### Machine Description

CNC, XXYZ, TMC5160, Makita spindle

### Configuration file

```yaml
board: 6 Pack
name: 6 Pack TMC5160 XXYZ
meta: 
stepping:
  engine: I2S_stream
  idle_ms: 250
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0
  segments: 12

axes:
  shared_stepper_disable_pin: NO_PIN
  shared_stepper_reset_pin: NO_PIN
  x:
    steps_per_mm: 160.000
    max_rate_mm_per_min: 6000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 800.000
    soft_limits: false
    homing:
      cycle: 1
      allow_single_axis: true
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 200.000
      seek_mm_per_min: 3000.000
      settle_ms: 250
      seek_scaler: 1.000
      feed_scaler: 1.500

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 4.500
      tmc_5160:
        step_pin: I2SO.2
        direction_pin: I2SO.1
        disable_pin: I2SO.0
        r_sense_ohms: 0.075
        run_amps: 1.800
        hold_amps: 1.250
        microsteps: 8
        toff_disable: 0
        toff_stealthchop: 5
        use_enable: false
        cs_pin: I2SO.3
        spi_index: -1
        run_mode: StealthChop
        homing_mode: StealthChop
        stallguard: 0
        stallguard_debug: false
        toff_coolstep: 3
        tpfd: 4

    motor1:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 5.000
      tmc_5160:
        step_pin: I2SO.6
        direction_pin: I2SO.5
        disable_pin: I2SO.4
        r_sense_ohms: 0.075
        run_amps: 1.800
        hold_amps: 1.250
        microsteps: 8
        toff_disable: 0
        toff_stealthchop: 5
        use_enable: false
        cs_pin: I2SO.7
        spi_index: -1
        run_mode: StealthChop
        homing_mode: StealthChop
        stallguard: 0
        stallguard_debug: false
        toff_coolstep: 3
        tpfd: 4

  y:
    steps_per_mm: 160.000
    max_rate_mm_per_min: 6000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 550.000
    soft_limits: false
    homing:
      cycle: 1
      allow_single_axis: true
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 200.000
      seek_mm_per_min: 3000.000
      settle_ms: 250
      seek_scaler: 1.000
      feed_scaler: 1.500

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 5.000
      tmc_5160:
        step_pin: I2SO.18
        direction_pin: I2SO.17
        disable_pin: I2SO.16
        r_sense_ohms: 0.075
        run_amps: 0.750
        hold_amps: 0.500
        microsteps: 8
        toff_disable: 0
        toff_stealthchop: 5
        use_enable: false
        cs_pin: I2SO.19
        spi_index: -1
        run_mode: StealthChop
        homing_mode: StealthChop
        stallguard: 0
        stallguard_debug: false
        toff_coolstep: 3
        tpfd: 4

  z:
    steps_per_mm: 400.000
    max_rate_mm_per_min: 2000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 300.000
    soft_limits: false
    homing:
      cycle: 1
      allow_single_axis: true
      positive_direction: true
      mpos_mm: 0.000
      feed_mm_per_min: 200.000
      seek_mm_per_min: 1000.000
      settle_ms: 250
      seek_scaler: 1.000
      feed_scaler: 1.500

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 1.000
      tmc_5160:
        step_pin: I2SO.22
        direction_pin: I2SO.21
        disable_pin: I2SO.20
        r_sense_ohms: 0.075
        run_amps: 0.750
        hold_amps: 0.500
        microsteps: 8
        toff_disable: 0
        toff_stealthchop: 5
        use_enable: false
        cs_pin: I2SO.23
        spi_index: -1
        run_mode: StealthChop
        homing_mode: StealthChop
        stallguard: 0
        stallguard_debug: false
        toff_coolstep: 3
        tpfd: 4

kinematics:
  Cartesian:

i2so:
  bck_pin: gpio.22
  data_pin: gpio.21
  ws_pin: gpio.17

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  cs_pin: gpio.5
  card_detect_pin: NO_PIN

control:
  safety_door_pin: NO_PIN
  reset_pin: NO_PIN
  feed_hold_pin: NO_PIN
  cycle_start_pin: NO_PIN
  macro0_pin: NO_PIN
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN

coolant:
  flood_pin: NO_PIN
  mist_pin: NO_PIN
  delay_ms: 0

probe:
  pin: NO_PIN
  check_mode_start: true

macros:
  startup_line0: 
  startup_line1: 
  macro0: 
  macro1: 
  macro2: 
  macro3: 

start:
  must_home: false
  deactivate_parking: false
  check_limits: false

user_outputs:
  analog0_pin: NO_PIN
  analog1_pin: NO_PIN
  analog2_pin: NO_PIN
  analog3_pin: NO_PIN
  analog0_hz: 5000
  analog1_hz: 5000
  analog2_hz: 5000
  analog3_hz: 5000
  digital0_pin: NO_PIN
  digital1_pin: NO_PIN
  digital2_pin: NO_PIN
  digital3_pin: NO_PIN

arc_tolerance_mm: 0.002
junction_deviation_mm: 0.010
verbose_errors: false
report_inches: false
enable_parking_override_control: false
use_line_numbers: false
planner_blocks: 16
NoSpindle:
```

### Startup Messages

```text
[MSG:INFO: FluidNC v3.5.0]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
[MSG:INFO: Local filesystem type is SPIFFS]
[MSG:INFO: Configuration file:config.yaml]
[MSG:INFO: Machine 6 Pack TMC5160 XXYZ]
[MSG:INFO: Board 6 Pack]
[MSG:INFO: I2SO BCK:gpio.22 WS:gpio.17 DATA:gpio.21]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN]
[MSG:INFO: Stepping:I2S_stream Pulse:4us Dsbl Delay:0us Dir Delay:1us Idle Delay:250ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (0.000,800.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_5160 Step:I2SO.2 Dir:I2SO.1 CS:I2SO.3 Disable:I2SO.0 Index:-1 R:0.075]
[MSG:INFO:   Motor1]
[MSG:INFO:     tmc_5160 Step:I2SO.6 Dir:I2SO.5 CS:I2SO.7 Disable:I2SO.4 Index:-1 R:0.075]
[MSG:INFO: Axis Y (0.000,550.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_5160 Step:I2SO.18 Dir:I2SO.17 CS:I2SO.19 Disable:I2SO.16 Index:-1 R:0.075]
[MSG:INFO: Axis Z (-300.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_5160 Step:I2SO.22 Dir:I2SO.21 CS:I2SO.23 Disable:I2SO.20 Index:-1 R:0.075]
[MSG:ERR: X Axis driver test failed. Check motor power]
[MSG:ERR: X2 Axis driver test failed. Check motor power]
[MSG:ERR: Y Axis driver test failed. Check motor power]
[MSG:ERR: Z Axis driver test failed. Check motor power]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Using spindle NoSpindle]
[MSG:INFO: STA SSID is not set]
[MSG:INFO: AP SSID FluidNC IP 192.168.0.1 mask 255.255.255.0 channel 1]
[MSG:INFO: AP started]
[MSG:INFO: WiFi on]
[MSG:INFO: Captive Portal Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
```

### User Interface Software

FluidTerm

### What happened?

Motors don't move with G0

lagnat, 2022-08-21:
I started with the XYYZ sample config and started making changes from there.  Thanks!

bdring, 2022-08-22:
Your pin numbering is wrong for the motors.

[See this wiki page](https://github.com/bdring/6-Pack_CNC_Controller/wiki/Socket-Pin-Number-Mapping#fluidnc)

lagnat, 2022-08-22:
Hi Bart.  I swear I tripple checked the numbers but you're right.  I had a couple that were wrong.  As for that page, it says to use `ms3_pin` instead of `cs_pin` for motors 1, 2, and 3 however that didn't work.
```
[MSG:ERR: Initialization error at /axes/x/motor0/tmc_5160: TMC cs_pin: pin must be configured]
```

bdring, 2022-08-22:
For SPI drivers replace with `cs_pin;`

bdring, 2022-08-22:
Post your new config

lagnat, 2022-08-22:
The wording is confusing (to me).  I understood that to mean that I needed to specify ms3_pin because I'm using SPI but it's the other way around.  With cs_pin and the corrected numbering, I have the motors passing the self-test now.  The Z motor doesn't seem to move right and one of the X motors doesn't move at all, but I'll try to debug that first.  I'll probably downgrade to 3.4.4 first.  I'll close this issue and open a new one if I have further problems.  Thanks!

closed 2022-08-22 by lagnat