bdring/FluidNC #584: Feature: DC motor control of spindle

solawc, 2022-08-24:
### Machine Context

3018 CNC

### Feature Description

Acceleration and deceleration start of spindle DC motor， I try to modify the code so that the PWM is separated by 1ms, and the duty ratio of PWM is increased a little, and the spindle start will become softer. I have replaced a larger spindle, and the motherboard will not restart during startup. But obviously this is not the best way. I just did a test.

### Other Approaches

I don't have a better way to kill him now

### How I Can Help

When controlling the spindle output, the PWM value should be modified step by step instead of directly outputting the corresponding duty ratio.

solawc, 2022-08-24:
This is the first time I use this template. The format may not be very good. Please forgive me.

MitchBradley, 2022-08-24:
We have an experimental version of spindle soft-start already, in [this branch](https://github.com/bdring/FluidNC/tree/BartsFixedIntervalRamp).  We might deploy it eventually, after we get caught up (if we ever do get caught up).

solawc, 2022-08-25:
> We have an experimental version of spindle soft-start already, in [this branch](https://github.com/bdring/FluidNC/tree/BartsFixedIntervalRamp). We might deploy it eventually, after we get caught up (if we ever do get caught up).

I'm glad that your method is much the same as mine. I didn't calculate the duty ratio correctly, probably because it was too hasty. I noticed_ ramp_ Interval = 50, which means that the fastest time to change the duty ratio is 50ms. During my own test, I used 2ms. At the beginning, I used 10ms. I found that the whole machine moved slowly. Of course, if the slice mode is uniform, it can make the startup more stable.

closed 2024-12-12 by bdring