bdring/FluidNC #585: il laser non si spegne in modalita M3

distebia, 2022-08-24:
### Controller Board

mks dlc 2.1 

### Help From Board Vendor

- [ ] Yes
- [ ] No
- [X] Not Applicable

### Machine Description

mks dlc32 v2.1 

### Configuration file

```yaml
board: MKS DLC32 2.1
name: MKS DLC32 XYZ
meta: (08/07/2022) Biagio

arc_tolerance_mm: 0.002
junction_deviation_mm: 0.010
verbose_errors: false
report_inches: false
enable_parking_override_control: false
use_line_numbers: false
planner_blocks: 16

stepping:
  engine: I2S_STATIC
  idle_ms: 255
  pulse_us: 10
  dir_delay_us: 1
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: i2so.0
  x:
    steps_per_mm: 79.973
    max_rate_mm_per_min: 20000.000
    acceleration_mm_per_sec2: 1000.000
    max_travel_mm: 1226.000
    soft_limits: true
    homing:
      cycle: 1
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 1000.000
      seek_mm_per_min: 5000.000
      settle_ms: 250.000
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.36:low
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm:1.500
      stepstick:
        step_pin:  i2so.1
        direction_pin: I2SO.2

  y:
    steps_per_mm: 80.194
    max_rate_mm_per_min: 10000.000
    acceleration_mm_per_sec2: 300.000
    max_travel_mm: 898.000
    soft_limits: true
    homing:
      cycle: 1
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 1000.000
      seek_mm_per_min: 5000.000
      settle_ms: 250.000
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.35:low
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm:1.500
      stepstick:
        step_pin: I2SO.5
        direction_pin: I2SO.6

  z:
    steps_per_mm: 640.000
    max_rate_mm_per_min: 200.000
    acceleration_mm_per_sec2: 50.000
    max_travel_mm: 170.000
    soft_limits: true
    homing:
      cycle: 0
      positive_direction: true
      mpos_mm: 0.000
      feed_mm_per_min: 200.000
      seek_mm_per_min: 200.000
      settle_ms: 250.000
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.34:low
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm:1.000
      stepstick:
        step_pin: I2SO.3
        direction_pin: I2SO.4:low

i2so:
  bck_pin: gpio.16
  data_pin: gpio.21
  ws_pin: gpio.17

spi:
  miso_pin: gpio.12
  mosi_pin: gpio.13
  sck_pin: gpio.14

sdcard:
  card_detect_pin: NO_PIN
  cs_pin: gpio.15

coolant:
  flood_pin: gpio.4
  mist_pin: NO_PIN
  delay_ms: 0

probe:
  pin: gpio.2
  check_mode_start: false

start:
  must_home: false

Laser:
  pwm_hz: 10000
  output_pin: gpio.32
  enable_pin: NO_PIN
  disable_with_s0: false
  s0_with_disable: true
# speed_map: 0=10.000% 1000=100.000%
  speed_map: 0=0.000% 0=10.000% 1000=100.000%
# speed_map: 0=0.000% 1000=100.000%
```

### Startup Messages

```text
no errore
```

### User Interface Software

lightburn

### What happened?

Salve, ho un laser co2 autocostruito e, fra alimentatore e tubo, in arduino uno impostavo in config.h il comando PWM_MIN_VALUE=25 per accendere il laser al 1% in Lightburn (LB configurato con SMax=1000). Passo a MKS DLC32 e carico FluidNc 3.4.4 per un problema di scansione e funziona perfettamente. Riesco ad accendere il laser al 1% (quindi il mio Speed_map funziona) e ho M4 all'avvio...tutto perfetto...Ieri ho effettuato un taglio con Costant power mode, quindi M3, e il laser non si spegneva nei movimenti rapidi...dove sbaglio?

bdring, 2022-08-25:
I think this was fixed. It should show up in the next release.

See https://github.com/bdring/FluidNC/issues/486

distebia, 2022-08-25:
salve e grazie per la risposta. purtroppo non posso attendere le nuova versione in quanto, per problemi di scansione e quindi usare "stepping:  engine: I2S_STATIC al posto di STREAM, sono bloccato alla 3.4.4. posso modificare qualcosa nella speed_map? la mia è questa: speed_map: 0=0.000% 0=10.000% 1000=100.000% perchè ho bisogno di aumentare i mV di uscita per accendere il laser a 1%.

MitchBradley, 2022-08-25:
Changing the speed map will not fix the G0 problem

distebia, 2022-08-25:
Ok. quindi è proprio un bug che avete corretto con l'ultimo aggiornamento? oltre ad usare M4 posso fare altro restando su 3.4.4?

bdring, 2022-08-25:
Try your speed map with m4

distebia, 2022-08-25:
ok.

distebia, 2022-08-25:
ma se metto l'ultima versione e uso I2S_STATIC ho sempre problemi di scansione?

MitchBradley, 2022-08-25:
We are trying to fix the problems with the 3.5 series but instead of working on them, we are overwhelmed with people asking for new features.

distebia, 2022-08-25:
quindi per ora utilizzo la M4 con la 3.4.4 e tengo d'occhio gli aggiornamenti. Grazie

distebia, 2022-09-07:
Salve, la versione 3.6.0 risolve i problemi della modalità M3  che non spegne il laser per i movimenti G0? inoltre per la sincronizzazione degli stepper al pwm #537 siamo sempre limitati ad avere  IS2_STATIC o è stato risolto? grazie

MitchBradley, 2022-09-07:
Regarding M3, why don't you try it and see?

Regarding I2S_STATIC, that may never be solved.  I2S_STREAM inherently adds a delay between power adjustment and stepping.  Compensating for that delay would be difficult, and since there is a workaround, we have little incentive to make the complex changes that would be necessary.

distebia, 2022-09-07:
ok, stasera provo la 3.6.0 e vi faccio sapere per il problema M3. lascio sempre I2S_STATIC. il mio file yaml con la nuova gestione dei finecorsa ha bisogno di modifiche o è sempre lo stesso?

MitchBradley, 2022-09-07:
Same yaml

distebia, 2022-09-07:
ok, per passare dalla 3.4.4 alla 3.6.0 devo lanciare install-wifi.bat direttamente oppure 1- erase.bat + 2- install-fs.bat + 3- install-wifi.bat?

Skorpi08, 2022-09-07:
https://www.youtube.com/watch?v=NcOWp8gJRXo&ab_channel=Skorpi

MitchBradley, 2022-09-07:
install-wifi.bat

Please try stuff instead of asking a lot of simple questions.

distebia, 2022-09-08:
Mi scuso per le troppe domande, è perchè la pagina wiki non è tanto chiara, parla di nuova installazione ma non parla di un aggiornamento di versione...forse causa della traduzione di google...

distebia, 2022-09-08:
Salve, ho aggiornato il programma dalla 3.4.4 alla 3.6.0 mediante install-wifi.bat e ho risolto il problema M3, il laser quando esegue i movimenti G0 si spegne come atteso. continuo ad usate I2S_STATIC e la scansione è perfetta. Grazie ancora per l'aiuto

distebia, 2022-09-08:
alla fine sono ritornato alla 3.4.4 in quanto ho problemi con i finecorsa, in particolare Z (con 3.4.4 tutto ok). Il finecorsa Z è un sensore induttivo pnp e ho collegato tramite pc417 e pulldown al pin. Ripeto...con la 3.4.4 questa configurazione funziona perfettamente mentre con la 3.6.0 su 10 volte 7 non riconosce l'ostacolo, anche se il comando ? riconosce il finecorsa impegnato e libero.

MitchBradley, 2022-09-08:
Please show us a schematic of how the sensor and pc417 is connected.

MitchBradley, 2022-09-08:
You have the Z limit set for active low:
```
    motor0:
      limit_neg_pin: gpio.34:low
```
but that is an active-high circuit

distebia, 2022-09-08:
z:
    steps_per_mm: 640.000
    max_rate_mm_per_min: 200.000
    acceleration_mm_per_sec2: 50.000
    max_travel_mm: 170.000
    soft_limits: true
    homing:
      cycle: 0
      positive_direction: true
      mpos_mm: 0.000
      feed_mm_per_min: 200.000
      seek_mm_per_min: 200.000
      settle_ms: 250.000
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.34
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm:1.000
      stepstick:
        step_pin: I2SO.3
        direction_pin: I2SO.4:low

questa è la mia attuale configurazione...ora provvedo ad aggiornare in alto

MitchBradley, 2022-09-09:
Try the test release https://github.com/bdring/FluidNC/releases/tag/TestPullups

I found an obscure bug that only affects limit pins in active high mode with high-value pulldown resistors.

distebia, 2022-09-09:
la versione suggerita funziona perfettamente per la mia configurazione del finecorsa Z. Mi chiedo se X e Y (finecorsa) hanno subito, con questa versione, delle modifiche al pullup interno. Funzionano come prima ma sono uguali a prima?

MitchBradley, 2022-09-09:
All the switches are the same.

Can we close this ticket?

distebia, 2022-09-09:
Si certo, seguirà una versione stabile? con questa non funziona la radio wifi

MitchBradley, 2022-09-09:
Of course a stable version will follow - but I cannot do release engineering and answer a neverending stream of questions at the same time.

closed 2022-09-09 by MitchBradley