bdring/FluidNC #586: Feature:  M66 Command.

mac7988, 2022-08-24:
### Machine Context

Custom bead machine.

### Feature Description

I was wondering if M66 can be implemented.

I want to index an Axis until a sensor is read the go to the next line of the gcode

### Other Approaches

Someone on discord suggested to use an external mcu to send dynanmic code via serial.

I think this makes it way to complex.

### How I Can Help

I can be provide funds for the time investment.

bdring, 2022-08-25:
https://linuxcnc.org/docs/html/gcode/m-code.html#mcode:m66

I can add that to our dev issues tracker for future consideration.  It does not sound too complicated, but we are working on more basic things right now.

mac7988, 2022-08-27:
> https://linuxcnc.org/docs/html/gcode/m-code.html#mcode:m66
> 
> I can add that to our dev issues tracker for future consideration. It does not sound too complicated, but we are working on more basic things right now.

Thank you

mac7988, 2022-12-16:
Did you guys get a chance to look at this? Just wondering.

bdring, 2022-12-16:
No, we are working on other things right now. This would have limited appeal, so there is no priority. Donations to the project are very low, so we have been working on our personal priorities.

This is a relatively simple command, but could have far reaching implications, like what state are you in while waiting. Hold is one option. If we feel we need a new state, that affects the WebUI and senders.

bdring, 2022-12-16:
The "hold" state is simple, but does not work with the parameters.
```
P- - specifies the digital input number from 0 to 3.

E- - specifies the analog input number from 0 to 3.

L- - specifies the wait mode type.

Mode 0: IMMEDIATE - no waiting, returns immediately. The current value of the input is stored in parameter #5399

Mode 1: RISE - waits for the selected input to perform a rise event.

Mode 2: FALL - waits for the selected input to perform a fall event.

Mode 3: HIGH - waits for the selected input to go to the HIGH state.

Mode 4: LOW - waits for the selected input to go to the LOW state.

Q- - specifies the timeout in seconds for waiting. If the timeout is exceeded, the wait is interrupt, and the variable #5399 will be holding the value -1. The Q value is ignored if the L-word is zero (IMMEDIATE). A Q value of zero is an error if the L-word is non-zero.

Mode 0 is the only one permitted for an analog input.
```

bdring, 2022-12-16:
The M0 command will put the machine in hold mode. Could the sensor be the resume button?

mac7988, 2022-12-16:
I am trying to machine small parts that are indexing on a linear rail( not
part of your controller) I want to put a proximity sensor to start the
machining (fluidnc) when the sensor is active.

MitchBradley, 2022-12-16:
Could this be done with an external microprocessor that sends feedhold/cyclestart bytes based on the sensor?

hsyngndz85, 2024-04-30:
M66 konusunda gelişme varmı

bdring, 2024-12-12:
We have M66 now.

http://wiki.fluidnc.com/en/features/supported_gcodes#m66-read-user-input

closed 2024-12-12 by bdring

emplast, 2025-01-09:
where can I find more information about analog or digital input and M-code M66
I mean coupling the value of either feed or Override with the sensor that goes into the Tinnybee board

MitchBradley, 2025-01-09:
That is not possible directly within FluidNC.  You might be able to do it with an external MCU that communicates with Tinybee over a UART connected via the EXP connector.  The MCU would read a sensor and send override + and - realtime characters via the UART.

In the future, please do not re-use closed issues.

emplast, 2025-01-09:
I assume communication via uart and via exp it is possible, you can do something more because it is supposed to control the speed or override, it would probably be easier via override because the whole machine can speed up or slow down the whole process in real timePiotr Majewski56-300 Milicz Polskaul. Kościuszki 48email: ***@***.***: +48 7883088939 sty

emplast, 2025-01-09:
is there a board where I can do this?Piotr Majewski56-300 Milicz Polskaul. Kościuszki 48email: ***@***.***: +48 7883088939 sty

Ams-Laser, 2025-02-06:
wil "M66 wait for input" be supported sometime ?
Thanks