bdring/FluidNC #587: TCP connection faliure - fluidnc-v3.5.1

pagabee, 2022-08-25:
### Controller Board

No board, only NodeMCU-32.

### Help From Board Vendor

- [ ] Yes
- [ ] No
- [ ] Not Applicable

### Machine Description

No machine.

### Configuration file

```yaml
No config file.
```

### Startup Messages

```text
**** Connected to 192.168.100.43 @ 23 baud ****
```

### User Interface Software

ugsplatform-win 2.0.12

### What happened?

If the fluidnc-v3.5.1 is installed, the TCP connection does not work.
The message in the console is short and the status is disconnected:

[image]

The TCP connection works fine with fluidnc-v3.4.4:

[image]

bdring, 2022-08-25:
Please provide complete startup messages.

http://wiki.fluidnc.com/en/support/requesting_help#fluidnc-startup-messages

pagabee, 2022-08-25:
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:12812
load:0x40080400,len:3032
entry 0x400805e4

[MSG:INFO: FluidNC v3.5.1]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
[MSG:INFO: Local filesystem type is SPIFFS]
[MSG:WARN: Cannot open configuration file:config.yaml]
[MSG:INFO: Using default configuration]
[MSG:INFO: Axes: using defaults]
[MSG:INFO: Machine Default (Test Drive)]
[MSG:INFO: Board None]
[MSG:INFO: SPI not defined]
[MSG:INFO: Stepping:RMT Pulse:4us Dsbl Delay:0us Dir Delay:0us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (-1000.000,0.000)]
[MSG:INFO: Axis Y (-1000.000,0.000)]
[MSG:INFO: Axis Z (-1000.000,0.000)]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Using spindle NoSpindle]
[MSG:INFO: STA SSID is not set]
[MSG:INFO: AP SSID FluidNC IP 192.168.0.1 mask 255.255.255.0 channel 1]
[MSG:INFO: AP started]
[MSG:INFO: WiFi on]
[MSG:INFO: Captive Portal Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]

Grbl 3.5 [FluidNC v3.5.1 (wifi) '$' for help]
[MSG:INFO: '$H'|'$X' to unlock]

breiler, 2022-09-20:
This is an issue with UGS. Do not connect to FluidNC with the `GRBL ESP32` firmware profile.
That firmware profile does not properly detect the controller when connecting. It expects a GRBL-welcome string when connecting.

Use the `FluidNC` firmware profile instead.

bdring, 2022-09-20:
If you ever need to change the "welcome string", you can do it with the `$Start/Message` setting.

http://wiki.fluidnc.com/en/features/commands_and_settings#start_message

pagabee, 2022-09-21:
Yes, you are right.
It works fine with the FluidNC firmware profile.

Thank you for your support!

closed 2022-09-21 by pagabee