bdring/FluidNC #588: Problem: config.yaml / 2 axis not available

wollo-fpv, 2022-08-26:
### Controller Board

ESPDUINO-32 with external stepper driver

### Help From Board Vendor

- [ ] Yes
- [X] No
- [ ] Not Applicable

### Machine Description

4 axis hotwire with external steppers

### Configuration file

```yaml
board: WeMOS D1 R32 
name: 4_axis_hotwire
meta: 20/08/2022

kinematics:
  Cartesian:
  

stepping:
  pulse_us: 2
  dir_delay_us: 1
  disable_delay_us: 0
  
  
axes:
  x:
    # X
    steps_per_mm: 320
    max_rate_mm_per_min: 18000.000
    acceleration_mm_per_sec2: 1500.000
    max_travel_mm: 1300.000
    soft_limits: true
    homing:
      cycle: 0
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 300.000
      seek_mm_per_min: 5000.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.25
      hard_limits: false
      pulloff_mm: 2.000
      standard_stepper:
        step_pin: gpio.22
        direction_pin: gpio.21
        disable_pin: gpio.18

  y:
    # Y
    steps_per_mm: 320
    max_rate_mm_per_min: 12000.000
    acceleration_mm_per_sec2: 300.000
    max_travel_mm: 1300.000
    soft_limits: true
    homing:
      cycle: 0
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 300.000
[... 97 lines cut ...]
[MSG:INFO:     standard_stepper Step:gpio.22 Dir:gpio.21 Disable:gpio.18]                                                                                                        
[MSG:INFO:     Neg Limit gpio.25]                                                                                                                                                
[MSG:INFO: Axis Y (0.000,1300.000)]                                                                                                                                              
[MSG:INFO:   Motor0]                                                                                                                                                             
[MSG:INFO:     standard_stepper Step:gpio.19 Dir:gpio.23 Disable:gpio.5]                                                                                                         
[MSG:INFO:     Neg Limit gpio.26]                                                                                                                                                
[MSG:INFO: Axis Z (0.000,600.000)]                                                                                                                                               
[MSG:INFO: Axis A (0.000,600.000)]                                                                                                                                               
[MSG:INFO: Kinematic system: Cartesian]                                                                                                                                          
[MSG:INFO: Using spindle NoSpindle]
```

### User Interface Software

bCNC

### What happened?

I don't understand what is wrong with the config.yaml.
I have checked the gpio's pins on the wiki page.
Please help...

Best regards,

Wolf

bdring, 2022-08-26:
```
[MSG:ERR: Skipping key motor0 indent 5 thisIndent 4]     
```

It does not like motor0 at indent 5. Under your Z axis you have an extra intent before motor0. This makes it look like it is under that homing section. Homing does not know what to do with that stuff. 

Remove one space before it try again.

wollo-fpv, 2022-08-26:
@bdring 
Thx......

Topic can be closed

[MSG:INFO: FluidNC v3.5.1]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
[MSG:INFO: Local filesystem type is SPIFFS]
[MSG:INFO: Configuration file:config.yaml]
[MSG:INFO: Machine 4_axis_hotwire]
[MSG:INFO: Board WeMOS D1 R32 ]
[MSG:INFO: SPI not defined]
[MSG:INFO: Stepping:RMT Pulse:2us Dsbl Delay:0us Dir Delay:1us Idle Delay:255ms]
[MSG:INFO: Axis count 4]
[MSG:INFO: Axis X (0.000,1300.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:gpio.22 Dir:gpio.21 Disable:gpio.18]
[MSG:INFO:     Neg Limit gpio.25]
[MSG:INFO: Axis Y (0.000,1300.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:gpio.19 Dir:gpio.23 Disable:gpio.5]
[MSG:INFO:     Neg Limit gpio.26]
[MSG:INFO: Axis Z (0.000,600.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:gpio.13 Dir:gpio.12 Disable:gpio.14]
[MSG:INFO:     Neg Limit gpio.15]
[MSG:INFO: Axis A (0.000,600.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:gpio.27 Dir:gpio.16 Disable:gpio.17]
[MSG:INFO:     Neg Limit gpio.33]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Using spindle NoSpindle]
[MSG:INFO: Connecting to STA SSID:SW25]

closed 2022-08-26 by bdring