bdring/FluidNC #591: Crash with app3M_spiffs1M_8MB.csv For 8Meg ESP32

Skorpi08, 2022-08-31:
### Controller Board

MKS DLC32 and MKS Tinybee

### Help From Board Vendor

- [ ] Yes
- [X] No
- [ ] Not Applicable

### Machine Description

not needed

### Configuration file

```yaml
not needed
```

### Startup Messages

```text
With install-wifi.bat

$bye
[MSG:INFO: Restarting]
ok
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:12812
load:0x40080400,len:3032
entry 0x400805e4

[MSG:INFO: FluidNC EventQueue2-Test]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
[MSG:INFO: Local filesystem type is SPIFFS]
[MSG:WARN: Cannot open configuration file:config.yaml]
[MSG:INFO: Using default configuration]
[MSG:INFO: Axes: using defaults]
[MSG:INFO: Machine Default (Test Drive)]
[MSG:INFO: Board None]
[MSG:INFO: SPI not defined]
[MSG:INFO: Stepping:RMT Pulse:4us Dsbl Delay:0us Dir Delay:0us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (-1000.000,0.000)]
[MSG:INFO: Axis Y (-1000.000,0.000)]
[MSG:INFO: Axis Z (-1000.000,0.000)]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Using spindle NoSpindle]
[MSG:INFO: STA SSID is not set]
[MSG:INFO: AP SSID FluidNC IP 192.168.0.1 mask 255.255.255.0 channel 1]
[MSG:INFO: AP started]
[MSG:INFO: WiFi on]
[MSG:INFO: Captive Portal Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]

Grbl  [FluidNC EventQueue2-Test (wifi) '$' for help]
[MSG:INFO: '$H'|'$X' to unlock]

Compiled and Uploaded with Visual Studio Code and app3M_spiffs1M_8MB.csv:

rst:0x3 (SW_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:12812
load:0x40080400,len:3032
entry 0x400805e4
ets Jun  8 2016 00:22:57

rst:0x3 (SW_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:12812
load:0x40080400,len:3032
entry 0x400805e4
ets Jun  8 2016 00:22:57
```

### User Interface Software

YAT Terminal

### What happened?

Change the Line 59 in platformio.ini for 8MB ESP32 and it crash.

here an older screenshot where it had worked.
[image]

### Other Information

changed only the line 59
[image]

after flash
[image]

Output with install-wifi.bat on Tinybee

```
-flash_freq 80m --flash_size detect 0x1000 common\bootloader_dio_80m.bin 0xe000 common\boot_app0.bin 0x10000 wifi\firmware.bin 0x8000 wifi\partitions.bin
esptool.py v3.1
Found 2 serial ports
Serial port COM6
Connecting....
Chip is ESP32-D0WD (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 40:91:51:1b:1f:4c
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Auto-detected Flash size: 8MB
Flash will be erased from 0x00001000 to 0x00005fff...
Flash will be erased from 0x0000e000 to 0x0000ffff...
Flash will be erased from 0x00010000 to 0x00150fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash params set to 0x023f
Compressed 17120 bytes to 11840...
Wrote 17120 bytes (11840 compressed) at 0x00001000 in 0.4 seconds (effective 362.0 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 47...
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.1 seconds (effective 742.7 kbit/s)...
Hash of data verified.
Compressed 1314032 bytes to 823744...
Writing at 0x000d3bea... (58 %)
```

MitchBradley, 2022-09-01:
I guess you will have to debug it because we do not support that FLASH size yet.

Skorpi08, 2022-09-01:
https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/Marlin/src/HAL/ESP32/esp32.csv

[image]

Skorpi08, 2022-09-01:
board = dpu_esp32
board_build.partitions = default_8MB.csv

[image]

board = dpu_esp32
board_build.partitions = FluidNC/ld/esp32/app3M_spiffs1M_8MB.csv  ; For 8Meg ESP32
[image]

bdring, 2022-09-01:
http://wiki.fluidnc.com/en/installation#using-esp32s-with-larger-more-memory

Skorpi08, 2022-09-01:
it doesn't work, just comment out the line because esp32dev doesn't have 8MB modules. So i switched to dpu_esp32 board. https://docs.platformio.org/en/latest/boards/espressif32/esp32dev.html#espressif-esp32-dev-module

bdring, 2022-09-01:
It helps well you use words like that.

MitchBradley, 2022-09-01:
According to .platformio/platforms/espressif32/boards/dpu_esp32.json, dpu-esp32 uses a WROVER module with PSRAM, not a WROOM module that we support.  PSRAM requires some extra GPIO pins, typically GPIO16 and GPIO17, making them unavailable for other uses.

Skorpi08, 2022-09-01:
That's right, I couldn't find any other board quickly and it worked with the dpu_esp32, so I was able to solve the problem.

closed 2022-09-01 by Skorpi08

Skorpi08, 2022-09-01:
Maybe that's the better choice:

https://github.com/platformio/platform-espressif32/blob/master/boards/heltec_wireless_stick.json

Skorpi08, 2022-09-01:
add Lines 62 + 63
then it works with board = esp32dev

[image]
[image]