bdring/FluidNC #592: Problem: Steps per mm: Reality vs Code

JemandAnders, 2022-09-03:
### Controller Board

MKS DLC32 + DRV8825 (or TMC2209)

### Help From Board Vendor

- [ ] Yes
- [X] No
- [ ] Not Applicable

### Machine Description

Standard Cartesian Plotter

### Configuration file

```yaml
board: MKS-DLC32 V2.1
name: PenPlotter
meta: (01.09.2022) by Skorpi

kinematics:
  Cartesian:

start:
  must_home: false

stepping:
  engine: I2S_STREAM
  idle_ms: 250
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0
axes:
  shared_stepper_disable_pin: I2SO.0
  x:
    steps_per_mm: 18.520
    max_rate_mm_per_min: 1000.000
    acceleration_mm_per_sec2: 25.000
    max_travel_mm: 345.000
    soft_limits: true
    homing:
      cycle: 1
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.36
      hard_limits: false
      pulloff_mm: 2.000
      stepstick:
        step_pin: I2SO.1
        direction_pin: I2SO.2

  y:
    steps_per_mm: 250.000
    max_rate_mm_per_min: 1000.000
    acceleration_mm_per_sec2: 25.000
    max_travel_mm: 40.000
    soft_limits: true
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.35
      hard_limits: false
      pulloff_mm: 2.000
      stepstick:
        step_pin: I2SO.5
        direction_pin: I2SO.6:low

  z:
    steps_per_mm: 157.750
    max_rate_mm_per_min: 12000.000
    acceleration_mm_per_sec2: 500.000
    max_travel_mm: 80.000
    soft_limits: true
    homing:
      cycle: 0
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 300.000
      seek_mm_per_min: 1000.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.34
      hard_limits: false
      pulloff_mm: 1.000
      stepstick:
        step_pin: I2SO.3
        direction_pin: I2SO.4

i2so:
  bck_pin: gpio.16
  data_pin: gpio.21
  ws_pin: gpio.17

spi:
  miso_pin: gpio.12
  mosi_pin: gpio.13
  sck_pin: gpio.14

sdcard:
  cs_pin: gpio.15
  card_detect_pin: NO_PIN

control:
  safety_door_pin: NO_PIN
  reset_pin: NO_PIN
  feed_hold_pin: NO_PIN
  cycle_start_pin: NO_PIN
  macro0_pin: NO_PIN
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN

macros:
  startup_line0:
  startup_line1:
  macro0:
  macro1:
  macro2:
  macro3:

coolant:
  flood_pin: NO_PIN
  mist_pin: NO_PIN
  delay_ms: 0

probe:
  pin: NO_PIN
  check_mode_start: true

Laser:
  pwm_hz: 5000
  #L on Beeper / IN on TTL
  output_pin: gpio.32
  enable_pin: I2SO.7
  disable_with_s0: false
  s0_with_disable: false
  tool_num: 0
  speed_map: 0=0.000% 0=12.500% 1700=100.000%
# 135=0mA 270=5mA 400=10mA 700=16mA

user_outputs:
  analog0_pin: NO_PIN
  analog1_pin: NO_PIN
  analog2_pin: NO_PIN
  analog3_pin: NO_PIN
  analog0_hz: 5000
  analog1_hz: 5000
  analog2_hz: 5000
  analog3_hz: 5000
  digital0_pin: NO_PIN
  digital1_pin: NO_PIN
  digital2_pin: NO_PIN
  digital3_pin: NO_PIN
```

### Startup Messages

```text
[MSG:INFO: FluidNC v3.4.4]
[MSG:INFO: Compiled with ESP32 SDK:v3.3.5-1-g85c43024c]
[MSG:INFO: Configuration file:MKS_DLC32_21_XYZ2.yaml]
[MSG:INFO: Machine K40 Laser]
[MSG:INFO: Board MKS-DLC32 V2.1]
[MSG:INFO: I2SO BCK:gpio.16 WS:gpio.17 DATA:gpio.21]
[MSG:INFO: SPI SCK:gpio.14 MOSI:gpio.13 MISO:gpio.12]
[MSG:INFO: SD Card cs_pin:gpio.15 detect:NO_PIN]
[MSG:INFO: Stepping:I2S_stream Pulse:4us Dsbl Delay:0us Dir Delay:1us Idle Delay:250ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Shared stepper disable I2SO.0]
[MSG:INFO: Axis X (0.000,345.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.1 Dir:I2SO.2 Disable:NO_PIN]
[MSG:INFO:     Neg Limit gpio.36]
[MSG:INFO: Axis Y (0.000,40.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.5 Dir:I2SO.6:low Disable:NO_PIN]
[MSG:INFO:     Neg Limit gpio.35]
[MSG:INFO: Axis Z (0.000,80.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.3 Dir:I2SO.4 Disable:NO_PIN]
[MSG:INFO:     Neg Limit gpio.34]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Laser Spindle Ena:I2SO.7 Out:gpio.32 Freq:5000Hz Res:13bits Laser mode:On]
[MSG:INFO: Using spindle Laser]
[MSG:INFO: Connecting to STA SSID:HQ]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connected - IP is 192.168.42.28]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
```

### User Interface Software

WebUI

### What happened?

I'm trying to calibrate my Plotter, but having troubles with the steps per mm.
You can ignore everything in the config file exept the config for the y-axis. The config is still WIP, but already working (wrong) :-) 

My Plotter has a resolution of 250 Steps per mm. (1.8 deg Stepper with a 5x0.8 mm axis, full step)
When I use these values in the config file, and moving the axis over the WebUI 10mm, the corresponding axis is moving more then 10 mm. Roughly 13 mm.
I tested this behavior with a DRV8825 and a TMC2209 Driver. I also tested my calculations with a simple arduino Code. My calculated steps are correct if I use an arduino and the same driver. 200 pulses = 1 Revolution = 0,8 mm distance / 20 revolutions = 16 mm distance = what I'm measuring. 
What am I missing? Is there a way to see the calculated steps for a specific movement? 
Tested Versions 3.4.4 and 3.5.1

Kind regards,
Joachim

Skorpi08, 2022-09-03:
Ich hab am 01.09.22 ganz sicher keine Config veröffentlicht!
Wenn schon, denn nimm meine "offizielle" Config und trag da deine Steps ein.
Aber hey, wenn du DER Winterscheidt bist, dann verzeih ich das ;)

JemandAnders, 2022-09-03:
Sorry. :-)
Dein Name sollte da auch nicht mehr stehen. Ist korrigiert.

Skorpi08, 2022-09-03:
wie erwähnt, nimm meine config und ändere nur die Steps.
oder hats ein wichtigen Grund warum du I2S_STREAM nimmst?

MitchBradley, 2022-09-03:
Try I2S_STATIC and see if that makes a difference

JemandAnders, 2022-09-06:
Hi,
With I2S_STATIC all movements are correct.
I used I2S_STREAM because the Wiki says "I2S_STREAM is generally best for most users unless dynamically-speed-adjusted "spindles" are used."

MitchBradley, 2022-09-06:
It is possible that signal quality of the I2S wiring on the MKS DLC32 board is not as good as on the 6-pack board, where we checked that aspect of the design carefully.  If so, the faster timing of I2S_STREAM might be causing false edges on the step pulses.  But I am just guessing about that.

RootCNC, 2022-09-18:
I've noticed the same issue when it comes to calibrating a rotation axis and the mapping between degrees to `steps_per_mm`. isn't as simple as a 1:# mapping that you'll normally see with lead screws etc.

I noticed that everything is correct with `I2S_STREAM` providing there is no factional part in the steps_per_mm, IE the unit is 400.00. As soon as you add a faction into the `steps_per_mm` I got positional error within the rotational axis and its not a small error at all, we're talking a huge error. 

I have no real understanding of the inner workings but I thought I would share my observation and it might provoke a "a-ha moment". as changing the engine to I2S_STATIC resolved all my issues.

MitchBradley, 2022-09-18:
> I noticed that everything is correct with I2S_STREAM providing there is no factional part in the steps_per_mm, IE the unit is 400.00. As soon as you add a faction into the steps_per_mm I got positional error within the rotational axis and its not a small error at all, we're talking a huge error.

I am trying to duplicate this behavior.  With microstepping 8, I set steps_per_mm to 8, move by 200 mm, and get exactly one revolution of a 200 steps/rev motor.  Set it to 9 and I get 9/8 of a rev.  8.5 gives halfway between the two, as expected.

BUT - I am using the serial terminal to set $/axes/x/steps_per_mm=WHATEVER , and the very first move after I make the change is wildly wrong, while subsequent ones are correct.  That is easily explained as follows: The internal machine position is stored in steps.  When you do a move, it computes the distance in mm as (new_pos_mm - old_pos_mm).  But old_pos_mm has to be computed as old_pos_steps / steps_per_mm, and since you just changed steps_per_mm, old_pos_mm is wrong for the first move.

You can observe the position change as follows:
```
$10=1
ok
?
<Idle|MPos:3200.000,0.000,0.000|FS:0,0>
$/axes/x/steps_per_mm
$/axes/x/steps_per_mm=8.000
ok
$/axes/x/steps_per_mm=8.5
ok
?
<Idle|MPos:3011.765,0.000,0.000|FS:0,0>
```
If MPos happens to be 0, the change will not show up, obviously.

I don't know if this explains what you were seeing.

I will keep trying to find a repro for the problem.

RootCNC, 2022-09-18:
Interesting. I was uploading a new configuration each time and resetting the controller between uploads through Fuildterm (I didn't realise you could amend those parameters through the terminal).

In my instance the machine position through the query `?` would report the correct distance but the actual movement of the stepper was incorrect. I did go back a froth in my testing, IE `G0 A0` to `G0 A360` back to `G0 A0' and both forward and reverse turns would rotate in the same distance. IE it would start in one position - do a complete turn and then roughly 45 degrees more (I didn't actually measure this, it was obviously incorrect) and then return to the original starting position. Could this indicate that the issue (or mis-understanding) is persistent between moves?

MitchBradley, 2022-09-18:
There are some parameters that you can change on the fly and many that you cannot.  In particular, pin assignments cannot be changed.  On the fly changes are not persistent unless you dump the changed configuration back to the file with $cd=config.yaml - but I personally do not like to make persistent changes that way.  I prefer to edit the file on the host, so I have a copy, and re-download it.

So far I have been doing the tests with the X axis to see if it is some deep problem.  I will try with A.

MitchBradley, 2022-09-18:
I have had no success duplicating the problem with non-integer steps_per_mm using I2S_STREAM on a 6 pack.  I think I would need exact details of your setup with all parameters.

shnads, 2022-10-07:
I am using a MKS Tinybee board and I am experiencing the same behavior, if I ask my Y axis to move 10mm it moves 13mm. Interestingly my X axis (using the same transmission system and steps_per_mm) is fine. I am using I2S_STREAM, I will change to STATIC and report if that solves the problem. I have confirmed all the stepper driver DIP settings are the same and the DIP settings on the Tinybee are also the same for all axis. I was using an Arduino to drive the CNC previously with the same steps_per_mm and it was accurate. I am using external drivers.

Please do let me know if I can assist in tracking down the issue!

MitchBradley, 2022-10-07:
Looking forward to your I2S_STREAM results.

shnads, 2022-10-07:
@MitchBradley I2S_STATIC fixed the issue, both X and Y are now moving as expected.

I am using dual motors for Y, with one motor in the Y socket and one in the E0 socket. I observed both motors moving the incorrect distance, while the single motor for X was moving correctly.

MitchBradley, 2022-10-07:
The only hypothesis I have right now is that perhaps the Tinybee has signal quality issues that prevent I2S_STREAM mode from working.  STREAM is faster than STATIC so maybe it is more susceptible to signal ringing or setup times or something.  I don't have a Tinybee so I cannot analyze it more deeply.  Someone said they ordered a Tinybee to be shipped to me, but I never received it.

shnads, 2022-10-07:
Is there anything I can test to help? Should I move my X axis that works to a Y axis socket to see if it exhibits the same behavior?

I would much prefer to be using a 6 pack controller, but they do not ship to Australia so I'm stuck with the tinybee.

MitchBradley, 2022-10-07:
Debugging by remote control is extremely time consuming and has a low success rate.  Can you just use static and be done with it?  the advantages of stream are minimal to nonexistent in most cases.

shnads, 2022-10-07:
Oh yes I'm completely fine with that, just trying to be helpful because I appreciate how awesome FluidNC is.

I'll send a donation instead :)

MitchBradley, 2022-10-10:
Since the OP has the laser enable on an I2SO pin, the original problem is very likely to be a manifestation of #287 which we just fixed in v3.6.3.  Everybody who has responded to this ticket with "me too", please try 3.6.3 and report if it fixes things.

closed 2024-02-03 by bdring