bdring/FluidNC #597: Problem:  “Driver test failed check motor power” on new install

mskcnc, 2022-09-08:
### Controller Board

6 pack controller 

### Help From Board Vendor

- [ ] Yes
- [ ] No
- [X] Not Applicable

### Machine Description

input cards in spaces 1&2 and a 5v card in 3 and TMC2130s. Dual XY Laser. Originally using  GRBL ESP32 now trying to update to Fluidnc

### Configuration file

```yaml
name: 6 Pack MPCNC XYZXY V2 (StepStick)
board: 6-pack

kinematics:
  Cartesian:

stepping:
  engine: I2S_STATIC
  idle_ms: 255
  dir_delay_us: 0
  pulse_us: 4
  disable_delay_us: 0

axes:
  x:
    steps_per_mm: 100.000
    max_rate_mm_per_min: 2000.000
    acceleration_mm_per_sec2: 200.000
    max_travel_mm: 600.000
    soft_limits: false
    homing:
      cycle: 2
      mpos_mm: 0.000
      positive_direction: true
      settle_ms: 250.000
      seek_mm_per_min: 1000.000
      feed_mm_per_min: 50.000
      seek_scaler: 1.100
      feed_scaler: 5.000
    motor0:
      limit_neg_pin: gpio.33:low
      limit_all_pin: no_pin
      hard_limits: false
      pulloff_mm: 05.000
      tmc_2130:
        direction_pin: i2so.1
        step_pin: i2so.2
        disable_pin: i2so.0
        run_amps: 1.000
        hold_amps: 0.500
        microsteps: 16
        stallguard: 16
        stallguard_debug: false
        run_mode: StealthChop
        homing_mode: StealthChop
        use_enable: false
        r_sense_ohms: 0.110
        cs_pin: i2so.3:low
    motor1:
      limit_neg_pin: gpio.34:low
      hard_limits: false
      pulloff_mm: 05.000
      tmc_2130:
        direction_pin: i2so.12
        step_pin: i2so.13
        disable_pin: i2so.15
        run_amps: 1.000
        hold_amps: 0.500
        microsteps: 16
        stallguard: 16
        stallguard_debug: false
        run_mode: StealthChop
        homing_mode: StealthChop
        use_enable: false
        r_sense_ohms: 0.110
        cs_pin: i2so.14:low
  y:
    steps_per_mm: 100.000
    max_rate_mm_per_min: 2000.000
    acceleration_mm_per_sec2: 200.000
    max_travel_mm: 600.000
    soft_limits: false
    homing:
      cycle: 2
      mpos_mm: 0.000
      positive_direction: true
      settle_ms: 250.000
      seek_mm_per_min: 1000.000
      feed_mm_per_min: 500.000
      seek_scaler: 1.100
      feed_scaler: 5.000
    motor0:
      limit_neg_pin: gpio.32:low
      hard_limits: false
      pulloff_mm: 05.000
      tmc_2130:
        direction_pin: i2so.4
        step_pin: i2so.5
        disable_pin: i2so.7
        run_amps: 1.000
        hold_amps: 0.500
        microsteps: 16
        stallguard: 16
        stallguard_debug: false
        run_mode: StealthChop
        homing_mode: StealthChop
        use_enable: false
        r_sense_ohms: 0.110
        cs_pin: i2so.6:low
    motor1:
      limit_neg_pin: gpio.2:low
      hard_limits: false
      pulloff_mm: 05.000
      tmc_2130:
        direction_pin: i2so.17
        step_pin: i2so.18
        disable_pin: i2so.16
        run_amps: 1.000
        hold_amps: 0.500
        microsteps: 16
        stallguard: 16
        stallguard_debug: false
        run_mode: StealthChop
        homing_mode: StealthChop
        use_enable: false
        r_sense_ohms: 0.110
        cs_pin: i2so.19:low
  z:
    steps_per_mm: 100.000
    max_rate_mm_per_min: 1000.000
    acceleration_mm_per_sec2: 200.000
    max_travel_mm: 100.000
    soft_limits: false
    homing:
      cycle: 1
      mpos_mm: 0.000
      positive_direction: false
      settle_ms: 250.000
      seek_mm_per_min: 2000.000
      feed_mm_per_min: 200.000
      seek_scaler: 1.100
      feed_scaler: 5.000
    motor0:
      limit_neg_pin: no_pin
      hard_limits: false
      tmc_2130:
        direction_pin: i2so.9
        step_pin: i2so.10
        disable_pin: i2so.8
        run_amps: 1.000
        hold_amps: 0.500
        microsteps: 16
        stallguard: 16
        stallguard_debug: false
        run_mode: StealthChop
        homing_mode: StealthChop
        use_enable: false
        r_sense_ohms: 0.110
        cs_pin: i2so.11:low
    motor1:
      null_motor:

i2so:
  bck_pin: gpio.22
  data_pin: gpio.21
  ws_pin: gpio.17

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  cs_pin: gpio.5

control:

coolant:
  delay_ms: 1000.000

probe:
  check_mode_start: false

macros:
  startup_line0: 
  startup_line1: 
  macro0: 
  macro1: 
  macro2: 
  macro3: 

start:
  must_home: false
  check_limits: true
  deactivate_parking: false

user_outputs:

laser:
  tool_num: 10
  speed_map: 0=0.0% 255=100.0%
  output_pin: gpio.26
  disable_with_s0: false
  s0_with_disable: true
  pwm_hz: 1000
arc_tolerance_mm: 0.002
junction_deviation_mm: 0.010
verbose_errors: false
report_inches: false
enable_parking_override_control: false
use_line_numbers: false
```

### Startup Messages

```text
FluidNC v1.2.0 using COM3
Exit: Ctrl-C, Ctrl-Q or Ctrl-], Clear screen: CTRL-W
Upload: Ctrl-U, Reset ESP32: Ctrl-R, Send Override: Ctrl-O
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:12812
load:0x40080400,len:3032
entry 0x400805e4

[MSG:INFO: FluidNC v3.5.1]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
[MSG:INFO: Local filesystem type is SPIFFS]
[MSG:INFO: Configuration file:config.yaml]
[MSG:INFO: Machine 6 Pack MPCNC XYZXY V2 (StepStick)]
[MSG:INFO: Board 6-pack]
[MSG:INFO: I2SO BCK:gpio.22 WS:gpio.17 DATA:gpio.21]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN]
[MSG:INFO: Stepping:I2S_static Pulse:4us Dsbl Delay:0us Dir Delay:0us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (-600.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_2130 Step:I2SO.2 Dir:I2SO.1 CS:I2SO.3:low Disable:I2SO.0 Index:-1 R:0.110]
[MSG:INFO:     Neg Limit gpio.33:low]
[MSG:INFO:   Motor1]
[MSG:INFO:     tmc_2130 Step:I2SO.13 Dir:I2SO.12 CS:I2SO.14:low Disable:I2SO.15 Index:-1 R:0.110]
[MSG:INFO:     Neg Limit gpio.34:low]
[MSG:INFO: Axis Y (-600.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_2130 Step:I2SO.5 Dir:I2SO.4 CS:I2SO.6:low Disable:I2SO.7 Index:-1 R:0.110]
[MSG:INFO:     Neg Limit gpio.32:low]
[MSG:INFO:   Motor1]
[MSG:INFO:     tmc_2130 Step:I2SO.18 Dir:I2SO.17 CS:I2SO.19:low Disable:I2SO.16 Index:-1 R:0.110]
[MSG:INFO:     Neg Limit gpio.2:low]
[MSG:INFO: Axis Z (0.000,100.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_2130 Step:I2SO.10 Dir:I2SO.9 CS:I2SO.11:low Disable:I2SO.8 Index:-1 R:0.110]
[MSG:INFO:   Motor1]
[MSG:ERR: X Axis driver test failed. Check motor power]
[MSG:ERR: X2 Axis driver test failed. Check motor power]
[MSG:ERR: Y Axis driver test failed. Check motor power]
[MSG:ERR: Y2 Axis driver test failed. Check motor power]
[MSG:ERR: Z Axis driver test failed. Check motor power]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Laser Ena:NO_PIN Out:gpio.26 Freq:1000Hz Period:65535]
[MSG:INFO: Using spindle Laser]
[MSG:INFO: STA SSID is not set]
[MSG:INFO: AP SSID FluidNC IP 192.168.0.1 mask 255.255.255.0 channel 1]
[MSG:INFO: AP started]
[MSG:INFO: WiFi on]
[MSG:INFO: Captive Portal Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]

Grbl 3.5 [FluidNC v3.5.1 (wifi) '$' for help]
```

### What happened?

motors not recognized.  
[MSG:ERR: X Axis driver test failed. Check motor power] all motors.
I can install my old ESP32 and it works so it doesn’t appear to be a hardware issue. I used the “magic robot” to transfer my old setting. 
Any help is greatly appreciated!

MitchBradley, 2022-09-08:
Version 3.5.1 is bad.  Use version 3.6.0 or revert to 3.4.4

mskcnc, 2022-09-08:
Thanks Mitch! But still the same after going to 3.6.0.

MitchBradley, 2022-09-08:
Try adding spi_index: N to the tmc2130: sections, with N being 0, 1, 2, .. in left to right order of the slots on the 6 pack.  So for example
```
axes:
  x:
     ...
     motor0:
      ...
      tmc_2130:
        spi_index: 0
```
Index 1 would be for y motor0 since that seems to be in the second driver slot, etc.

MitchBradley, 2022-09-08:
Please note that I updated the comment above to correct the spi_index values.  They should be 1,2,3,... not 0,1,2,...

mskcnc, 2022-09-18:
Thanks for your help! While looking up SPI info for my 6 Pack controller I
found some configuration  examples. One was a pretty close fit and after a
few modifications it's working beautifully. Thanks again

closed 2022-09-18 by MitchBradley