bdring/FluidNC #598: reboot when execute job

BRUN-Richard, 2022-09-08:
### Controller Board

my own board built around ESP32-WROOM-32D

### Help From Board Vendor

- [ ] Yes
- [ ] No
- [ ] Not Applicable

### Machine Description

modified K40 laser (addition of a Z axis)
personal development control board around an ESP32-WROOM-32D module
3 external drivers based on TB6600
spi micro sd card reader (aliexpress built-in level conversion circuit, interface level can be 5V or 3.3V)
optocoupled mechanical fdc

### Configuration file

```yaml
name: "ESP32 Dev Controller V4"
board: "K40"

stepping:
  engine: RMT
  idle_ms: 255
  dir_delay_us: 1
  pulse_us: 8
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: gpio.13:low
  
  x:
    steps_per_mm: 160
    max_rate_mm_per_min: 3000
    acceleration_mm_per_sec2: 2000
    max_travel_mm: 297
    homing:
      cycle: 2
      allow_single_axis: true
      mpos_mm: 5
      positive_direction: false
      seek_mm_per_min: 1000
      feed_mm_per_min: 200
      
    motor0:
      limit_neg_pin: gpio.17:low:pu
      hard_limits: true
      stepstick:
        direction_pin: gpio.26
        step_pin: gpio.12
    motor1:
      null_motor:

  y:
    steps_per_mm: 160
    max_rate_mm_per_min: 3000
    acceleration_mm_per_sec2: 2000
    max_travel_mm: 210
    homing:
      cycle: 2
      allow_single_axis: true
      mpos_mm: 5
      positive_direction: false
      seek_mm_per_min: 1000
      feed_mm_per_min: 200

    motor0:
      limit_neg_pin: gpio.4:low:pu
      hard_limits: true
      stepstick:
        direction_pin: gpio.25
        step_pin: gpio.14
    motor1:
      null_motor:

  z:
    steps_per_mm: 800
    max_rate_mm_per_min: 200
    acceleration_mm_per_sec2: 100
    max_travel_mm: 80
    homing:
      cycle: 1
      allow_single_axis: true
      mpos_mm: 10
      positive_direction: true
      seek_mm_per_min: 200
      feed_mm_per_min: 50
     

    motor0:
      limit_neg_pin: gpio.15:low:pu
      hard_limits: true
      stepstick:
        direction_pin: gpio.33
        step_pin: gpio.27
    motor1:
      null_motor:

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  cs_pin: gpio.5
  card_detect_pin: NO_PIN

coolant:
  flood_pin: gpio.16:high
  mist_pin:  NO_PIN

        
probe:
  pin: gpio.32:low:pu

laser:
  pwm_hz: 20000
  output_pin: gpio.2
  enable_pin: NO_PIN
  disable_with_s0: false
  s0_with_disable: true
  tool_num: 0
  speed_map: 0=0.000% 1000=100.000%
```

### Startup Messages

```text
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:12812
load:0x40080400,len:3032
entry 0x400805e4

[MSG:INFO: FluidNC v3.6.0]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
[MSG:INFO: Local filesystem type is SPIFFS]
[MSG:INFO: Configuration file:config.yaml]
[MSG:INFO: Machine ESP32 Dev Controller V4]
[MSG:INFO: Board K40]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN]
[MSG:INFO: Stepping:RMT Pulse:8us Dsbl Delay:0us Dir Delay:1us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Shared stepper disable gpio.13:low]
[MSG:INFO: Axis X (5.000,302.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:gpio.12 Dir:gpio.26 Disable:NO_PIN]
[MSG:INFO:     Neg Limit gpio.17:low:pu]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Y (5.000,215.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:gpio.14 Dir:gpio.25 Disable:NO_PIN]
[MSG:INFO:     Neg Limit gpio.4:low:pu]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Z (-70.000,10.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:gpio.27 Dir:gpio.33 Disable:NO_PIN]
[MSG:INFO:     Neg Limit gpio.15:low:pu]
[MSG:INFO:   Motor1]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Laser Ena:NO_PIN Out:gpio.2 Freq:20000Hz Period:2047]
[MSG:INFO: Using spindle Laser]
[MSG:INFO: Flood coolant gpio.16]
[MSG:INFO: Probe Pin: gpio.32:low:pu]
[MSG:INFO: STA SSID is not set]
[MSG:INFO: AP SSID FluidNC IP 192.168.0.1 mask 255.255.255.0 channel 1]
[MSG:INFO: AP started]
[MSG:INFO: WiFi on]
[MSG:INFO: Captive Portal Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]

Grbl 3.6 [FluidNC v3.6.0 (wifi) '$' for help]
[MSG:INFO: '$H'|'$X' to unlock]
```

### User Interface Software

WebUI and LaserGRBL

### What happened?

when I launch a job (gcode file) fluidNC crashes (then reboot) after a few gcodes executed (variable number, I was even able to execute a complete program a few times)
the program is hard the SD card and the command sends
$SD/run=filexx.nc
from the webUI interface (systematic crash)
from a command line via usb (crash too but not systematic)

### Other Information

log console after crash (fluidNC reboot in minimal mode (test))

Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed). 

Core  1 register dump:
PC      : 0x4010c430  PS      : 0x00060035  A0      : 0x80089b50  A1      : 0x3ffbf19c  
A2      : 0x3ffb4778  A3      : 0x00000000  A4      : 0x3ffcbc74  A5      : 0x3ffbdbe8  
A6      : 0x00000003  A7      : 0x00060523  A8      : 0x80081e45  A9      : 0x00000002  
A10     : 0x0000001a  A11     : 0x00000000  A12     : 0x3ffb1c70  A13     : 0x3ffb1c30  
A14     : 0x3ffc043c  A15     : 0x3ffb2f20  SAR     : 0x00000000  EXCCAUSE: 0x00000007  
EXCVADDR: 0x00000000  LBEG    : 0x4008ad64  LEND    : 0x4008ad7a  LCOUNT  : 0x00000000  

Backtrace:0x4010c42d:0x3ffbf19c |<-CORRUPTED

ELF file SHA256: 0000000000000000

Rebooting...
ets Jun  8 2016 00:22:57

rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:12812
load:0x40080400,len:3032
entry 0x400805e4

[MSG:INFO: FluidNC v3.6.0]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
[MSG:INFO: Local filesystem type is SPIFFS]
[MSG:ERR: Skipping configuration file due to panic]
[MSG:INFO: Using default configuration]
[MSG:INFO: Axes: using defaults]
[MSG:INFO: Machine Default (Test Drive)]
[MSG:INFO: Board None]
[MSG:INFO: SPI not defined]
[MSG:INFO: Stepping:RMT Pulse:4us Dsbl Delay:0us Dir Delay:0us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (-1000.000,0.000)]
[MSG:INFO: Axis Y (-1000.000,0.000)]
[MSG:INFO: Axis Z (-1000.000,0.000)]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Using spindle NoSpindle]
[MSG:INFO: STA SSID is not set]
[MSG:INFO: AP SSID FluidNC IP 192.168.0.1 mask 255.255.255.0 channel 1]
[MSG:INFO: AP started]
[MSG:INFO: WiFi on]
[MSG:INFO: Captive Portal Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]

Grbl 3.6 [FluidNC v3.6.0 (wifi) '$' for help]
[MSG:INFO: '$H'|'$X' to unlock]

BRUN-Richard, 2022-09-08:
in what happened ? you should read "the program is on the SD card"  sorry!

Avataar120, 2022-09-08:
Hi,
I have reproduced your defects only in WIFI AP mode yesterday.
Can you try in WIFI STA mode ?
WIFI AP mode is not recommended

BRUN-Richard, 2022-09-08:
Hi 

no I have not tried in STA mode
but the fault also occurred when I run the command via the USB port
I will try in STA mode to see (I have to look at how to do it)

Avataar120, 2022-09-08:
Same for me, it occurs as well with USB connection but ... AP is active
To go to STA mode, you just have to send : 
$sta/SSID=YourWIFIName
$sta/Password=YourWIFIPassword (min 8 characters)
Then reboot your shield

you should see in the startup message something like : 
[MSG:INFO: Connecting to STA SSID:VotreNomDeReseauWifi]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connected - IP is 192.168.xxx.yyy]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]

BRUN-Richard, 2022-09-08:
hi

I connected in STA
time to fix my endstop Z (homing incident) and I'll tell you the result

BRUN-Richard, 2022-09-08:
hi 
hey
YES, it seems to work! :-)
I continue my tests and if everything is OK, we can close this issue
Thank you very much for your help, I was a hundred miles from thinking about that.
congratulations for your work!

Avataar120, 2022-09-08:
Hope it will completely fix your issue.
You can congrats @MitchBradley & @bdring and other developers for their wonderful job on this firmware !

BRUN-Richard, 2022-09-09:
I executed several times a file of 19000 lines of Gcode without any incident
I think the problem is solved
once again thank you very much for your help
:-)

closed 2022-09-09 by BRUN-Richard