bdring/FluidNC #602: Problem: Limit pin are lost after $cd=file.yaml command

Avataar120, 2022-09-10:
### Controller Board

Personal design

### Help From Board Vendor

- [ ] Yes
- [ ] No
- [X] Not Applicable

### Machine Description

4 steppers - Engine I2S_STATIC

### Configuration file

```yaml
name: K40_AvaShield
board: 4_Ava_Shield_7.x_XYZ

kinematics:
  Cartesian:

stepping:
  engine: I2S_STATIC
  idle_ms: 255
  dir_delay_us: 1
  pulse_us: 4
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: NO_PIN

  x:
    steps_per_mm: 78.74
    max_rate_mm_per_min: 20000.000
    acceleration_mm_per_sec2: 2000.000
    max_travel_mm: 310.000
    soft_limits: true
    homing:
      cycle: 1
      positive_direction: false
      mpos_mm: 5.000
      feed_mm_per_min: 200.000
      seek_mm_per_min: 2500.000
      settle_ms: 250.000
      seek_scaler: 1.500
      feed_scaler: 5.000

    motor0:
      limit_neg_pin: gpio.39
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 5.000
      stepstick:
        ms3_pin: I2SO.3
        step_pin: I2SO.2
        direction_pin: I2SO.1
        disable_pin: I2SO.0

  y:
    steps_per_mm: 78.74
    max_rate_mm_per_min: 20000.000
    acceleration_mm_per_sec2: 2000.000
    max_travel_mm: 210.000
    soft_limits: true
    homing:
      cycle: 1
      positive_direction: false
      mpos_mm: 5.000
      feed_mm_per_min: 200.000
      seek_mm_per_min: 2500.000
      settle_ms: 250.000
      seek_scaler: 1.500
      feed_scaler: 5.000

    motor0:
      limit_neg_pin: gpio.34
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 5.000
      stepstick:
        ms3_pin: I2SO.7
        step_pin: I2SO.6
        direction_pin: I2SO.5:low
        disable_pin: I2SO.4

  z:
    steps_per_mm: 1000.000
    max_rate_mm_per_min: 200.000
    acceleration_mm_per_sec2: 1000.000
    max_travel_mm: 20.000
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: true
      mpos_mm: 9.500
      feed_mm_per_min: 100.000
      seek_mm_per_min: 100.000
      settle_ms: 250.000
      seek_scaler: 1.500
      feed_scaler: 5.000

    motor0:
      limit_neg_pin: gpio.35:low
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 9.500
      stepstick:
        ms3_pin: I2SO.11
        step_pin: I2SO.10
        direction_pin: I2SO.9:low
        disable_pin: I2SO.8
    motor1:
      null_motor:

i2so:
  bck_pin: gpio.22
  data_pin: gpio.21
  ws_pin: gpio.17

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  cs_pin: gpio.5

control:
  safety_door_pin: gpio.27:low:pu
  reset_pin: gpio.12:low:pu
  feed_hold_pin: gpio.14:low:pu
  cycle_start_pin: gpio.13:low:pu

coolant:
  mist_pin: gpio.4
  delay_ms: 50.000

probe:
  pin: gpio.36:low
  check_mode_start: false

macros:
  startup_line0: 
  startup_line1: 
  macro0: 
  macro1: 
  macro2: 
  macro3: 

start:
  must_home: true
  check_limits: true
  deactivate_parking: false

user_outputs:

laser:
  tool_num: 0
  speed_map: 0=0.0% 1000=100.0%
  output_pin: gpio.15
  enable_pin: gpio.16:low
  disable_with_s0: false
  s0_with_disable: false
  pwm_hz: 15000

arc_tolerance_mm: 0.002
junction_deviation_mm: 0.010
verbose_errors: false
report_inches: false
enable_parking_override_control: false
use_line_numbers: false
```

### Startup Messages

```text
With Devt branch : 
Grbl  [FluidNC TestPullups (Devt-4e3ef6bd) (wifi) '$' for help]
[MSG:INFO: '$H'|'$X' to unlock]
[MSG:INFO: FluidNC TestPullups (Devt-4e3ef6bd)]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
[MSG:INFO: Local filesystem type is SPIFFS]
[MSG:INFO: Configuration file:config.yaml]
[MSG:INFO: Machine K40_AvaShield]
[MSG:INFO: Board 4_Ava_Shield_7.x_XYZ]
[MSG:INFO: I2SO BCK:gpio.22 WS:gpio.17 DATA:gpio.21]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN]
[MSG:INFO: Stepping:I2S_static Pulse:4us Dsbl Delay:0us Dir Delay:1us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (5.000,315.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.2 Dir:I2SO.1 Disable:I2SO.0]
[MSG:INFO:     Neg Limit gpio.39]
[MSG:INFO: Axis Y (5.000,215.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.6 Dir:I2SO.5:low Disable:I2SO.4]
[MSG:INFO:     Neg Limit gpio.34]
[MSG:INFO: Axis Z (-10.500,9.500)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.10 Dir:I2SO.9:low Disable:I2SO.8]
[MSG:INFO:     Neg Limit gpio.35:low]
[MSG:INFO:   Motor1]
[MSG:INFO: safety_door_pin gpio.27:low:pu]
[MSG:INFO: reset_pin gpio.12:low:pu]
[MSG:INFO: feed_hold_pin gpio.14:low:pu]
[MSG:INFO: cycle_start_pin gpio.13:low:pu]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Laser Ena:gpio.16:low Out:gpio.15 Freq:15000Hz Period:4095]
[MSG:INFO: Using spindle Laser]
[MSG:INFO: Mist coolant gpio.4]
[MSG:INFO: Probe Pin: gpio.36:low]
[MSG:INFO: STA SSID is not set]
[MSG:INFO: AP SSID FluidNC IP 192.168.0.1 mask 255.255.255.0 channel 1]
[MSG:INFO: AP started]
[MSG:INFO: WiFi on]
[MSG:INFO: Captive Portal Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]

With 3.6.0 branch : 
[MSG:INFO: FluidNC v3.6.0]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
[MSG:INFO: Local filesystem type is SPIFFS]
[MSG:INFO: Configuration file:config.yaml]
[MSG:INFO: Machine K40_AvaShield]
[MSG:INFO: Board 4_Ava_Shield_7.x_XYZ]
[MSG:INFO: I2SO BCK:gpio.22 WS:gpio.17 DATA:gpio.21]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN]
[MSG:INFO: Stepping:I2S_static Pulse:4us Dsbl Delay:0us Dir Delay:1us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (5.000,315.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.2 Dir:I2SO.1 Disable:I2SO.0]
[MSG:INFO:     Neg Limit gpio.39]
[MSG:INFO: Axis Y (5.000,215.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.6 Dir:I2SO.5:low Disable:I2SO.4]
[MSG:INFO:     Neg Limit gpio.34]
[MSG:INFO: Axis Z (-10.500,9.500)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.10 Dir:I2SO.9:low Disable:I2SO.8]
[MSG:INFO:     Neg Limit gpio.35:low]
[MSG:INFO:   Motor1]
[MSG:INFO: safety_door_pin gpio.27:low:pu]
[MSG:INFO: reset_pin gpio.12:low:pu]
[MSG:INFO: feed_hold_pin gpio.14:low:pu]
[MSG:INFO: cycle_start_pin gpio.13:low:pu]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Laser Ena:gpio.16:low Out:gpio.15 Freq:15000Hz Period:4095]
[MSG:INFO: Using spindle Laser]
[MSG:INFO: Mist coolant gpio.4]
[MSG:INFO: Probe Pin: gpio.36:low]
[MSG:INFO: STA SSID is not set]
[MSG:INFO: AP SSID FluidNC IP 192.168.0.1 mask 255.255.255.0 channel 1]
[MSG:INFO: AP started]
[MSG:INFO: WiFi on]
[MSG:INFO: Captive Portal Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
```

### User Interface Software

Fluidterm

### What happened?

I describe limit pin as 
limit_neg_pin: gpio.39 for X axis
limit_neg_pin: gpio.34 for Y axis
limit_neg_pin: gpio.35:low for Z axis

when issuing $cd command, here the result (on 3.60 & Devt branch)

  x:
    steps_per_mm: 78.740
    max_rate_mm_per_min: 20000.000
    acceleration_mm_per_sec2: 2000.000
    max_travel_mm: 310.000
    soft_limits: true
    homing:
      cycle: 1
      allow_single_axis: true
      positive_direction: false
      mpos_mm: 5.000
      feed_mm_per_min: 200.000
      seek_mm_per_min: 2500.000
      settle_ms: 250
      seek_scaler: 1.500
      feed_scaler: 5.000

    motor0:
      limit_neg_pin: **NO_PIN**
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN

....

  y:
    steps_per_mm: 78.740
    max_rate_mm_per_min: 20000.000
    acceleration_mm_per_sec2: 2000.000
    max_travel_mm: 210.000
    soft_limits: true
    homing:
      cycle: 1
      allow_single_axis: true
      positive_direction: false
      mpos_mm: 5.000
      feed_mm_per_min: 200.000
      seek_mm_per_min: 2500.000
      settle_ms: 250
      seek_scaler: 1.500
      feed_scaler: 5.000

    motor0:
      limit_neg_pin: **NO_PIN**
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN

...

  z:
    steps_per_mm: 1000.000
    max_rate_mm_per_min: 200.000
    acceleration_mm_per_sec2: 1000.000
    max_travel_mm: 20.000
    soft_limits: false
    homing:
      cycle: 2
      allow_single_axis: true
      positive_direction: true
      mpos_mm: 9.500
      feed_mm_per_min: 100.000
      seek_mm_per_min: 100.000
      settle_ms: 250
      seek_scaler: 1.500
      feed_scaler: 5.000

    motor0:
      limit_neg_pin: **NO_PIN**
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN

However, no error message during boot and limit pins are even well detected during boot (see boot message)

Then ...
 if I change on the fly some settings, 
then I save them with $cd=config.yaml, 

**limit pins are lost**

Was not the case on previous versions

Avataar120, 2022-09-11:
Hi,

Even if it was easily reproductible, I confirm that this commit is fixing the issue on my tests
Thanks Mitch,

Avataar120, 2022-09-11:
Arg, side effect ...
When asking $H, X & Y axis are going to home position : OK
Pull back 5mm for both axis : OK
Looking for both end stops, X is found, but Y is not -> Y axis is crashing into the machine

I reopen the issue, but if you prefer, I can open another one

MitchBradley, 2022-09-16:
Please supply complete schematics for your limit pin circuitry.  Many recent problems have been traced to electrical issues with such circuitry.

Avataar120, 2022-09-16:
Hi, please find it below : 

[image]
[image]

MitchBradley, 2022-09-16:
And the switches are just ordinary switches connected to GND?  Or something else?

Avataar120, 2022-09-16:
I have both setup.
One setup with ordinary switches on one machine (wired on normally closed pins)
One other machine with optical switches (+5V when triggered, 0V when head not in home position)

Avataar120, 2022-09-17:
Please find more information here :

With 3.4.x version, when I home (with debug info), I get those messages : 
MSG:DBG: Homing XY]
[MSG:DBG: PrePulloff]
[MSG:DBG: Fast approach]
[MSG:DBG: X target -465.000 rate 2500.000]
[MSG:DBG: Y target -315.000 rate 2500.000]
[MSG:DBG: Pulloff0]
[MSG:DBG: X target 5.000 rate 200.000]
[MSG:DBG: Y target 5.000 rate 200.000]
[MSG:DBG: Slow approach]
[MSG:DBG: X target -25.000 rate 200.000]
[MSG:DBG: Y target -25.000 rate 200.000]
[MSG:DBG: Pulloff1]
[MSG:DBG: X target 5.000 rate 200.000]
[MSG:DBG: Y target 5.000 rate 200.000]
[MSG:DBG: Homing Z]
[MSG:DBG: PrePulloff]
[MSG:DBG: Fast approach]
[MSG:DBG: Z target 30.000 rate 100.000]
[MSG:DBG: Pulloff0]
[MSG:DBG: Z target -9.500 rate 100.000]
[MSG:DBG: Slow approach]
[MSG:DBG: Z target 47.500 rate 100.000]
[MSG:DBG: Pulloff1]
[MSG:DBG: Z target -9.500 rate 100.000]
ok

With 3.5.1-Pre-FS : 
[MSG:DBG: Homing Cycle XY]
[MSG:DBG: Homing nextPhase FastApproach]
[MSG:DBG: Starting from 0.000,0.000,0.000]
[MSG:DBG: X target -465.000 rate 2500.000]
[MSG:DBG: Y target -315.000 rate 2500.000]
[MSG:DBG: planned move to -465.000,-315.000,0.000 @ 3535.534]
[MSG:DBG: Homing limited X]
[MSG:DBG: Homing limited XY]
[MSG:DBG: Homing nextPhase Pulloff0]
[MSG:DBG: Starting from -139.167,-112.776,0.000]
[MSG:DBG: X target -134.167 rate 200.000]
[MSG:DBG: Y target -107.776 rate 200.000]
[MSG:DBG: planned move to -134.167,-107.776,0.000 @ 282.843]
[MSG:DBG: Homing nextPhase SlowApproach]
[MSG:DBG: Starting from -134.163,-107.772,0.000]
[MSG:DBG: X target -159.163 rate 200.000]
[MSG:DBG: Y target -132.772 rate 200.000]
[MSG:DBG: planned move to -159.163,-132.772,0.000 @ 282.843]
[MSG:DBG: Homing limited X]
[MSG:DBG: Homing limited X]
ALARM:9
Homing fail. Could not find limit switch within search distance. Defined as 1.5 * max_travel on search and 5 * pulloff on locate phases.

You can see that Y limit switch is not found (100% reproducible) during slow approach.
Targets seem odd to me ... But I don't have the background to judge.

MitchBradley, 2022-09-18:
Try the [ParkingRedux branch ](https://github.com/bdring/FluidNC/tree/ParkingRedux).  It has some improvements to limit switch handling and more debug output for Homing.

Avataar120, 2022-09-18:
Hello,

[VER: FluidNC TestParking2 (ParkingRedux-26420a78):]
[OPT:MPHS]
[MSG: Machine: K40_AvaShield]
[MSG: Mode=STA:SSID=:Status=Not connected:IP=0.0.0.0:MAC=EC-94-CB-6D-3F-00]
[MSG:Mode=AP:SSID=FluidNC:IP=192.168.0.1:MAC=EC-94-CB-6D-3F-01]
ok

No movement at all when issuing $h command

closed 2022-09-23 by Avataar120