bdring/FluidNC #603: Problem: Scan offset, diode laser

xpozec, 2022-09-10:
### Controller Board

custom board 
[image]

### Help From Board Vendor

- [ ] Yes
- [ ] No
- [X] Not Applicable

### Machine Description

Laser engraver, custom made. using A4988 stepper drivers, 7.5W diode laser
120W 12V power supply (meanwell) 
2 machines 
[image]
[image]
[image]

### Configuration file

```yaml
name: "Laser"
board: "Byte 4axis"

stepping:
  engine: RMT 
  idle_ms: 255
  dir_delay_us: 1
  pulse_us: 2
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: gpio.13:high
  
  x:
    steps_per_mm: 80
    max_rate_mm_per_min: 4800
    acceleration_mm_per_sec2: 1200
    max_travel_mm: 450
    soft_limits: true
    homing:
      cycle: 1
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 400.000
      seek_mm_per_min: 3000.000
      settle_ms: 250
      seek_scaler: 1.0
      feed_scaler: 1.0
    
    motor0:
      limit_neg_pin: gpio.34:low
      pulloff_mm: 3.000
      stepstick:
        direction_pin: gpio.2:low
        step_pin: gpio.0
    motor1:
      null_motor:

  y:
    steps_per_mm: 80
    max_rate_mm_per_min: 4800
    acceleration_mm_per_sec2: 1200
    max_travel_mm: 450
    soft_limits: true
    homing:
      cycle: 1
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 400.000
      seek_mm_per_min: 3000.000
      settle_ms: 250
      seek_scaler: 1.0
      feed_scaler: 1.0

    motor0:
      limit_neg_pin: gpio.35:low
      pulloff_mm: 3.000
      stepstick:
        direction_pin: gpio.15:high
        step_pin: gpio.26
    motor1:
      null_motor:

 
  # z:
  #   steps_per_mm: 2000
  #   max_rate_mm_per_min: 500
  #   acceleration_mm_per_sec2: 100
  #   max_travel_mm: 5
  #   homing:
  #     cycle: 0
  #     mpos_mm: 0
  #     positive_direction: false

  #   motor0:
  #     rc_servo:
  #       pwm_hz: 50
  #       output_pin: gpio.27
  #       min_pulse_us: 700
  #       max_pulse_us: 2200
        
spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  cs_pin: gpio.5
  card_detect_pin: NO_PIN

coolant:
  flood_pin: NO_PIN
  mist_pin:  gpio.21

probe:
  pin: gpio.32:low:pu
  
Laser:
  pwm_hz: 5000
  output_pin: gpio.25
  enable_pin: NO_PIN
  disable_with_s0: false
  s0_with_disable: true
  tool_num: 0
  speed_map: 0=0.000% 1000=100.000%
```

### Startup Messages

```text
no errors, all green
```

### User Interface Software

Lightburn, lasergrbl

### What happened?

Engraving over USB. 
There is some kind of strange  offset happening with 3.4.4
With new version, i am unable to test, as at this moment I am unable to home my machines, it only works like 20% of the time. 
when using Arduino board, or ESP32grbl (discontinued) it does not happen photos of those beautiful lades and Robbin Williams, done on the same machine arduino board and same board with ESP32grbl (discontinued), this rules out hardware issues i think
Engraving with 35mm/s 67% power
If you need any more information in order to be able to help, please ask, I will do my best to provide all.

### Other Information

[image]
[image]
[image]
[image]

xpozec, 2022-09-10:
Did some more tests, as shown here: 
[image]
this is 0% overscaning, as i figured maybe this is the problem
[image]
2% and 5%
[image]
10% 15% 20%
[image]
upper one was not  a bidirectional scan, only in one direction...
[image]
and finally bottom one was 90degree scan, as in not scan on x but on y axis 10% overscaning... 
best result was the unidirectional scan, but that was not perfect as well.

MitchBradley, 2022-09-10:
Can you do some simpler tests like gray rectangles?  I need a test case that is easy to reproduce and think about.  Looking at Daenerys Targaeryen is too distracting.

Also try saving the file and running it from an SD card.  Trying to eliminate variables.

xpozec, 2022-09-10:
Sure, it is night time now, i will conduct those tests, and post my results in about 12 hours from now i hope

xpozec, 2022-09-11:
update
ran a interval test in LB
so no fancy Targaeryens today 
[image]
[image]
[image]
[image]
the 0.05 is an overkill, and only looks ''good'' because it is an overburn same for 0.07
where a problem really shows is 0.20
this crosshatch and wavy result but as per photo only intermittently

Avataar120, 2022-09-11:
Hello,

Could you share your lightburn file (.lbrn2) please ?

I'd like to test on my setup

Thx

xpozec, 2022-09-11:
[image]
LB does not allow me to save it as such, it is a laser tool, interval test, settings as per photo
I can save gcode file, not sure if that would help, but here it goes 
[test.zip](https://github.com/bdring/FluidNC/files/9542442/test.zip)

Avataar120, 2022-09-11:
Test on my side. 100mm/s - 10% power (K40)
It seems I don't have the same issue

[image]

xpozec, 2022-09-11:
I am no expert, but does that . in 0.20 not look wrong to you? offset is bad at my side.
Is it me or am I seeing same or similar wavy pattern on your test as well 
I have 2 lasers for test, and both are giving me same results. My CO2 is not having these problems

xpozec, 2022-09-12:
more tests were performed: 
re compiled the GRBL esp32, same hardware. 
results:

JHarding86, 2022-09-12:
This looks like LightBurns scan offset adjustment. Base on how the .05 doesnt show the issue. That is because .05 steps are smaller than the width of the focal point of the laser. If you perform the scan offset adjustments as [show here](https://docs.lightburnsoftware.com/ScanningOffsetAdjustment.html), your issues should go away.

xpozec, 2022-09-12:
you are absolutely right about lightburn scanning offset , however another issue is the wavy lines this firmware produces, and the person doing the same test replicated the same problem.
Just because laser cuts well, does not mean it is doing a nice engraving, and diode lasers are all about photo engraving.
But here, lets say that I can fix the offset, how am I supposed to get rid of the miniature arcs, and non parallel lines?
I have a test in mind.
Discussing this with other people as well we have reached a conclusion that y motor is for some reason pulsing while x is scanning. I will run a cycle with oscilloscope in a day or 2, post my result here.
When it comes to scanning offset, having to adjust it on speed of 35mm/s is not something any other firmware required, I have a huge CO2 laser 1300x900mm and can achieve speeds of up to 900mm/s without need for scanning offset.
I found a similar issue in one of the closed issues, their solution was to use I2C stepper drivers, something I am unable to do.
I do however thank each and every person for looking in to my issue and taking the time to try and help

MitchBradley, 2022-09-12:
I am trying to duplicate the problem, while juggling about 5 other issues simultaneously.

xpozec, 2022-09-12:
@MitchBradley thank you, I am available to share more information and do some more tests if need be, I do posses oscilloscope and intermediate knowledge in both hardware and software design, maintenance and am a veteran at laser technologies. So if there is more information you need let me know, in the meantime i will keep using this topic as a sort of a blog of what was done to try to solve problem, and results I am getting if you dont mind, after all more data we all have, more likely we are to reach some sensible conclusion, cheers

MitchBradley, 2022-09-12:
Can you just try a single line to see if it is wavy?

xpozec, 2022-09-13:
[image]
top one is esp32 grbl, bottom one is fluid nc, single line does not look like is an issue, some more tests will be performed today

MitchBradley, 2022-09-13:
Please attach the GCode file for the test above.

xpozec, 2022-09-13:
[test.zip](https://github.com/bdring/FluidNC/files/9558947/test.zip)

MitchBradley, 2022-09-13:
Try running that gcode file from SD card instead of streaming it from lightburn

xpozec, 2022-09-13:
I will install SD card reader, and let you know when I am done with tests

MitchBradley, 2022-09-14:
I converted an old EleksLaser gantry with a 2.5W laser to FluidNC by replacing the ancient EleksLaser Arduino Nano board with a [FYSETC E4 board](https://www.aliexpress.com/item/3256801518098396.html) that has TMC2209 drivers.  Here is the configuration file.  
[e4laser.yaml.txt](https://github.com/bdring/FluidNC/files/9570443/e4laser.yaml.txt)
The firmware is build from the[ StdFilesystem branch](https://github.com/bdring/FluidNC/tree/StdFilesystem).  Compared to v3.6.0, that branch has no changes that are relevant to motion or spindle/power control.
The laser is an old 2-pin one where the PWM is applied directly to the 12V power - not a "TTL PWM".  The E4 board's Extruder MOSFET switches the low side of the laser power on and off via PWM, with the high side of the laser connected directly to +12V.'

I edited your test file to remove the labels and engrave only 6 lines of the test pattern - since 6 lines is enough to show any systematic problems.  I also changed the distance between lines from 0.2 mm to 0.6 mm so the lines show up better and there is more Y travel to reveal any momentum issues.  I set the laser power to S300 which makes easy to read marks on yellow notepad paper without burning through.  Here is the file:
[test_small.gc.txt](https://github.com/bdring/FluidNC/files/9570468/test_small.gc.txt)
 I ran that file from SD using the WebUI Tablet UI.
Originally there was some waviness in the lines, but I was able to reduce it to a fairly small amount by dialing down the acceleration and making sure the gantry was square (the gantry has dual Y motors each with a separate driver).  Here are some runs:
[image]
* A100 SQ is acceleration=100 mm/sec2 with the gantry square.  I think the result is pretty good.
* A500 Sq is acceleration 500 mm/sec2 with the gantry square.  You can see some wobble (red arrows) due to the Y momentum still settling after the Y motion.
* A100 R1 is after I introduced some racking by pulling one end of the gantry manually by one step.  You can see some wobble (blue arrow) as the gantry tries to dissipate the mechanical stress.
* A100 R2 is the same; I pulled out out of square again by one step
In an earlier run - not shown - the gantry was badly racked and there was a significant amount of waviness.

xpozec, 2022-09-15:
Sure enough as soon as I reduced acceleration to 100 mm/sec2 wavy pattern was gone, however this is not good enough for engraving machines, not for the results, speeds etc that I am looking for. My machines have 1 y motor, and are shaft driven to the other end, this works perfectly and insures no skipping or any desync... 
If the ESP32 board and firmware gives out indication that it is the bottleneck on machines it is back to AVR boards, since Fluid NC can not handle fast raster, GRBL ESP32 randomly reports panicked core, sometimes after 10 min, sometimes after 12hours, ultimately ESP grbl ports are not there yet. 
Since I still have hardware, I will keep following the progress and do experiments as development moves forward.
Thank you guys 
[image]

MitchBradley, 2022-09-15:
Closing since OP seems to be giving up.

closed 2022-09-15 by MitchBradley