bdring/FluidNC #608: Problem: Homing fails, feed speed not correct. Works fine in 3.4.4 but not in 3.6.

kartom, 2022-09-11:
### Controller Board

Own built

### Help From Board Vendor

- [ ] Yes
- [ ] No
- [X] Not Applicable

### Machine Description

Router with DM542 steppers. Mechanical end switches.

### Configuration file

```yaml
name: "ESP32 Dev Controller V4"
board: "3 Axis Router"

start:
  must_home: false
  deactivate_parking: false
  check_limits: true

control:
  safety_door_pin: gpio.32:pu
  cycle_start_pin: gpio.33:low:pu
probe:
  pin: gpio.35:low
 
stepping:
  engine: RMT
  idle_ms: 250
  dir_delay_us: 5
  pulse_us: 3
  disable_delay_us: 0

axes: 
  x:
    steps_per_mm: 320.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 200.000
    max_travel_mm: 380.0
    soft_limits: true
    homing:
      cycle: 2
      mpos_mm: 380.000
      seek_mm_per_min: 400
      feed_mm_per_min: 50
    motor0:
      limit_pos_pin: gpio.34
      standard_stepper:
        step_pin: gpio.22
        direction_pin: gpio.21:low
  y:
    steps_per_mm: 320.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 200.000
    max_travel_mm: 595.0
    soft_limits: true
    homing:
      cycle: 2
      mpos_mm: 595.000
      seek_mm_per_min: 400
      feed_mm_per_min: 50
    motor0:
      limit_pos_pin: gpio.39
      standard_stepper:
        step_pin: gpio.17
        direction_pin: gpio.16:low
  z:
    steps_per_mm: 400.000
    max_rate_mm_per_min: 4000.000
    acceleration_mm_per_sec2: 200.000
    max_travel_mm: 115.000
    soft_limits: true
    homing:
      cycle: 1
      mpos_mm: 115.000
      seek_mm_per_min: 200
      feed_mm_per_min: 50
    motor0:
      limit_pos_pin: gpio.36
      standard_stepper:
        step_pin: gpio.4
        direction_pin: gpio.15:low

PWM:
  pwm_hz: 1000
  output_pin: gpio.13:low
  enable_pin: gpio.2
  disable_with_s0: false
  s0_with_disable: true
  spinup_ms: 5000
  spindown_ms: 5000
  tool_num: 0
  speed_map: 0=0.000% 24000=100.000%
  
spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  cs_pin: gpio.5
```

### Startup Messages

```text
[MSG:INFO: FluidNC v3.6.0]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
[MSG:INFO: Local filesystem type is SPIFFS]
[MSG:INFO: Configuration file:config.yaml]
[MSG:INFO: Machine ESP32 Dev Controller V4]
[MSG:INFO: Board 3 Axis Router]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN]
[MSG:INFO: Stepping:RMT Pulse:3us Dsbl Delay:0us Dir Delay:5us Idle Delay:250ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (0.000,380.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:gpio.22 Dir:gpio.21:low Disable:NO_PIN]
[MSG:INFO:     Pos Limit gpio.34]
[MSG:INFO: Axis Y (0.000,595.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:gpio.17 Dir:gpio.16:low Disable:NO_PIN]
[MSG:INFO:     Pos Limit gpio.39]
[MSG:INFO: Axis Z (0.000,115.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:gpio.4 Dir:gpio.15:low Disable:NO_PIN]
[MSG:INFO:     Pos Limit gpio.36]
[MSG:INFO: safety_door_pin gpio.32:pu]
[MSG:INFO: cycle_start_pin gpio.33:low:pu]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: PWM Spindle Ena:gpio.2 Out:gpio.13:low Dir:NO_PIN Freq:1000Hz Period:65535]
[MSG:INFO: Using spindle PWM]
[MSG:INFO: Probe Pin: gpio.35:low]
[MSG:INFO: Connecting to STA SSID:nnnnnnn]
[MSG:INFO: Connecting.]
[MSG:INFO: Connected - IP is 192.168.nnn.nnn]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
```

### User Interface Software

WebUI

### What happened?

Homing fails because pull off fials. And the reason for that is that the uses a far to low speed (barely moves). 
The jogging speed of the machine does not change when the speed in the WebUI is changed (the corresponding G-code command sends the new speed). 
There is something wrong with the calculation of the pulses to the steppers. Version 3.5 of the code could occasionally show the same problem and therefore i went back to 2.4.4 until this new 3.6 version was released, but it seams that the problem persists.

kartom, 2022-09-11:
BTW, the Z-axis works just fine. It's the X-and Y axis that act strange. That makes me believe that there is some overflow in some integer calculations that fails. The same way that if you use the above setup for the PWM frequency the output PWM wraps around and spindle speeds above 50% becomes PWM from 0-50%.

daxliniere, 2022-09-16:
> There is something wrong with the calculation of the pulses to the steppers.

I have seen exactly this in earlier versions, but not since 3.5, I think.

MitchBradley, 2022-10-10:
Please retry with v3.6.3.  We fixed a lot of things.

MitchBradley, 2022-10-16:
Please try this https://github.com/bdring/FluidNC/releases/tag/v3.6.4-pre1

closed 2024-12-19 by bdring