bdring/FluidNC #610: when stpper driver enable , does the gcode able be pause ?

kweilim98, 2022-09-13:
### Machine Context

Uploading WhatsApp Video 2022-09-13 at 9.49.52 AM.mp4…

### Feature Description

I am using the enable port with a relay for the safety door switch. When my door is open, the relay will cut the current and enable the stepper motor. The issue is the movement of the stepper motor is able to pause, but the Gcode sending still keeps streaming and does not pause. 

### Other Approaches

.

### How I Can Help

So does this problem able to change from the program? If this issue able to solve then please share the code, thx u

MitchBradley, 2022-09-13:
What sender program is that?  The video is too blurry to tell.

kweilim98, 2022-09-13:
> What sender program is that? The video is too blurry to tell.

it is not important which sender using, because i have try a lot sender already, all are the same program, like ugs, candle, cnc3d commander, and so on

kweilim98, 2022-09-13:
because I have to change the grbl parameter setting for the $4=0 by default into $4=1 to active the enable port for the stepper motor

MitchBradley, 2022-09-13:
The above statement makes no sense because FluidNC does not support $4.  You must use :low on the disable_pin: or shared_stepper_disable_pin: gpio specification in order to invert the logic sense for stepper enable.

We cannot solve problems without complete information about your configuration.  Please open a new issue using the Problem Report template and provide all of the information that it asks for.  Even if you think that some information is irrelevant to the problem, we need to understand the complete context in order to evaluate what could be going wrong.

kweilim98, 2022-09-13:
> The above statement makes no sense because FluidNC does not support $4. You must use :low on the disable_pin: or shared_stepper_disable_pin: gpio specification in order to invert the logic sense for [...]

hi bradley, actually the problem issue is very simple only. I just hope when the g-code is streaming and I enable the stepper motor, the streaming g-code will pause and when disabling the stepper motor, the streaming will continue back to the last stop position

kweilim98, 2022-09-13:
[image]

kweilim98, 2022-09-13:
[image]
this is my wiring connection

MitchBradley, 2022-09-13:
FluidNC has an input pin for the safety door state, configured via, for example:
```yaml
control:
   safety_door_pin: gpio.13:low:pu
```

When that input pin becomes active, FluidNC does the following:
* If any motion is in progress, decelerating to a controlled stop
* If parking is configured and possible, performs the parking procedure by retracting the parking axis, stopping the spindle, then moving the parking axis to the parking position.
* If parking is not possible, just stops the spindle
* The state as reported by the ? status report request changes from Cycle or Idle to Door:1, informing the sender that the controller is blocked on the safety door.
* GCode commands are not acknowledged with "ok", so even if the sender does not respect the status report information, it will stop sending because of the flow control effect.

The spindle stays off and motion stays off so long as the safety door input remains active.  Furthermore, FluidNC stops issuing "ok" responses to When the safety door input becomes inactive, the above steps are reversed; the spindle turns back and any parking motion is undone, so the tool returns to the place where it stopped.  Then the previously interrupted motion resumes from the stopping point and normal GCode reception and execution is resumed.

We will not support the scheme that you propose.  It is better suited for an Emergency Stop than 

The relay technique that you use has the problem that it forces stepping to stop instantly with no opportunity for controlled deceleration.  That will typically result in loss of accurate machine position because the inertia of the mechanical system will move it to a place that does not correspond to the software's position model.  That prevents clean restart after the door is closed.

Your scheme is better suited for an emergency stop situation than for Grbl's safety door behavior.  We do not intent to support your scheme in the safety door context.  If you want the door to act as an E-Stop, it should be connected to the reset input via control: reset_pin:

closed 2022-09-14 by MitchBradley