bdring/FluidNC #612: Feature: Solenoid as a spindle

brakthehun, 2022-09-15:
### Machine Context

I would like to have a row of pens that are controlled each by solenoids and using tool select (M6) to select different colors.

It would also be cool to use a solenoid as an engraving oscillator.

### Feature Description

solenoid:
  output_pin: gpio.26
  pwm_hz: 5000
  enable_pin: NO_PIN
  disable_with_s0: false
  s0_with_disable: true
  tool_num: 0
  off_percent: 0.000
  pull_percent: 100.000
  hold_percent: 20.000
  pull_ms: 75
  direction_invert: false

### Other Approaches

I currently use an auxilary Arduino UNO that reads the M7 and M8 to select between 4 different pens. It reads the m3 output to drop and pickup pens

### How I Can Help

I can Amazon a 4ch Mosfet and 4 12v solenoids to you guys.

bdring, 2022-09-16:
Can you useoutputs? user ouputs?

http://wiki.fluidnc.com/en/config/user_outputs

You could implement the pull and hold with gcode delays G4...

brakthehun, 2022-09-20:
Not quite. It's that percentage hold that I like after the initial stroke.

bdring, 2022-09-20:
This gcode has a pull and hold
```gcode
M67 E0 Q100.00 ; pull level 100%
G4 P0.1 ; wait 0.1 sec
M67 E0 Q50.00 ; hold level 50%
```

brakthehun, 2022-09-20:
It does,, But I don't know if I can make any of my Gcode creators can output that. I would also like to be drawing within 10ms so I don't make spots / puddles on what ever I'm drawing on.

closed 2024-12-12 by bdring