bdring/FluidNC #615: Reporting error code 38

daxliniere, 2022-09-15:
### Controller Board

MKS DLC32

### Help From Board Vendor

- [ ] Yes
- [ ] No
- [X] Not Applicable

### Machine Description

3-axis CNC mill

### Configuration file

```yaml
board: MKS-DLC32 V2.5
name: SWOLE-CNC
meta: (2202.09.07) Dax Liniere, originally by Skorpi

kinematics:
  Cartesian:

stepping:
  engine: I2S_STATIC
  idle_ms: 0
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0
  
axes:
  shared_stepper_disable_pin: I2SO.0
  x:
    steps_per_mm: 800.000
    #was 802.341
    max_rate_mm_per_min: 1700.000
# was 1500.000
    acceleration_mm_per_sec2: 80.000
    max_travel_mm: 302.000
    soft_limits: true
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 20.000
      seek_mm_per_min: 200.000
# was 100 in first SWOLE-CNC iteration running MKS firmware
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.36:high
      limit_pos_pin: gpio.34:high
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 1.000
      stepstick:
        step_pin: I2SO.1
        direction_pin: I2SO.2:low
        disable_pin: NO_PIN
        ms1_pin: NO_PIN
        ms2_pin: NO_PIN
        ms3_pin: NO_PIN
        reset_pin: NO_PIN

  y:
    steps_per_mm: 802.322
    #tested 806.062, features too small
    max_rate_mm_per_min: 1700.000
    acceleration_mm_per_sec2: 80.000
# was 300.000
    max_travel_mm: 186.000
    soft_limits: true
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 20.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.35:high
      limit_pos_pin: gpio.23:high:pd
      #this pin doesn't have the same input circuitry as the 3 dedicated on-board limit switch inputs
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 1.000
      stepstick:
        step_pin: I2SO.5
        direction_pin: I2SO.6:low
        disable_pin: NO_PIN
        ms1_pin: NO_PIN
        ms2_pin: NO_PIN
        ms3_pin: NO_PIN
        reset_pin: NO_PIN
  z:
    steps_per_mm: 802.497
    max_rate_mm_per_min: 1000.000
    acceleration_mm_per_sec2: 50.000
# was 500.000
    max_travel_mm: 46.000
    soft_limits: true
    homing:
      cycle: 1
      positive_direction: true
      mpos_mm: 0.000
      feed_mm_per_min: 20.000
      seek_mm_per_min: 100.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.18:high:pd
      #this pin doesn't have the same input circuitry as the 3 dedicated on-board limit switch inputs
      hard_limits: true
      pulloff_mm: 1.000
      stepstick:
        step_pin: I2SO.3
        direction_pin: I2SO.4:low
        disable_pin: NO_PIN
        ms1_pin: NO_PIN
        ms2_pin: NO_PIN
        ms3_pin: NO_PIN
        reset_pin: NO_PIN

i2so:
  bck_pin: gpio.16
  data_pin: gpio.21
  ws_pin: gpio.17

spi:
  miso_pin: gpio.12
  mosi_pin: gpio.13
  sck_pin: gpio.14

sdcard:
  cs_pin: gpio.15
  card_detect_pin: NO_PIN
# This could be GPIO.39, but Card Detect has no supported functions in FluidNC

control:
  safety_door_pin: gpio.33:low:pu
# Does the start pin need to be wired to the other side of the switch??
  cycle_start_pin: NO_PIN
  feed_hold_pin: NO_PIN
  reset_pin: NO_PIN
  macro0_pin: gpio.19:low:pu
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN

macros:
  startup_line0:
  startup_line1:
  macro0: $X&$H
  macro1:
  macro2:
  macro3: $SD/Run=lasertest.gcode

coolant:
  flood_pin: gpio.0
# continuous air
  mist_pin:  gpio.4
# pulsed air 
  delay_ms: 0

probe:
  pin: gpio.22:low
  check_mode_start: true
  

10V:
# Spindle
  direction_pin: NO_PIN
  forward_pin: gpio.5:low
  reverse_pin: NO_PIN
  output_pin: gpio.32
  enable_pin: NO_PIN
  pwm_hz: 5000
  disable_with_s0: false
  s0_with_disable: true
  spinup_ms: 10000
  spindown_ms: 12000
  tool_num: 0
  speed_map: 0=0% 0=25% 5868=25% 23237=99% 24000=100%
#DLC32 Spindle TTL max output voltage is 4.89v, so it can never reach 24000rpm/400Hz

#pwm:
#  direction_pin: NO_PIN
#  output_pin: gpio.32
#  enable_pin: gpio.5:low
#  pwm_hz: 5000
#  disable_with_s0: false
#  s0_with_disable: true
#  spinup_ms: 10000
#  spindown_ms: 12000
#  tool_num: 0
#  speed_map: 0=0% 0=25% 5868=25% 23237=99% 24000=100%

user_outputs:
  analog0_pin: NO_PIN
  analog1_pin: NO_PIN
  analog2_pin: NO_PIN
  analog3_pin: NO_PIN
  analog0_hz: 5000
  analog1_hz: 5000
  analog2_hz: 5000
  analog3_hz: 5000
  digital0_pin: NO_PIN
  digital1_pin: NO_PIN
  digital2_pin: NO_PIN
  digital3_pin: NO_PIN

start:
  must_home: false
# Pins that could be used: 0(SDA), 4(SCL), 5, 18, 19, 22, 23, 25, 26, 27, 32 (TTL spindle control), 33, 39, I2SO.7
# Pins not currently in use: 25 (TMC_CS_Y), 26 (TMC_CS_X), 27 (TMC_CS_Z), (39 SD card detect), I2SO.7
```

### Startup Messages

```text
*** Connecting to jserialcomm://COM5:115200
*** Fetching device status
*** Fetching device firmware version
*** Fetching device status codes
*** Fetching device state
*** Fetching device settings
*** Connected to FluidNC 3.6.0
```

### User Interface Software

UGS 2.0.12+dev

### What happened?

This was triggered by a tool number higher than the controller was expecting/set up to accept. It simply pauses and waits for the user to continue the program.
Looking up error 38 shows "Gcode max value exceeded" - is that generic, as in, a maximum anything value or is it just for tool number?
If it is the latter, could you change the text for that error code to mention tool number? (as it was in GRBL)
If it is the former, can we please get a separate error code for 'max tool number exceeded', please?

bdring, 2022-09-17:
What was the gcode line that that caused this error?

The tool number is stored in an unsigned 8 bit value. This limits the number to 255. This is the same as Grbl. That error is a generic gcode parsing error.

FluidNC assigns tool number ranges to to spindles. The range starts from the `tool_num:` value and goes to the next spindle's `tool_num:`  or to 255 if there is no next spindle.

The error is caught here in FNC

```cpp
                     case 'T':
                        axis_word_bit = GCodeWord::T;
                        if (value > MaxToolNumber) {
                            FAIL(Error::GcodeMaxValueExceeded);
                        }
                        log_info("Tool No: " << int_value);
                        gc_state.tool = int_value;
                        break;
                        
```

The error is here in [Grbl](https://github.com/gnea/grbl/blob/master/grbl/gcode.c#L315)

```c
case 'T': word_bit = WORD_T; 
  if (value > MAX_TOOL_NUMBER) { FAIL(STATUS_GCODE_MAX_VALUE_EXCEEDED); }
  gc_block.values.t = int_value;
  break;
						
```

I think it is best to keep the error number the same. There is little benefit to a num number and it could cause confusion with old gcode senders. We could add a `[MSG: Gcode T value exceeded max of 255]`. We could also increase the tool number range.

daxliniere, 2022-09-17:
I use long tool numbers so I can identify tools without having to renumber all of them when a new tool is bought that fits between existing tools.
The gcode would have been >30000 and <100000.

bdring, 2022-09-17:
I'll add the suggestion to our development tracker wish list.

bdring, 2022-09-17:
Huh?

daxliniere, 2022-09-17:
My current CNC tools:

[image]

bdring, 2022-09-17:
What do you mean by "defaulting to 255 if line is not present"?

daxliniere, 2022-09-17:
Just saying that if the current built-in default is 255 tools, if, after this change is implemented, the max_tools= line is missimg from config.yaml, FNC defaults to 255 tools.

bdring, 2022-09-17:
There is no configurable max. It is hard coded into the firmware. It would change for everyone. We would probably make it an unsigned 32 bit number, if we changed it. We could constrain it, but why?

daxliniere, 2022-09-17:
Yes, sure, that's even better.

closed 2022-10-02 by bdring