bdring/FluidNC #616: Probing motion doesn't always stop when UGS clearly shows ALARM: PROBE

daxliniere, 2022-09-15:
### Controller Board

MKS DLC32

### Help From Board Vendor

- [ ] Yes
- [ ] No
- [X] Not Applicable

### Machine Description

3-axis CNC mill

### Configuration file

```yaml
board: MKS-DLC32 V2.5
name: SWOLE-CNC
meta: (2202.09.07) Dax Liniere, originally by Skorpi

kinematics:
  Cartesian:

stepping:
  engine: I2S_STATIC
  idle_ms: 0
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0
  
axes:
  shared_stepper_disable_pin: I2SO.0
  x:
    steps_per_mm: 800.000
    #was 802.341
    max_rate_mm_per_min: 1700.000
# was 1500.000
    acceleration_mm_per_sec2: 80.000
    max_travel_mm: 302.000
    soft_limits: true
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 20.000
      seek_mm_per_min: 200.000
# was 100 in first SWOLE-CNC iteration running MKS firmware
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.36:high
      limit_pos_pin: gpio.34:high
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 1.000
      stepstick:
        step_pin: I2SO.1
        direction_pin: I2SO.2:low
        disable_pin: NO_PIN
        ms1_pin: NO_PIN
        ms2_pin: NO_PIN
        ms3_pin: NO_PIN
        reset_pin: NO_PIN

  y:
    steps_per_mm: 802.322
    #tested 806.062, features too small
    max_rate_mm_per_min: 1700.000
    acceleration_mm_per_sec2: 80.000
# was 300.000
    max_travel_mm: 186.000
    soft_limits: true
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 20.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.35:high
      limit_pos_pin: gpio.23:high:pd
      #this pin doesn't have the same input circuitry as the 3 dedicated on-board limit switch inputs
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 1.000
      stepstick:
        step_pin: I2SO.5
        direction_pin: I2SO.6:low
        disable_pin: NO_PIN
        ms1_pin: NO_PIN
        ms2_pin: NO_PIN
        ms3_pin: NO_PIN
        reset_pin: NO_PIN
  z:
    steps_per_mm: 802.497
    max_rate_mm_per_min: 1000.000
    acceleration_mm_per_sec2: 50.000
# was 500.000
    max_travel_mm: 46.000
    soft_limits: true
    homing:
      cycle: 1
      positive_direction: true
      mpos_mm: 0.000
      feed_mm_per_min: 20.000
      seek_mm_per_min: 100.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.18:high:pd
      #this pin doesn't have the same input circuitry as the 3 dedicated on-board limit switch inputs
      hard_limits: true
      pulloff_mm: 1.000
      stepstick:
        step_pin: I2SO.3
        direction_pin: I2SO.4:low
        disable_pin: NO_PIN
        ms1_pin: NO_PIN
        ms2_pin: NO_PIN
        ms3_pin: NO_PIN
        reset_pin: NO_PIN

i2so:
  bck_pin: gpio.16
  data_pin: gpio.21
  ws_pin: gpio.17

spi:
  miso_pin: gpio.12
  mosi_pin: gpio.13
  sck_pin: gpio.14

sdcard:
  cs_pin: gpio.15
  card_detect_pin: NO_PIN
# This could be GPIO.39, but Card Detect has no supported functions in FluidNC

control:
  safety_door_pin: gpio.33:low:pu
# Does the start pin need to be wired to the other side of the switch??
  cycle_start_pin: NO_PIN
  feed_hold_pin: NO_PIN
  reset_pin: NO_PIN
  macro0_pin: gpio.19:low:pu
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN

macros:
  startup_line0:
  startup_line1:
  macro0: $X&$H
  macro1:
  macro2:
  macro3: $SD/Run=lasertest.gcode

coolant:
  flood_pin: gpio.0
# continuous air
  mist_pin:  gpio.4
# pulsed air 
  delay_ms: 0

probe:
  pin: gpio.22:low
  check_mode_start: true
  

10V:
# Spindle
  direction_pin: NO_PIN
  forward_pin: gpio.5:low
  reverse_pin: NO_PIN
  output_pin: gpio.32
  enable_pin: NO_PIN
  pwm_hz: 5000
  disable_with_s0: false
  s0_with_disable: true
  spinup_ms: 10000
  spindown_ms: 12000
  tool_num: 0
  speed_map: 0=0% 0=25% 5868=25% 23237=99% 24000=100%
#DLC32 Spindle TTL max output voltage is 4.89v, so it can never reach 24000rpm/400Hz

#pwm:
#  direction_pin: NO_PIN
#  output_pin: gpio.32
#  enable_pin: gpio.5:low
#  pwm_hz: 5000
#  disable_with_s0: false
#  s0_with_disable: true
#  spinup_ms: 10000
#  spindown_ms: 12000
#  tool_num: 0
#  speed_map: 0=0% 0=25% 5868=25% 23237=99% 24000=100%

user_outputs:
  analog0_pin: NO_PIN
  analog1_pin: NO_PIN
  analog2_pin: NO_PIN
  analog3_pin: NO_PIN
  analog0_hz: 5000
  analog1_hz: 5000
  analog2_hz: 5000
  analog3_hz: 5000
  digital0_pin: NO_PIN
  digital1_pin: NO_PIN
  digital2_pin: NO_PIN
  digital3_pin: NO_PIN

start:
  must_home: false
# Pins that could be used: 0(SDA), 4(SCL), 5, 18, 19, 22, 23, 25, 26, 27, 32 (TTL spindle control), 33, 39, I2SO.7
# Pins not currently in use: 25 (TMC_CS_Y), 26 (TMC_CS_X), 27 (TMC_CS_Z), (39 SD card detect), I2SO.7
```

### Startup Messages

```text
*** Connecting to jserialcomm://COM5:115200
*** Fetching device status
*** Fetching device firmware version
*** Fetching device status codes
*** Fetching device state
*** Fetching device settings
*** Connected to FluidNC 3.6.0
```

### User Interface Software

UGS 2.0.12+dev

### What happened?

Probing motion doesn't always stop when UGS clearly shows ALARM: PROBE.

daxliniere, 2022-09-15:
I just had this happen again now. Perhaps my z-probe macro has a problem, but for some reason even though probe triggering is being detected correctly by FNC (displaued in UGS), it's not stopping.

> G21
> G53 G38.2 Z-46 F100
> G0 Z1
> G38.2 Z-2 F50
> G10 L20 P1 Z4.985
> ok
> [PRB:7.177,16.006,-9.414:1]
> ok
> ok
> ALARM:4
> ok
> G10 L20 P2 Z4.985
error:9
*** The communicator has been paused
> G10 L20 P3 Z4.985
error:9
*** The communicator has been paused
error:9
*** The communicator has been paused
> G10 L20 P4 Z4.985
error:9
*** The communicator has been paused
> G10 L20 P5 Z4.985
error:9
*** The communicator has been paused
> G10 L20 P6 Z4.985
error:9
*** The communicator has been paused
> G90 G0 Z20
> G90
error:9
*** The communicator has been paused
error:9

MitchBradley, 2022-09-16:
Probing is working perfectly for me with my latest test build from the StdFilesystem branch.  The motor stops when the probe is touched, and also when the distance expires without touching the problem.  If I run the g38.2 when the probe is already active, I get an immediate ALARM:4 and the motor does not move at all.

I think I see a problem in youi script.  The "G0 Z1" line is ambiguous because the initial position in work space is unknown, so who knows which direction is Z1 ?  Am I missing something?

MitchBradley, 2022-09-26:
Any update on this?  Is it still a problem?

bdring, 2022-10-02:
You might consider using the new P parameter in your macro. It will be more accurate, because your macro does not account for deceleration.

http://wiki.fluidnc.com/en/features/supported_gcodes#g38-probing

The double touch is typically not needed for accuracy if you probe correctly. If you want to use that consider 

G91
G0 Z1
G90

as the way to retract 1 mm.

daxliniere, 2022-10-02:
> You might consider using the new P parameter in your macro. It will be more accurate, because your macro does not account for deceleration.
>
Rock 'n' roll. 🤘

Also, I updated a few things on that wiki page. The biggest changes were to the descriptions of G38 commands. Take a look to make sure I didn't get anything wrong?

> The double touch is typically not needed for accuracy if you probe correctly. If you want to use that consider

Interesting. What do you describe as correctly? Typically I'll probe at 80mm/min, then double-probe at 20mm/min. The few seconds extra doesn't bother me it if truly results in a more accurate probe measurement.

MitchBradley, 2022-10-02:
Have you measured to see if you get better/more repeatable results at 20 vs 80?

daxliniere, 2022-10-02:
Can you propose some meaningful tests I could perform? What about putting a flat-ended piece of bar stock into the collet and a DTI under it to measure some difference?

MitchBradley, 2022-11-09:
Is there anything more to be done on this issue?  If not can we close it?