bdring/FluidNC #617: FluidTerm doesn't display what is being typed

daxliniere, 2022-09-15:
### Controller Board

N/A

### Help From Board Vendor

- [ ] Yes
- [ ] No
- [ ] Not Applicable

### Machine Description

N/A

### Configuration file

```yaml
N/A
```

### Startup Messages

```text
N/A
```

### User Interface Software

FluidTerm

### What happened?

FluidTerm does not display keystrokes after a CTRL+R reset, though it seems FNC buffers them.
I found that I could type a command like $h and pressing enter would start the command, but nothing is displayed in FT.

You can see it happening from here: https://youtu.be/aow6SEUhhIM?t=156

bdring, 2022-09-16:
See advanced terminal mode.

TLDR; try ctrl+L

http://wiki.fluidnc.com/en/features/serial_terminals

daxliniere, 2022-09-16:
Just to see the cursor and what I'm typing? Surely that's a bug, right?

daxliniere, 2022-09-16:
I sent CTRL+L and it makes no difference - what I'm typing is not being displayed. (though I can confirm it is definitely being "read"/"captured" by FNC.)

MitchBradley, 2022-09-16:
CTRL-L turns off echoing.  To turn it on you need to use an editing character like right-arrow.

MitchBradley, 2022-09-16:
Keep in mind that classic GRBL has no line editing and never echos characters.  Most senders collect lines for you, with echoing and editing, then send the completed line to GRBL.  Senders do not expect GRBL to echo and typically will fail if echoing happens.  Se we cannot enable echoing by default lest we break every sender out there.  The finesse, for use with terminal programs like FluidTerm, is that echoing is only turned on after you type the first line-editing character (in the case of arrow keys, it is not actually a single character but rather a complicated sequence that begins with the ESC character) - which is presumably safe since a sender will never send such a line-editing character or sequence.

In an attempt to make things easy for users, FluidTerm sends right-arrow immediately after it starts up, thus turning on echoing.  That mostly works but there is a failure mode whereby if you run FluidTerm, then close it and start a sender without resetting the ESP32, echoing might already be on and the sender would get confused.  For this scenario and senders that know about FluidNC, we offer the CTRL-L command to turn off echoing.

There is a further problem related to hard-resetting the ESP32 with CTRL-R or the reset button.  Doing so returns to the default no-echo state, but FluidTerm does not know that a reset has occurred in the reset button case.  In the CTRL-R case, FluidTerm does know, and attempts to issue a right-arrow after a startup delay.  That sometimes works but there are cases where if fails, for example if startup takes a long time due to Wifi connection or whatever.  Ideally, FluidTerm would notice the startup messages and issue the right-arrow at exactly the right time, but doing it that way is a lot of work and there are a lot of higher priority things to do.

So:
* No good deed goes unpunished (the good deed in this case being line editing and echoing)\
* This is a lot more complicated than people seem to think
* <rant>Coexisting with the GRBL ecosystem is not easy.  Many GRBL design decisions that were made due to lack of program space in AVRs cause much grief to those of us trying to modernize it - and the implications of those decisions are "cast in entropy" by all the many senders that treat GRBL as "what you see is all you can ever get" </rant>

daxliniere, 2022-09-16:
Ahhhhhh, I see! Right. Yes, that does make sense. Is there any value in creating a "new GRBL"? As in, just do what you want to do and get the various sender developers to adopt this new FluidNC standard?

closed 2022-09-16 by daxliniere

MitchBradley, 2022-09-16:
"new GRBL" - "the various sender developers" is an almost unbounded list.  Anybody and his brother, armed with 10 minutes of programming experience, can make a sender, and somebody else will want to use it.  Trying to find them all and get them on board would be a herculean effort.  I have been involved in standards committees and it is more hassle than you can believe.  That said, I did try to get the grblHAL guy to coordinate on a simple specification for the identification string "Grbl <version> (other stuff)".  He was not interested at all.

We do have a lot of traction with the UGS developer so we are trying to make that work.

There are a few good/well-supported/actively-developed senders out there - LightBurn, Candle, Commander - so we will try to work with them as time and their own interest permits.