bdring/FluidNC #618: Limit switch is ignored if already triggered at time of homing

daxliniere, 2022-09-15:
### Controller Board

MKS DLC32

### Help From Board Vendor

- [ ] Yes
- [ ] No
- [X] Not Applicable

### Machine Description

3-axis CNC mill

### Configuration file

```yaml
board: MKS-DLC32 V2.5
name: SWOLE-CNC
meta: (2202.09.07) Dax Liniere, originally by Skorpi

kinematics:
  Cartesian:

stepping:
  engine: I2S_STATIC
  idle_ms: 0
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0
  
axes:
  shared_stepper_disable_pin: I2SO.0
  x:
    steps_per_mm: 800.000
    #was 802.341
    max_rate_mm_per_min: 1700.000
# was 1500.000
    acceleration_mm_per_sec2: 80.000
    max_travel_mm: 302.000
    soft_limits: true
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 20.000
      seek_mm_per_min: 200.000
# was 100 in first SWOLE-CNC iteration running MKS firmware
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.36:high
      limit_pos_pin: gpio.34:high
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 1.000
      stepstick:
        step_pin: I2SO.1
        direction_pin: I2SO.2:low
        disable_pin: NO_PIN
        ms1_pin: NO_PIN
        ms2_pin: NO_PIN
        ms3_pin: NO_PIN
        reset_pin: NO_PIN

  y:
    steps_per_mm: 802.322
    #tested 806.062, features too small
    max_rate_mm_per_min: 1700.000
    acceleration_mm_per_sec2: 80.000
# was 300.000
    max_travel_mm: 186.000
    soft_limits: true
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 20.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.35:high
      limit_pos_pin: gpio.23:high:pd
      #this pin doesn't have the same input circuitry as the 3 dedicated on-board limit switch inputs
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 1.000
      stepstick:
        step_pin: I2SO.5
        direction_pin: I2SO.6:low
        disable_pin: NO_PIN
        ms1_pin: NO_PIN
        ms2_pin: NO_PIN
        ms3_pin: NO_PIN
        reset_pin: NO_PIN
  z:
    steps_per_mm: 802.497
    max_rate_mm_per_min: 1000.000
    acceleration_mm_per_sec2: 50.000
# was 500.000
    max_travel_mm: 46.000
    soft_limits: true
    homing:
      cycle: 1
      positive_direction: true
      mpos_mm: 0.000
      feed_mm_per_min: 20.000
      seek_mm_per_min: 100.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.18:high:pd
      #this pin doesn't have the same input circuitry as the 3 dedicated on-board limit switch inputs
      hard_limits: true
      pulloff_mm: 1.000
      stepstick:
        step_pin: I2SO.3
        direction_pin: I2SO.4:low
        disable_pin: NO_PIN
        ms1_pin: NO_PIN
        ms2_pin: NO_PIN
        ms3_pin: NO_PIN
        reset_pin: NO_PIN

i2so:
  bck_pin: gpio.16
  data_pin: gpio.21
  ws_pin: gpio.17

spi:
  miso_pin: gpio.12
  mosi_pin: gpio.13
  sck_pin: gpio.14

sdcard:
  cs_pin: gpio.15
  card_detect_pin: NO_PIN
# This could be GPIO.39, but Card Detect has no supported functions in FluidNC

control:
  safety_door_pin: gpio.33:low:pu
# Does the start pin need to be wired to the other side of the switch??
  cycle_start_pin: NO_PIN
  feed_hold_pin: NO_PIN
  reset_pin: NO_PIN
  macro0_pin: gpio.19:low:pu
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN

macros:
  startup_line0:
  startup_line1:
  macro0: $X&$H
  macro1:
  macro2:
  macro3: $SD/Run=lasertest.gcode

coolant:
  flood_pin: gpio.0
# continuous air
  mist_pin:  gpio.4
# pulsed air 
  delay_ms: 0

probe:
  pin: gpio.22:low
  check_mode_start: true
  

10V:
# Spindle
  direction_pin: NO_PIN
  forward_pin: gpio.5:low
  reverse_pin: NO_PIN
  output_pin: gpio.32
  enable_pin: NO_PIN
  pwm_hz: 5000
  disable_with_s0: false
  s0_with_disable: true
  spinup_ms: 10000
  spindown_ms: 12000
  tool_num: 0
  speed_map: 0=0% 0=25% 5868=25% 23237=99% 24000=100%
#DLC32 Spindle TTL max output voltage is 4.89v, so it can never reach 24000rpm/400Hz

#pwm:
#  direction_pin: NO_PIN
#  output_pin: gpio.32
#  enable_pin: gpio.5:low
#  pwm_hz: 5000
#  disable_with_s0: false
#  s0_with_disable: true
#  spinup_ms: 10000
#  spindown_ms: 12000
#  tool_num: 0
#  speed_map: 0=0% 0=25% 5868=25% 23237=99% 24000=100%

user_outputs:
  analog0_pin: NO_PIN
  analog1_pin: NO_PIN
  analog2_pin: NO_PIN
  analog3_pin: NO_PIN
  analog0_hz: 5000
  analog1_hz: 5000
  analog2_hz: 5000
  analog3_hz: 5000
  digital0_pin: NO_PIN
  digital1_pin: NO_PIN
  digital2_pin: NO_PIN
  digital3_pin: NO_PIN

start:
  must_home: false
# Pins that could be used: 0(SDA), 4(SCL), 5, 18, 19, 22, 23, 25, 26, 27, 32 (TTL spindle control), 33, 39, I2SO.7
# Pins not currently in use: 25 (TMC_CS_Y), 26 (TMC_CS_X), 27 (TMC_CS_Z), (39 SD card detect), I2SO.7
```

### Startup Messages

```text
*** Connecting to jserialcomm://COM5:115200
*** Fetching device status
*** Fetching device firmware version
*** Fetching device status codes
*** Fetching device state
*** Fetching device settings
*** Connected to FluidNC 3.6.0
```

### User Interface Software

UGS 2.0.12+dev

### What happened?

FNC 3.6.1-pre1-FS

Limit switch is ignored if already triggered at time of homing, resulting in carriage grinding up again end stop until controller is reset.

See it happen here: https://www.youtube.com/watch?v=aow6SEUhhIM

MitchBradley, 2022-09-16:
I would like to see the circuit for your limit switches.  Something strange is happening that I am having trouble reproducing,

MitchBradley, 2022-09-16:
Using 3.5.1-pre1-FS on a laser gantry with dual motors and dual negative limit switches, squared homing is flawless.  If a switch is tripped when I issue $HY, the first move is a pulloff, then the homing cycle proceeds as expected.

robercy, 2022-09-16:
Same problem to me. Only Firmware 3.4.4 works flawless.  Firmware 3.5.1-pre1-FS have the same problem.

Avataar120, 2022-09-16:
Same problem here (please see my last comment in #602 when I reopened it)

daxliniere, 2022-09-16:
> I would like to see the circuit for your limit switches. Something strange is happening that I am having trouble reproducing,

[image]

MitchBradley, 2022-09-16:
So why is :pd enabled for such switches?  pd effectively connects a resistor across the transistor, making it appear to be always on.

MitchBradley, 2022-09-16:
I can imagine why it used to "work".  Since some versions of FluidNC had the bug of always enabling pullups, even when you did not ask for them, the :pu would turn into :pu + :pd.  The combination of pu and pd creates a voltage divider that sets the transistor-off-state voltage at some intermediate value that just happens to be above the input threshold, albeit with a narrow noise margin.

daxliniere, 2022-09-16:
> So why is :pd enabled for such switches? pd effectively connects a resistor across the transistor, making it appear to be always on.

From what I recall, I tried the correct solution, but it wasn't behaving as expected, but I'll try it again and see what TF goes on.

MitchBradley, 2022-09-16:
I don't see a schematic for v2.5 of DLC32 on the MKS github.  The closest I can find is v2.1.  It shows the circuits for X-, Y-, Z- (GPIO 36, 35, 34) but does not show the values of the pullup and series resistors in those circuits.  Lacking those values, it is impossible to calculate the voltages when :pd is active.  But I can say for sure that :pd is not helpful for those cases.

For GPIO 18 (Z limit_neg_pin) AKA LCD_SCK, :pd is definitely wrong, and there is no pullup at all.  So that pin will never go high.  You really should add an external pullup as the internal ones on ESP32 are too weak to give decent noise margin even if you say :pu

MitchBradley, 2022-09-16:
GPIO 23 - same comments as for GPIO 18 - omit :pd and add external pullup of say 10K.  A 0.1uF cap to ground wouldn't hurt either.

daxliniere, 2022-09-16:
V2.5 is my file version. The DLC32 is v2.1 rev1

daxliniere, 2022-09-16:
> GPIO 23 - same comments as for GPIO 18 - omit :pd and add external pullup of say 10K. A 0.1uF cap to ground wouldn't hurt either.

Sure. I'll make those changes tonight.

daxliniere, 2022-09-16:
So should I remove :pd from the 3 built-in limit inputs?

MitchBradley, 2022-09-16:
The only case where :pd would be useful would be a high-side switch - connected between V+ and the input.  Even then, it would be better to use an external pulldown resistor.  The internal resistors on ESP32 are really intended to keep the inputs from floating to an indeterminate state.  They are too weak for signals going off-board.

sjonholle, 2022-09-16:
I've  noticed this a few times with homing. My pulloff_mm setting was 1.500 mm and i changed it to 2.000 mm. But it could be that i'm thinking too simple. But at this moment is is working again.

sjonholle, 2022-09-17:
in that case: you made fluid-nc even better :-)

jnoam, 2022-09-17:
Sorry to barge in with what might be my newbiness. But I seem to have the same problem. Whenever something bad happens and I reach the limit switch (don't know why this is happening at all - shouldn't the limit be at 0-pulloff_mm?) then I cannot re-home and if I reboot the controller and then home - it  drives straight through the limit. 
my version is 3.6.0; my relevant config is 
`  x:
    steps_per_mm: 80.16339034975746
    max_rate_mm_per_min: 8000.000
    acceleration_mm_per_sec2: 1500.000
    max_travel_mm: 740
    soft_limits: true
 # moved Z connector here as there are two plugs there
    homing:
      cycle: 1
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 300.000
      seek_mm_per_min: 5000.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100
    motor0:
      limit_neg_pin: gpio.22
      hard_limits: TRUE
      pulloff_mm: 4.000`

sjonholle, 2022-09-21:
This is what i noticed and i hope this is helpful (maybe not): my pulse_us was 3 and during homing, i got several times alarm:6 (pulse must be greater than 3 us) so i changed this setting to 4. It looks like it is working fine now.  https://github.com/bdring/Grbl_Esp32/issues/1246 (i didn't notice the 3 us)

MitchBradley, 2022-09-26:
I am closing this because the OP's problem is solved.  If @jnoam is still having problems, please open a new issue with the complete information about your machine, per the issue template.

closed 2022-09-26 by MitchBradley

daxliniere, 2022-10-11:
Unfortunately this is still not solved for me with 3.6.3 and 10K PU resistor between +3.3v and Z- input pin. :(

bdring, 2022-10-11:
Show your current schematic and config for the pin.

daxliniere, 2022-10-11:
Here's video. Answer to your question coming soon.

bdring, 2022-10-11:
Have you measured the voltage at the gpio pin with the switch in both positions?

daxliniere, 2022-10-11:
Should it matter? The $limits test indicates it is working correctly.
Is there something I need to know about the limits test?

daxliniere, 2022-10-11:
Okay, so there's a problem here. I changed **limit_neg_pin: gpio.18:high** to **limit_neg_pin: gpio.18** and _$limits_ still performs the exact same way. I believe it should be reporting the opposite, right?

daxliniere, 2022-10-11:
Right, so even after changing from **:high**, I still get the same failure.

bdring, 2022-10-11:
No high is the default. You should switch to `:low`

daxliniere, 2022-10-11:
Changed to `:low`, now it tries to home and just keeps jamming against the top of the limit. This time is doesn't stop after 3 seconds and show error 8.

They are NC switches through an optocoupler, so in the non-triggered state, the IR LED is on and the pin is pulled to ground by the phototransistor. 

Both my Z- and Y- pins are showing their states as inverted without using `:high`

bdring, 2022-10-11:
OK, that active state change was your idea. You were just doing it wrong.

Set $message/level=debug and reply with the debug text during a Z homing.

daxliniere, 2022-10-11:
Ahh, so it's trying to pulling off in the wrong direction, I guess.

bdring, 2022-10-11:
Paste you Z axis config. Use the yaml code block formatting.

daxliniere, 2022-10-11:
Updated first post.

bdring, 2022-10-11:
I think it should be a `limit_pos_pin:`

daxliniere, 2022-10-11:
> I think it should be a `limit_pos_pin:`

Argh. Right, this was the solution all along. 🤦‍♂️

bdring, 2022-10-11:
It needs to know which way to pull_off from a stuck limit switch. If it is touching a negative switch it knows it must move in the positive direction. If it is a `limit_all_switch:` it will just error right away.