bdring/FluidNC #619: Problem: No laser PWM output with 3.6.0

truglodite, 2022-09-18:
### Controller Board

6-pack v1p5, 2x quad inputs, 1x rs485, 1x 5V out, and 1x fet out, in that order.

### Help From Board Vendor

- [ ] Yes
- [ ] No
- [ ] Not Applicable

### Machine Description

MPCNC Primo w/ 5x DM542 drivers @48V

### Configuration file

```yaml
name: TrugMPCNC
board: 6-pack

# MPCNC Primo J w/ dual endstops, z-max, & 1-start z screw
# 6-pack v1p5 controller
# 4x DM542 drivers @3200/rev (1/16)
# 4x DM542 drivers @800/rev (1/4)
# X1X2Y1Y2Z driver layout (left to right... vs default X1Y1ZX2Y2)
# 2x quad input, 1x RS485, 1x 5V output, 1x fet 12V output
# Huanyang rs485 VFD+1.5kW spindle & PWM laser (M6 T0 spindle, M6 T1 laser)
# Pause and Start buttons
# LED output0, 12VDC & 120VAC spindle relays output1 (M62/63 P0 and M62/63 P1)

kinematics:
  Cartesian:

stepping:
  engine: I2S_static
  idle_ms: 255
  dir_delay_us: 0
  pulse_us: 4
  disable_delay_us: 0

axes:
  x:
    steps_per_mm: 100.000
    max_rate_mm_per_min: 7500.000
    acceleration_mm_per_sec2: 127.000
    max_travel_mm: 562.000
    soft_limits: true
    homing:
      cycle: 2
      mpos_mm: -562.000
      positive_direction: false
      settle_ms: 250.000
      seek_mm_per_min: 2000.000
      feed_mm_per_min: 300.000
      seek_scaler: 1.100
      feed_scaler: 5.000
    motor0:
      limit_neg_pin: gpio.33
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 1.000
      stepstick:
        direction_pin: i2so.1:low
        step_pin: i2so.2
        disable_pin: i2so.0
        ms3_pin: i2so.3
    motor1:
      limit_neg_pin: gpio.32
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 1.000
      stepstick:
        direction_pin: i2so.4
        step_pin: i2so.5
        disable_pin: i2so.7
        ms3_pin: i2so.6
  y:
    steps_per_mm: 100.000
    max_rate_mm_per_min: 7500.000
    acceleration_mm_per_sec2: 127.000
    max_travel_mm: 562.000
    soft_limits: true
    homing:
      cycle: 2
      mpos_mm: -562.000
      positive_direction: false
      settle_ms: 250.000
      seek_mm_per_min: 2000.000
      feed_mm_per_min: 300.000
      seek_scaler: 1.100
      feed_scaler: 5.000
    motor0:
      limit_neg_pin: gpio.35
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 1.000
      stepstick:
        direction_pin: i2so.9
        step_pin: i2so.10
        disable_pin: i2so.8
        ms3_pin: i2so.11
    motor1:
      limit_neg_pin: gpio.34
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 1.000
      stepstick:
        direction_pin: i2so.12:low
        step_pin: i2so.13
        disable_pin: i2so.15
        ms3_pin: i2so.14
  z:
    steps_per_mm: 400.000
    max_rate_mm_per_min: 4000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 150.000
    soft_limits: true
    homing:
      cycle: 1
      mpos_mm: 0.000
      positive_direction: true
      settle_ms: 250.000
      seek_mm_per_min: 700.000
      feed_mm_per_min: 300.000
      seek_scaler: 1.100
      feed_scaler: 5.000
    motor0:
      limit_neg_pin: gpio.25
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 1.000
      stepstick:
        direction_pin: i2so.17
        step_pin: i2so.18
        disable_pin: i2so.16
        ms3_pin: i2so.19
    motor1:
      null_motor:

i2so:
  bck_pin: gpio.22
  data_pin: gpio.21
  ws_pin: gpio.17

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  cs_pin: gpio.5

control:
  feed_hold_pin: gpio.36
  cycle_start_pin: gpio.2:pu
  
Huanyang:
  uart:
    txd_pin: gpio.26
    rxd_pin: gpio.16
    rts_pin: gpio.4
    baud: 9600
    mode: 8N1
  modbus_id: 1
  tool_num: 0
  speed_map: 0=0% 0=25% 6000=25% 24000=100%

coolant:
  flood_pin: i2so.24
  mist_pin: gpio.15
  delay_ms: 1000.000

probe:
  pin: gpio.39
  check_mode_start: false

macros:
  startup_line0: 
  startup_line1: 
  macro0: 
  macro1: 
  macro2: 
  macro3: 

start:
  must_home: true
  check_limits: true
  deactivate_parking: false

user_outputs:
  digital0_pin: i2so.25
  digital1_pin: gpio.14

laser:
  tool_num: 1
  speeds: 0=0.0% 1000=100.0%
  output_pin: gpio.13
  disable_with_s0: false
  s0_with_disable: false
  pwm_hz: 5000
arc_tolerance_mm: 0.002
junction_deviation_mm: 0.010
verbose_errors: false
report_inches: false
enable_parking_override_control: false
use_line_numbers: false
```

### Startup Messages

```text
<Alarm|MPos:0.000,0.000,0.000|FS:0,0>
<Alarm|MPos:0.000,0.000,0.000|FS:0,0>
<Alarm|MPos:0.000,0.000,0.000|FS:0,0>
<Alarm|MPos:0.000,0.000,0.000|FS:0,0>
$ss
<Alarm|MPos:0.000,0.000,0.000|FS:0,0>
<Alarm|MPos:0.000,0.000,0.000|FS:0,0>

...?
```

### User Interface Software

bCNC

### What happened?

The exact same setup and config file worked fine with 3.4.4. I tested 3.6.0 and everything else works but laser pwm (stays 0V on the scope), and terminal reports no errors. Then I flashed 3.4.4, and it works again.

### Other Information

 Sorry I can't think of any other helpful info to share about this problem.

MitchBradley, 2022-09-18:
Connect with FluidTerm and show the complete startup messages.

trihwangyudi, 2022-09-18:
i have the same issue but in may motors, they don't move, has no power troug them, i'm flshing again with 3.4.4 they have same no moving

bdring, 2022-09-18:
It is important to show us the startup messages. [Here is a link to the wiki on that](http://wiki.fluidnc.com/en/support/requesting_help#fluidnc-startup-messages).

It loaded your file and got an error. In your laser config you have `speeds:` it should be `speed_map:`. I think it will load a default, I fixed that and my startup messages look like this.

```
[MSG:INFO: FluidNC v3.6.0]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
[MSG:INFO: Local filesystem type is SPIFFS]
[MSG:INFO: Configuration file:trug.yaml]
[MSG:INFO: Machine TrugMPCNC]
[MSG:INFO: Board 6-pack]
[MSG:INFO: I2SO BCK:gpio.22 WS:gpio.17 DATA:gpio.21]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN]
[MSG:INFO: Stepping:I2S_static Pulse:4us Dsbl Delay:0us Dir Delay:0us Idle Delay:255ms]
[MSG:INFO: User Digital Output:0 on Pin:I2SO.25]
[MSG:INFO: User Digital Output:1 on Pin:gpio.14]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (-562.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.2 Dir:I2SO.1:low Disable:I2SO.0]
[MSG:INFO:     Neg Limit gpio.33]
[MSG:INFO:   Motor1]
[MSG:INFO:     stepstick Step:I2SO.5 Dir:I2SO.4 Disable:I2SO.7]
[MSG:INFO:     Neg Limit gpio.32]
[MSG:INFO: Axis Y (-562.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.10 Dir:I2SO.9 Disable:I2SO.8]
[MSG:INFO:     Neg Limit gpio.35]
[MSG:INFO:   Motor1]
[MSG:INFO:     stepstick Step:I2SO.13 Dir:I2SO.12:low Disable:I2SO.15]
[MSG:INFO:     Neg Limit gpio.34]
[MSG:INFO: Axis Z (-150.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.18 Dir:I2SO.17 Disable:I2SO.16]
[MSG:INFO:     Neg Limit gpio.25]
[MSG:INFO:   Motor1]
[MSG:INFO: feed_hold_pin gpio.36]
[MSG:INFO: cycle_start_pin gpio.2:pu]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Huanyang Spindle Uart Tx:gpio.26 Rx:gpio.16 RTS:gpio.4 Baud:9600]
[MSG:INFO: Laser Ena:NO_PIN Out:gpio.13 Freq:5000Hz Period:8191]
[MSG:INFO: Using spindle Huanyang]
[MSG:INFO: Flood coolant I2SO.24]
[MSG:INFO: Mist coolant gpio.15]
[MSG:INFO: Probe Pin: gpio.39]
[MSG:INFO: WiFi is disabled]
[MSG:INFO: VFD setState:0 SpindleSpeed:0]
```

Here is the session showing the code I used to test it.

```
Grbl 3.6 [FluidNC v3.6.0 (wifi) '$' for help]
[MSG:INFO: '$H'|'$X' to unlock]
$X ; clear the alarm. I don't have switches to home it.
[MSG:INFO: Caution: Unlocked]
ok
M6T1 ;change to the laser tool
[MSG:INFO: Tool No: 1]
[MSG:INFO: Using spindle Laser]
ok
M4S500 ;set the power in M4 mode
ok
G1 X-10 F100 ; do a G1 move to turn on the laser.
ok
```
It worked for me. The laser turned on for the duration of the move.

truglodite, 2022-09-18:
Hmm, not sure how I missed that, or why it worked with 344. I fixed that to speed_map and it works, but with a quirk that wasn't there before. First, Here is the boot message:

```
=~=~=~=~=~=~=~=~=~=~=~= PuTTY log 2022.09.18 11:25:14 =~=~=~=~=~=~=~=~=~=~=~=
[MSG:INFO: FluidNC v3.6.0]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
[MSG:INFO: Local filesystem type is SPIFFS]
[MSG:INFO: Configuration file:config.yaml]
[MSG:INFO: Machine TrugMPCNC]
[MSG:INFO: Board 6-pack]
[MSG:INFO: I2SO BCK:gpio.22 WS:gpio.17 DATA:gpio.21]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN]
[MSG:INFO: Stepping:I2S_static Pulse:4us Dsbl Delay:0us Dir Delay:0us Idle Delay:255ms]
[MSG:INFO: User Digital Output:0 on Pin:I2SO.25]
[MSG:INFO: User Digital Output:1 on Pin:gpio.14]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (-562.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.2 Dir:I2SO.1:low Disable:I2SO.0]
[MSG:INFO:     Neg Limit gpio.33]
[MSG:INFO:   Motor1]
[MSG:INFO:     stepstick Step:I2SO.5 Dir:I2SO.4 Disable:I2SO.7]
[MSG:INFO:     Neg Limit gpio.32]
[MSG:INFO: Axis Y (-562.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.10 Dir:I2SO.9 Disable:I2SO.8]
[MSG:INFO:     Neg Limit gpio.35]
[MSG:INFO:   Motor1]
[MSG:INFO:     stepstick Step:I2SO.13 Dir:I2SO.12:low Disable:I2SO.15]
[MSG:INFO:     Neg Limit gpio.34]
[MSG:INFO: Axis Z (-150.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.18 Dir:I2SO.17 Disable:I2SO.16]
[MSG:INFO:     Neg Limit gpio.25]
[MSG:INFO:   Motor1]
[MSG:INFO: feed_hold_pin gpio.36]
[MSG:INFO: cycle_start_pin gpio.2:pu]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Huanyang Spindle Uart Tx:gpio.26 Rx:gpio.16 RTS:gpio.4 Baud:9600]
[MSG:INFO: Laser Ena:NO_PIN Out:gpio.13 Freq:5000Hz Period:8191]
[MSG:INFO: Using spindle Huanyang]
[MSG:INFO: Flood coolant I2SO.24]
[MSG:INFO: Mist coolant gpio.15]
[MSG:INFO: Probe Pin: gpio.39]
[MSG:INFO: Connecting to STA SSID:xxx]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connected - IP is xxx]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://xxx.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
ok
```

To explain the quirk... I have a macro button in bCNC that runs this gcode script to help me align the laser without burning stuff:

```
 G91
	M6 T1
	M62 P0
	M62 P1
	M3S5
	G1X0Y0
	G90
```

With 344, just hitting the macro button the laser would turn on. Now with 360, after running that script, I have to actually move first before the laser lights up. Incidentally, moving would not get it to light with 360 when I had the "speeds" error in my config. I'm OK with this, as I understand it is a feature to prevent the laser making fire sitting in one spot for a long time. I can just add X-0.1 like Bart to get that macro working the way I want.

So that I think explains this issue... a config error as well as a minor change in behavior. Also Bart, thanks for spotting that silly error in my config... I was wondering why the boot messages had that line about "speeds" but figured I could ignore it since it was working, lol.

closed 2022-09-18 by truglodite

MitchBradley, 2022-09-18:
There is no way to write documentation that someone will not misunderstand.

MitchBradley, 2022-09-18:
We are trying to be consistent with [Grbl 1.1 laser mode](https://github.com/gnea/grbl/wiki/Grbl-v1.1-Laser-Mode#laser-mode-operation). In particular:
[image]

Note that you do not actually have to move; it suffices to simply say "G1" (if a feedrate has already been established) or "G1 F1000" (if not).

MitchBradley, 2022-09-18:
I tried running fluidterm.sh from the posix installer package fluidnc-v3.5.1-pre-FS-posix.zip on a Pi Zero W.  The only problem is that you have to point it at the USB serial port, as in
```
  ./fluidterm.sh /dev/ttyUSB0
```
Otherwise it tends to pick /dev/ttyAMA0, the onboard serial port that is rarely used.

MitchBradley, 2022-09-18:
G0 ?

truglodite, 2022-09-18:
…typo

MitchBradley, 2022-09-18:
I just tried this test on my laser, after a fresh reset:
```
m3s7  (laser did not turn on)
g1f1  (laser turned on)
g0 (laser turned off)
```
Also tried replacing g1f1 with g1x0y0f100 and that worked too, although the gantry moved because the initial position was not 0 0 due to a work coordinate offset.

MitchBradley, 2022-09-18:
It works in g91 mode too.

truglodite, 2022-09-18:
Hmm, what are your “s0 with disable” and “disable with s0” settings? Just noticed my laser off script left pwm on… didn’t notice before opening my box and seeing the led lit up, because I also use the m62/63 p0 to turn on/off a 12v relay that feeds the laser. I had to add ‘g1x0.1’ to get pwm to turn off as well.

MitchBradley, 2022-09-18:
They are defaulted.  My complete laser section is
```yaml
laser:
  output_pin: gpio.2
  speed_map: 0=0% 1000=100%

```

MitchBradley, 2022-09-18:
So false for disable_with_s0 and true for s0_with_disable