bdring/FluidNC #626: Problem:  Laser does not turn off in Hold state

mani-fa, 2022-09-23:
### Controller Board

pen/laser controller
development controller v4.1

### Help From Board Vendor

- [ ] Yes
- [ ] No
- [X] Not Applicable

### Machine Description

pen/laser controller
development controller v4.1

### Input Circuits

```yaml
gpio.17 as 3.3v pwm signal goes to level shift input and gpio.22 as enable pin goes to level shifter OE pin
```

### Configuration file

```yaml
name: "ESP32 Pen/Laser V3"
board: "ESP32 Pen/Laser V3"

stepping:
  engine: RMT
  idle_ms: 250
  dir_delay_us: 1
  pulse_us: 2
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: gpio.13
  
  x:
    steps_per_mm: 80
    max_rate_mm_per_min: 10000
    acceleration_mm_per_sec2: 500
    max_travel_mm: 320
    homing:
      cycle: 2
      mpos_mm: 10
      positive_direction: false
    
    motor0:
      limit_all_pin: gpio.15:low:pu
      stepstick:
        direction_pin: gpio.26
        step_pin: gpio.12
    motor1:
      null_motor:

  y:
    steps_per_mm: 80
    max_rate_mm_per_min: 10000
    acceleration_mm_per_sec2: 500
    max_travel_mm: 240
    homing:
      cycle: 2
      mpos_mm: 10
      positive_direction: false

    motor0:
      limit_all_pin: gpio.4:low:pu
      stepstick:
        direction_pin: gpio.25
        step_pin: gpio.14
    motor1:
      null_motor:

  z:
    steps_per_mm: 400
    max_rate_mm_per_min: 2000
    acceleration_mm_per_sec2: 500
    max_travel_mm: 5.00
    soft_limits: false
    homing:
      cycle: 1
      positive_direction: true
      mpos_mm: 5
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100
    
    motor0:
      rc_servo:
        pwm_hz: 50
        output_pin: gpio.27
        min_pulse_us: 2100
        max_pulse_us: 1000

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  cs_pin: gpio.5
  card_detect_pin: NO_PIN

control:
  safety_door_pin: NO_PIN
  reset_pin: gpio.34:low
  feed_hold_pin: gpio.36:low
  cycle_start_pin: gpio.39:low
  macro0_pin: NO_PIN
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN

coolant:
  flood_pin: NO_PIN
  mist_pin:  NO_PIN
  delay_ms: 0
        
probe:
  pin: NO_PIN

PWM:
  pwm_hz: 5000
  output_pin: gpio.17
  enable_pin: gpio.22
  direction_pin: NO_PIN
  disable_with_s0: false
  s0_with_disable: true
  spinup_ms: 0
  spindown_ms: 0
  tool_num: 0
  speed_map: 0=0% 1000=100%
```

### Startup Messages

```text
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:12812
load:0x40080400,len:3032
entry 0x400805e4

[MSG:INFO: FluidNC v3.6.1]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
[MSG:INFO: Local filesystem type is SPIFFS]
[MSG:INFO: Configuration file:Pen_laser_V3.yaml]
[MSG:INFO: Machine ESP32 Pen/Laser V3]
[MSG:INFO: Board ESP32 Pen/Laser V3]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN]
[MSG:INFO: Stepping:RMT Pulse:2us Dsbl Delay:0us Dir Delay:1us Idle Delay:250ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Shared stepper disable gpio.13]
[MSG:INFO: Axis X (10.000,330.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:gpio.12 Dir:gpio.26 Disable:NO_PIN]
[MSG:INFO:  X All Limit gpio.15:low:pu]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Y (10.000,250.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:gpio.14 Dir:gpio.25 Disable:NO_PIN]
[MSG:INFO:  Y All Limit gpio.4:low:pu]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Z (0.000,5.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     rc_servo Pin:gpio.27 Pulse Len(2100,1000 period:1048575)]
[MSG:INFO: reset_pin gpio.34:low]
[MSG:INFO: feed_hold_pin gpio.36:low]
[MSG:INFO: cycle_start_pin gpio.39:low]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: PWM Spindle Ena:gpio.22 Out:gpio.17 Dir:NO_PIN Freq:5000Hz Period:8191]
[MSG:INFO: Using spindle PWM]
[MSG:INFO: STA SSID is not set]
[MSG:INFO: AP SSID FluidNC IP 192.168.0.1 mask 255.255.255.0 channel 1]
[MSG:INFO: AP started]
[MSG:INFO: WiFi on]
[MSG:INFO: Captive Portal Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]

Grbl 3.6 [FluidNC v3.6.1 (wifi) '$' for help]
[MSG:INFO: '$H'|'$X' to unlock]
```

### User Interface Software

WebUI , LasrGrbl , Lightburn and ...

### What happened?

I have pen/laser controller and development controller, both have the same problem , when I put machine in hold state , laser does not turn off and signal still in pwm pin , my boards have shift level (txs0101 or similar) , with shift level or without it , problem is still present. maybe $32=1 does not work properly. In fluidnc I do not see any setting to put my machine in Laser mode.
pen/laser board has Fluidnc 3.6.1 and development controller has esp32 grbl 1.3a.

bdring, 2022-09-23:
Try changing to a `laser:` spindle and both of these true.

```yaml
  disable_with_s0: true
  s0_with_disable: true
  ```

mani-fa, 2022-09-23:
I am doing it right now

bdring, 2022-09-23:
That level shifter is not ideal for that use. It might be best to force the enable to always be on.

mani-fa, 2022-09-23:
I'm trying to short OE pin with 3.3v to put it on
actually my level shifter is 74lvc1t45 with same pins and instead of OE pin has DIR pin with same function, If I put this pin in High (wire it to 3.3v) , Force level shifter to be always on from A port to B port

mani-fa, 2022-09-23:
I took off esp32 from board and changed Spindle to laser: and disabled enable_pin as well
One Led connecter to GND and gpio17 (pwm 3.3)
In arduino and in laser mode Led goes off in hold state , but here with all changes , LED remain On again

mani-fa, 2022-09-23:
> Try changing to a `laser:` spindle and both of these true.
> 
> ```yaml
>   disable_with_s0: true
>   s0_with_disable: true
> ```

I took off esp32 from board and changed Spindle to laser: and disabled enable_pin as well
for test One Led connecterd to GND and gpio17 (pwm 3.3)
In arduino and in laser mode Led goes off in hold state , but here with all changes , LED remain On again

One more thing , Speed is too slow while pwm signal is changing in gcode, somehow I guess my board not in laser mode

mani-fa, 2022-09-23:
> That level shifter is not ideal for that use. It might be best to force the enable to always be on.

I'm trying to short OE pin with 3.3v to put it on
actually my level shifter is 74lvc1t45 with same pins and instead of OE pin has DIR pin with same function, If I put this pin in High (wire it to 3.3v) , Force level shifter to be always on from A port to B port

MitchBradley, 2022-09-23:
Repost your config file with your latest changes.  Also draw a schematic of your circuit and post that.

mani-fa, 2022-09-23:
Here the Dip file of my design
https://drive.google.com/file/d/1GC8EPwmAtAB5Fo3oDXkHXheeI_PayLlM/view?usp=sharing

and this is My Config File:
https://drive.google.com/file/d/1eK_67Ev7q08GC26T0ssiI6S5gbgcuTzI/view?

MitchBradley, 2022-09-23:
I said the schematic, not the PCB layout.  dch file, not dip file.

mani-fa, 2022-09-23:
ok
Just a sec

mani-fa, 2022-09-23:
Sorry to make it ready
This is a PDF file , sorry

https://drive.google.com/file/d/1ZJlOVuNw6dq01kvEodOJ9JqfTw-0v_qJ/view?usp=sharing

mani-fa, 2022-09-25:
This problem is present in eap32_grbl latest version and someone reported it and problem leaved unsolved.
Is there any solution for it?
my board is just for pen and laser purposes.

MitchBradley, 2022-09-25:
Somehow my earlier reply got lost.  The problem is in your circuit
[image]
When LaserEN is low, the direction is from B to A and the B pin is an input, not an output.  When B is an input, it cannot supply much current, so the PWM5v signal floats to whatever level the laser input pulls it to.  Most lasers have a pullup on such signals, so you must drive them actively low to turn them off.  In your circuit, when LaserEN is low, there is no such active drive.
To fix this you might be able to find a pulldown resistor value that is small enough to overcome the laser's default state yet large enough that it can be overdriven when LaserEN is high and thus makes B an output.
A better solution would be a different circuit.
Another solution would be as Bart suggested, to force LaserEN to be always on.  That could be done by not configuring GPIO22, thus letting it default to an input, and then pulling it up with a resistor so it stays on all the time.

mani-fa, 2022-09-25:
I did test this way:
I disconnect Dir pin from Laser En pin and connect it directly to 3.3v to put in in high state and now It is in active mode always from A as input and B as output.
But again while hold laser has a signal , for example before pause the pwm value was 100 and after pause I see 100 pwm value in output and if it was in 1000 value , after pause I see 1000 pwm in output. Infact my laser acts like spindle
in spindle hold state does not turn the spindle off!! just remain on with pwm value before pausing.

another way to test the issue is connecting one Led to pwm 3.3v output pin in esp32 and other pin of Led goes to GND , in hold state Lef must be off as I test in arduino with grbl1.1 , but here it remain on in it was on before hold and off if it was off before hold state!!

the main issue maybe is another place not in level shifter

mani-fa, 2022-09-25:
> Somehow my earlier reply got lost. The problem is in your circuit [image] When LaserEN is low, the direction is from B to A and the B pin is an input, not an output. When B is an input, it cannot [...]

I did test this way:
I disconnect Dir pin from Laser En pin and connect it directly to 3.3v to put in in high state and now It is in active mode always from A as input and B as output.
But again while hold laser has a signal , for example before pause the pwm value was 100 and after pause I see 100 pwm value in output and if it was in 1000 value , after pause I see 1000 pwm in output. Infact my laser acts like spindle
in spindle hold state does not turn the spindle off!! just remain on with pwm value before pausing.

another way to test the issue is connecting one Led to pwm 3.3v output pin in esp32 and other pin of Led goes to GND , in hold state Lef must be off as I test in arduino with grbl1.1 , but here it remain on in it was on before hold and off if it was off before hold state!!

the main issue maybe is another place not in level shifter

mani-fa, 2022-09-25:
And in my code How can I force gpio22 pin to be always High and not using  Laser En pin anymore

mani-fa, 2022-09-25:
One more thing
I have MKS DLC32 board
with its own frimware laser goes off when I press hold.
I just upload fluidnc to mks and upload config file too.
Now in hold state laser will be on again if it was on before pause!!!!

MitchBradley, 2022-09-25:
One way might be to define GPIO22 as a user output that is high unless explicitly turned on, as in:
```yaml
user_outputs:
   digital0_pin: gpio.22:low
```

MitchBradley, 2022-09-25:
For the MKS board, make a new issue with all the information

bdring, 2022-09-26:
I am generally not motivated respond to questions or statements with multiple exclamation points.

I am able to repeat your problem in M3 mode. The laser does turn off in M4 mode. 

With M3 mode the laser will be on for feed rate controlled motion (G1,M2,M3, etc). It will be off for rapid moves (G0). The laser will stay on after the G1 move until it sees a G0 move. It will also stay on in hold, because it is still in G1 mode.

M4 is the preferred method for lasers. It uses proportional power to speed. It will turn off at the end of G1 moves and hold, because the feed rate is 0.

I have noticed the cycle start is not working after a hold. We will look into that.

We will also discuss the design intent of M3.

mani-fa, 2022-09-26:
Tnx a lot for responding
I am waiting to next releases , I thought It was my mistake in design or configuration part.

closed 2022-09-29 by MitchBradley