bdring/FluidNC #629: Problem: New 4x TMC2209 v1.0 boards report all axes communications check failed

dakoner, 2022-09-24:
### Controller Board

I have 2 https://www.tindie.com/products/33366583/4-axis-tmc2209-cnc-controller/ boards that I just purchased.  I have installed FluidNC 3.6.1 on them, and uploaded this config: https://github.com/bdring/FluidNC/blob/main/example_configs/tmc2209_laser.yaml 
The board is connected to 12V power and the green power light is on.

When I boot the board, I see an error about not being able to communicate with the motors.

### Help From Board Vendor

- [ ] Yes
- [ ] No
- [X] Not Applicable

### Machine Description

It's a microscope using two NEMA11.  I have already got it working with the 6pack controller and TMC2130s.

### Input Circuits

```yaml
No inputs
```

### Configuration file

```yaml
# https://raw.githubusercontent.com/bdring/FluidNC/main/example_configs/tmc2209_laser.yaml (3.6.1)
[MSG:INFO: FluidNC v3.6.1]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
[MSG:INFO: Local filesystem type is SPIFFS]
[MSG:INFO: Configuration file:tmc2209_laser.yaml]
[MSG:INFO: Machine TMC2209 XYYZ Laser]
[MSG:INFO: Board TMC2209 4x DK]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN]
[MSG:INFO: Stepping:RMT Pulse:2us Dsbl Delay:0us Dir Delay:1us Idle Delay:255ms]
[MSG:INFO: Axis count 4]
[MSG:INFO: Shared stepper disable gpio.25]
[MSG:INFO: Axis X (150.000,450.000)]
[MSG:INFO:   Motor0]
[MSG:INFO: Trinamic Stepper Uart Tx:gpio.22 Rx:gpio.21 RTS:NO_PIN Baud:115200]
[MSG:INFO:     tmc_2209 Step:gpio.26 Dir:gpio.27 Disable:NO_PIN Addr:0 R:0.110]
[MSG:INFO:  X Pos Limit gpio.35]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Y (150.000,450.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_2209 Step:gpio.33 Dir:gpio.32 Disable:NO_PIN Addr:1 R:0.110]
[MSG:INFO:  Y Pos Limit gpio.34]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Z (150.000,450.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_2209 Step:gpio.2 Dir:gpio.14 Disable:NO_PIN Addr:2 R:0.110]
[MSG:INFO:  Z Pos Limit gpio.39]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis A (150.000,450.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_2209 Step:gpio.16 Dir:gpio.15 Disable:NO_PIN Addr:3 R:0.110]
[MSG:INFO:  A Pos Limit gpio.36]
[MSG:INFO:   Motor1]
[MSG:INFO: X Axis communications check failed]
[MSG:INFO: Y Axis communications check failed]
[MSG:INFO: Z Axis communications check failed]
[MSG:INFO: A Axis communications check failed]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Laser Ena:gpio.12 Out:gpio.4 Freq:5000Hz Period:8191]
[MSG:INFO: Using spindle Laser]
[MSG:INFO: Connecting to STA SSID:artdeco]
[MSG:INFO: Connecting.]
[MSG:INFO: Connected - IP is 192.168.1.227]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://4pack.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
```

### Startup Messages

```text
`
[MSG:INFO: FluidNC v3.6.1]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
[MSG:INFO: Local filesystem type is SPIFFS]
[MSG:INFO: Configuration file:tmc2209_laser.yaml]
[MSG:INFO: Machine TMC2209 XYYZ Laser]
[MSG:INFO: Board TMC2209 4x DK]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN]
[MSG:INFO: Stepping:RMT Pulse:2us Dsbl Delay:0us Dir Delay:1us Idle Delay:255ms]
[MSG:INFO: Axis count 4]
[MSG:INFO: Shared stepper disable gpio.25]
[MSG:INFO: Axis X (150.000,450.000)]
[MSG:INFO:   Motor0]
[MSG:INFO: Trinamic Stepper Uart Tx:gpio.22 Rx:gpio.21 RTS:NO_PIN Baud:115200]
[MSG:INFO:     tmc_2209 Step:gpio.26 Dir:gpio.27 Disable:NO_PIN Addr:0 R:0.110]
[MSG:INFO:  X Pos Limit gpio.35]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Y (150.000,450.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_2209 Step:gpio.33 Dir:gpio.32 Disable:NO_PIN Addr:1 R:0.110]
[MSG:INFO:  Y Pos Limit gpio.34]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Z (150.000,450.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_2209 Step:gpio.2 Dir:gpio.14 Disable:NO_PIN Addr:2 R:0.110]
[MSG:INFO:  Z Pos Limit gpio.39]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis A (150.000,450.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_2209 Step:gpio.16 Dir:gpio.15 Disable:NO_PIN Addr:3 R:0.110]
[MSG:INFO:  A Pos Limit gpio.36]
[MSG:INFO:   Motor1]
[MSG:INFO: X Axis communications check failed]
[MSG:INFO: Y Axis communications check failed]
[MSG:INFO: Z Axis communications check failed]
[MSG:INFO: A Axis communications check failed]`
```

### User Interface Software

fluidterm

### What happened?

I purchased the board, flashed FluidNC 3.6.1 wifi and fs, then uploaded an example config and looked at the serial output.

I was expecting this config, or one of the other configs in fluidnc/example_configs, to work out of the box and be able to communicate with the motors.

I have exactly one motor connected, to the X axis.  Removing or connecting motors doesn't affect the communications issue,.

MitchBradley, 2022-09-24:
The "configuration file" that you posted is the startup messages.  We need the .yaml file - renamed to .txt so GitHub will accept it.

dakoner, 2022-09-24:
[tmc2209_laser.txt](https://github.com/bdring/FluidNC/files/9639186/tmc2209_laser.txt)

bdring, 2022-09-24:
Ideally the file is posted in a yaml code block.

http://wiki.fluidnc.com/en/support/requesting_help#pasting-the-config-file-on-github

dakoner, 2022-09-24:
`name: "TMC2209 XYYZ Laser"
board: "TMC2209 4x DK"

stepping:
  engine: RMT
  idle_ms: 255
  dir_delay_us: 1
  pulse_us: 2
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: gpio.25:high
  
  x:
    steps_per_mm: 800.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 300.000
    soft_limits: false
    homing:
      cycle: 1
      positive_direction: false
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: gpio.35
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      tmc_2209:
        uart:
          txd_pin: gpio.22
          rxd_pin: gpio.21
          rts_pin: NO_PIN
          cts_pin: NO_PIN
          baud: 115200
          mode: 8N1

        addr: 0
        r_sense_ohms: 0.110
        run_amps: 1.200
        hold_amps: 0.500
        microsteps: 8
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: StealthChop
        homing_mode: StealthChop
        use_enable: false
        step_pin: gpio.26
        direction_pin: gpio.27
        disable_pin: NO_PIN

    motor1:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      null_motor:

  y:
    steps_per_mm: 800.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 300.000
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: gpio.34
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      tmc_2209:
        addr: 1
        r_sense_ohms: 0.110
        run_amps: 1.200
        hold_amps: 0.500
        microsteps: 8
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: StealthChop
        homing_mode: StealthChop
        use_enable: false
        step_pin: gpio.33
        direction_pin: gpio.32
        disable_pin: NO_PIN

    motor1:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      null_motor:

  z:
    steps_per_mm: 800.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 300.000
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: gpio.39
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      tmc_2209:
        addr: 2
        r_sense_ohms: 0.110
        run_amps: 1.200
        hold_amps: 0.500
        microsteps: 8
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: StealthChop
        homing_mode: StealthChop
        use_enable: false
        step_pin: gpio.2
        direction_pin: gpio.14
        disable_pin: NO_PIN

    motor1:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      null_motor:

  a:
    steps_per_mm: 800.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 300.000
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: gpio.36
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      tmc_2209:
        addr: 3
        r_sense_ohms: 0.110
        run_amps: 1.200
        hold_amps: 0.500
        microsteps: 8
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: StealthChop
        homing_mode: StealthChop
        use_enable: false
        step_pin: gpio.16
        direction_pin: gpio.15
        disable_pin: NO_PIN

    motor1:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      null_motor:

      
spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  card_detect_pin: NO_PIN
  cs_pin: gpio.5

probe:
  pin: NO_PIN

start:
  must_home: false

Laser:
  pwm_hz: 5000
  output_pin: gpio.4
  enable_pin: gpio.12
  disable_with_s0: false
  s0_with_disable: true
  tool_num: 100
  speed_map: 0=0.000% 255=100.000%
`

dakoner, 2022-09-24:
Uploaded the YAML config file- it's straight out of the fluidnc example_configs dir.

bdring, 2022-09-24:
That config file was for a different controller.
https://www.tindie.com/products/33366583/tmc2209-4-axis-grbl_esp32-cnc-controller/

You controller is documented here...

https://oshwlab.com/bdring/tmc2130-2-axis_copy_copy
http://wiki.fluidnc.com/en/hardware/official/TMC2209_4_Axis

dakoner, 2022-09-24:
Thanks for the pointer.  I tried logging into the OSHW site but it just always pops me back to not logged in.

I tried the config at http://wiki.fluidnc.com/en/hardware/official/TMC2209_4_Axis
and got an error so I added
```

i2so:
  bck_pin: gpio.22
  data_pin: gpio.21
  ws_pin: gpio.17
```

and now the board works.

Here was the error without the i2so:
```
[MSG:INFO: Configuration file:tmc2209_new.yaml]

Backtrace:0x400F00A1:0x3FFB23600x400F0E6F:0x3FFB24E0 0x400D4029:0x3FFB2510 0x400DCF91:0x3FFB2560 0x400DD93D:0x3FFB25A0 0x400DDE2F:0x3FFB2620 0x400DDF61:0x3FFB27B0 0x400DEB9B:0x3FFB27E0 0x4010DB3E:0x3FFB2820

[MSG:ERR: Initialization error at /stepping: I2SO bus must be configured for this stepping type]
[MSG:INFO: Configuration is invalid]
[MSG:INFO: Using default configuration]
[MSG:INFO: Axes: using defaults]

Backtrace:0x400F00A1:0x3FFB24F00x400F0E6F:0x3FFB2670 0x400D4029:0x3FFB26A0 0x400DCF91:0x3FFB26F0 0x400DD93D:0x3FFB2730 0x400DDF99:0x3FFB27B0 0x400DEB9B:0x3FFB27E0 0x4010DB3E:0x3FFB2820

[MSG:ERR: Initialization error at /stepping: I2SO bus must be configured for this stepping type]
[MSG:INFO: Configuration is invalid]
[MSG:INFO: Connecting to STA SSID:artdeco]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connected - IP is 192.168.1.227]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://4pack.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]

Grbl 3.6 [FluidNC v3.6.1 (wifi) '$' for help]
[MSG:INFO: Configuration is invalid. Check boot messages for ERR's.]
```

The config I'm using is copied straight from the page http://wiki.fluidnc.com/en/hardware/official/TMC2209_4_Axis

```
stepping:
  engine: I2S_STREAM
  idle_ms: 255
  dir_delay_us: 1
  pulse_us: 4
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: NO_PIN

  x:
    steps_per_mm: 53.33
    max_rate_mm_per_min: 18000
    acceleration_mm_per_sec2: 950
    max_travel_mm: 750
    homing:
      cycle: 1
      mpos_mm: 0
      positive_direction: false
      seek_mm_per_min: 3000
      feed_mm_per_min: 240

    motor0:
      limit_neg_pin: gpio.36:low
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      tmc_2209:
        uart:
          txd_pin: gpio.16
          rxd_pin: gpio.4
          rts_pin: NO_PIN
          cts_pin: NO_PIN
          baud: 115200
          mode: 8N1

        addr: 0
        r_sense_ohms: 0.110
        run_amps: 1.000
        hold_amps: 0.500
        microsteps: 4
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: CoolStep
        homing_mode: CoolStep
        use_enable: false
        direction_pin: i2so.1
        step_pin: i2so.2
        disable_pin: i2so.0

  y:
    steps_per_mm: 53.33
    max_rate_mm_per_min: 18000
    acceleration_mm_per_sec2: 950
    max_travel_mm: 410
    homing:
      cycle: 1
      mpos_mm: 0
      positive_direction: false
      seek_mm_per_min: 3000
      feed_mm_per_min: 240

    motor0:
      limit_neg_pin: gpio.39:low
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      tmc_2209:
        addr: 1
        r_sense_ohms: 0.110
        run_amps: 1.000
        hold_amps: 0.500
        microsteps: 16
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: StealthChop
        homing_mode: StealthChop
        use_enable: false
        direction_pin: i2so.4
        step_pin: i2so.5
        disable_pin: i2so.7

  z:
    steps_per_mm: 400
    max_rate_mm_per_min: 200
    acceleration_mm_per_sec2: 25
    max_travel_mm: 200
    homing:
      cycle: 2
      mpos_mm: 0
      positive_direction: false
      seek_mm_per_min: 200
      feed_mm_per_min: 100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: gpio.34:low
      limit_all_pin: NO_PIN
      tmc_2209:
        addr: 2
        r_sense_ohms: 0.110
        run_amps: 1.000
        hold_amps: 0.500
        microsteps: 16
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: StealthChop
        homing_mode: StealthChop
        use_enable: false
        direction_pin: i2so.9
        step_pin: i2so.10
        disable_pin: i2so.8

  a:
    steps_per_mm: 400
    max_rate_mm_per_min: 200
    acceleration_mm_per_sec2: 25
    max_travel_mm: 200
    homing:
      cycle: 2
      mpos_mm: 0
      positive_direction: false
      seek_mm_per_min: 200
      feed_mm_per_min: 100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: gpio.35:low
      limit_all_pin: NO_PIN
      tmc_2209:
        addr: 3
        r_sense_ohms: 0.110
        run_amps: 1.000
        hold_amps: 0.500
        microsteps: 16
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: StealthChop
        homing_mode: StealthChop
        use_enable: false
        direction_pin: i2so.12
        step_pin: i2so.13
        disable_pin: i2so.15
 
```

dakoner, 2022-09-24:
Could probably save time for noobs like me if there's a well-named file in the fluidnc configs that is the same as the motor section in http://wiki.fluidnc.com/en/hardware/official/TMC2209_4_Axis but with the i2s config and a board name added.

closed 2022-09-24 by dakoner