bdring/FluidNC #631: Repurposing FluidNC for telescope control

Cyrille-de-Brebisson, 2022-09-25:
### Machine Context

Hello,

I am trying to see if I can repurpose FluidNC to control a telescope mount.
A telescope is a 2 axis system (with a 3rd motor used for focus).
I am interested in being able to use a cheap, ready made, HW board with a LCD to control my telesope...

### Feature Description

I am an experienced SW engineer (embeded, C/C++ and app development)...
BUT, I have no experience wit VS Code or platform IO :-(.

What I most would like to understand is where the GUI that gets displayed on the screen is managed/handled so that I can start "hacking" at he existing code and see if I can get anywhere from there...

Can you give me some "pointers" "here where/how to get started" word of advice? I would be "eternally grateful" :-)

### Other Approaches

Ultimately, if I am successful, I would do the following:
- reuse the low level libraries/setup from FluidNC (OS, coms, stepper control, graphical stuff)
- recode a different UI on top which would allow the user to:
  - configure his mount
  - see the night sky (start map)
  - control the telescope pointing
  

### How I Can Help

well, I don't know exactly what U can do, but I am sure that I can help!!!

Thanks

MitchBradley, 2022-09-25:
There is no real need to use VSCode; it is just a convenient editor with good platformio integration.  Platformio is required, since it is the build system, akin to "make".  It controls which files are compiled, which external libraries and frameworks to use, etc.  Platformio differs from most other build systems by having very good features for installing toolchains, SDKs, frameworks, and libraries - steps that other build systems either do not automate or only do so with a lot of tedious hand-written rules.  Platformio is a command-line application written in Python.  It has one main command "pio run" which performs the compilation, akin to typing "make" to do a compilation.  Many editors like VSCode, Atom, and whatnot have varying degrees of platformio integration so you fire off the "pio run" by poking at a button, but it is entirely feasible to use an old-school editor like vi and then type "pio run" in a command window.

Regarding your actual question, the FluidNC UI is ESP3D_WebUI which is a separate project.  We maintain [a fork](https://github.com/MitchBradley/ESP3D-WEBUI/tree/relayout) for various reasons, but the fundamental code is in [another repo](https://github.com/luc-github/ESP3D-WEBUI).  WebUI is written javascript / HTML.  Its source is combined into one file and minified into a single "index.html.gz" file.  That file is stored in a file that resides on the FLASH filesystem on the ESP32 module.  FluidNC includes simple web server code and has builtin WiFi support.  From an external browser, either on a PC or smartphone or tablet, you can contact that web server, which will return the index.html.gz file and it will run in the browser, thus running the UI code.

We support a couple of small OLED screens that can be directly attached to the FluidNC controller, but they are only for information display, not full UI with input capabilities.  Various people have written LCD-based UIs that run on separate microprocessors - such as another ESP32 module - communicating with FluidNC over WiFi or serial.  Given the low cost of an MCU and the modularity advantages of having the UI code run on a separate processor, we recommend that approach and do not plan to integrate LCD UI code into the main FluidNC code structure.

Cyrille-de-Brebisson, 2022-09-26:
Hello,

Thanks for all this info!
So, if I understand well, if I wanted to use something like this board, which comes with a LCD https://fr.aliexpress.com/item/1005004255629268.html

The LCD is in fact a full "system", which has a web browser and gets the data from the main module through serial (rather than HTTP, but who cares)...

So, if I wanted to make a telescope control, what I would really need to do is look at that ESP3D_WebUI project and see how I can add a new type of UI which would handle a telescope (I assume that you already have configurations for cnc, laser, 3D printers...)

Is this relatively accurate?

Cyrille

MitchBradley, 2022-09-26:
You could use a 10" ($65) or 7" ($35) Android tablet and add a UI panel that is optimized for telescopes.  The TabletUI tab is an example of such a customized UI; I optimized it for the stuff I do all the time in the production shop, eliminating small hard-to-touch controls that I never use for job running.

MitchBradley, 2022-09-26:
Regarding the board that you mention, its source code was adapted from our older Grbl_Esp32 project and, in violation of the GPL license, they did not publish their mods anywhere that I have seen.

Cyrille-de-Brebisson, 2022-09-27:
Thanks for the info. I will look into it.

Sorry for the guys taking your work improperly. I have seen it happen also on my end and it is very frustrating!

Cyrille

MitchBradley, 2022-10-03:
I think that the questions have been answered.

closed 2022-10-03 by MitchBradley