bdring/FluidNC #633: Problem:  Feed and speed override not responding

LilBoiAlvin, 2022-09-26:
### Controller Board

MKS DLC32

### Help From Board Vendor

- [ ] Yes
- [ ] No
- [X] Not Applicable

### Machine Description

Gantry router DM556 stepper drives.

### Configuration file

```yaml
board: MKS-DLC32 V2.1
name: Router
meta: (01.01.2022) by Skorpi

kinematics:
  Cartesian:

stepping:
  engine: I2S_STATIC
  idle_ms: 0
  pulse_us: 10
  dir_delay_us: 1
  disable_delay_us: 0
axes:
  shared_stepper_disable_pin: I2SO.0
  x:
    steps_per_mm: 157.750
    max_rate_mm_per_min: 12500
    acceleration_mm_per_sec2: 500
    max_travel_mm: 100000
    soft_limits: true
    homing:
      cycle: 1
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 300.000
      seek_mm_per_min: 5000.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.36
      hard_limits: false
      pulloff_mm: 2.000
      stepstick:
        step_pin: I2SO.1
        direction_pin: I2SO.2

  y:
    steps_per_mm: 157.750
    max_rate_mm_per_min: 20000
    acceleration_mm_per_sec2: 30
    max_travel_mm: 100000
    soft_limits: true
    homing:
      cycle: 1
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 300.000
      seek_mm_per_min: 5000.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.35
      hard_limits: false
      pulloff_mm: 2.000
      stepstick:
        step_pin: I2SO.5
        direction_pin: I2SO.6:low

  z:
    steps_per_mm: 157.750
    max_rate_mm_per_min: 12000.000
    acceleration_mm_per_sec2: 500.000
    max_travel_mm: 80.000
    soft_limits: true
    homing:
      cycle: 0
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 300.000
      seek_mm_per_min: 1000.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.34
      hard_limits: false
      pulloff_mm: 1.000
      stepstick:
        step_pin: I2SO.3
        direction_pin: I2SO.4

i2so:
  bck_pin: gpio.16
  data_pin: gpio.21
  ws_pin: gpio.17

spi:
  miso_pin: gpio.12
  mosi_pin: gpio.13
  sck_pin: gpio.14

sdcard:
  cs_pin: gpio.15
  card_detect_pin: NO_PIN

control:
  safety_door_pin: NO_PIN
  reset_pin: NO_PIN
  feed_hold_pin:  gpio.18:low:pu
  cycle_start_pin: gpio.23:low:pu
  macro0_pin: gpio.33:low:pu
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN

macros:
  startup_line0:
  startup_line1:
  macro0: $SD/Run=lasertest.gcode
  macro1: $SD/Run=home.gcode
  macro2:
  macro3:

coolant:
  flood_pin: NO_PIN
  mist_pin: NO_PIN
  delay_ms: 0

probe:
  pin: gpio.22
  check_mode_start: true

user_outputs:
  analog0_pin: NO_PIN
  analog1_pin: NO_PIN
  analog2_pin: NO_PIN
  analog3_pin: NO_PIN
  analog0_hz: 5000
  analog1_hz: 5000
  analog2_hz: 5000
  analog3_hz: 5000
  digital0_pin: NO_PIN
  digital1_pin: NO_PIN
  digital2_pin: NO_PIN
  digital3_pin: NO_PIN
  

PWM:
  pwm_hz: 5000
  output_pin: gpio.26
  enable_pin: gpio.4
  direction_pin: NO_PIN
  disable_with_s0: false
  s0_with_disable: true
  spinup_ms: 1000
  spindown_ms: 1000
  tool_num: 0
  speed_map: 0=0.000% 1000=100.000%

start:
  must_home: false

# 5,18,19,22,23,25,26,27,32,33,39,I2SO.7
# SDA 0 / SCL 4
```

### Startup Messages

```text
No errors
```

### User Interface Software

WebUI

### What happened?

Used the WebUI and tried using the override features but there is no response. Do I have to add something in the Configuration file?

bdring, 2022-09-27:
Works for me.

I tested this with FluidTerm. Ctrl+O allows you to send overrides

I also tested with the WebUI.

What commands did you send? Try sending a slow move like G1 X100 F100 and then try the overrides. The webui overrides only work for feedrate based moves like G1,G2, G3.

LilBoiAlvin, 2022-09-27:
> Works for me.
> 
> I tested this with FluidTerm. Ctrl+O allows you to send overrides
> 
> I also tested with the WebUI.
> 
> What commands did you send? Try sending a slow move like G1 X100 F100 and then try the overrides. The webui overrides only work for feedrate based moves like G1,G2, G3.

Thanks for the help, the terminal works perfectly, however the WebUI still doesn't have any response on feed change. The G-code is all G1. Is there anything else I could try?

MitchBradley, 2022-09-27:
I just tried WebUI feed override and it worked fine.  In the Commands / Send Command area I typed `G91 G1 X100 F100` to start running a slow move that will take a minute to complete.  I then clicked on the F10% override 
[image]
 and the motion sped up a little.  Repeated clicks continued to increase the speed until it maxed out at 200% of the original rate, as designed, and the operation completed in about half the time that it would otherwise have taken.

Please use the above description as an example of the level of descriptive detail that we need in order to solve problems.  "The WebUI still doesn't have any response on feed change" does not tell us exactly what you typed or clicked, what you expected to observe, and what you actually observed, so we are unable to detect misinterpretations of how things are supposed to work or replicate the exact of operations that might trigger a problem.

MitchBradley, 2022-09-27:
Another problem is that you did not include the startup messages.  "No errors" tells us what is lacking but not what is present.  Notably, the very first startup message gives the firmware version.  That is very important because some problems only show up in some versions.  If there happens to be a problem with overrides that only applies to one version, and we are testing with a different version than you, that wastes our time and yours.

LilBoiAlvin, 2022-09-27:
> I just tried WebUI feed override and it worked fine. In the Commands / Send Command area I typed `G91 G1 X100 F100` to start running a slow move that will take a minute to complete. I then clicked [...]

Thank you for your help, your sample g-code works perfectly. However when I try to run the following g-code, I am still not able to use the feed over-rides.

(Exported by FreeCAD)
(Post Processor: grbl_post)
(Output Time:2022-09-27 00:03:52.219848)
(Begin preamble)
G17 G90
G21 G54
(Begin operation: G54)
(Path: G54)
(Finish operation: G54)
(Begin operation: TC: Centre_Drill_3mm)
(Path: TC: Centre_Drill_3mm)
(TC: Centre_Drill_3mm)
(Begin toolchange)
( M6 T1 )
(Finish operation: TC: Centre_Drill_3mm)
(Begin operation: Drilling)
(Path: Drilling)
(Drilling)
(Begin Drilling)
G0 Z5.000
G90
G0 X10.000 Y10.000
G0 Z3.000
(G81 X10.000 Y10.000 Z-5.000 F70.000 R3.000)
G0 X10.000 Y10.000
G1 Z-5.000 F70.00
G0 Z3.000
G0 X40.000 Y10.000
G0 Z3.000
(G81 X40.000 Y10.000 Z-5.000 F70.000 R3.000)
G0 X40.000 Y10.000
G1 Z-5.000 F70.00
G0 Z3.000
G0 X40.000 Y55.000
G0 Z3.000
(G81 X40.000 Y55.000 Z-5.000 F70.000 R3.000)
G0 X40.000 Y55.000
G1 Z-5.000 F70.00
G0 Z3.000
G0 X10.000 Y55.000
G0 Z3.000
(G81 X10.000 Y55.000 Z-5.000 F70.000 R3.000)
G0 X10.000 Y55.000
G1 Z-5.000 F70.00
G0 Z3.000
G0 X20.000 Y78.000
G0 Z3.000
(G81 X20.000 Y78.000 Z-5.000 F70.000 R3.000)
G0 X20.000 Y78.000
G1 Z-5.000 F70.00
G0 Z3.000
G0 X42.500 Y108.000
G0 Z3.000
(G81 X42.500 Y108.000 Z-5.000 F70.000 R3.000)
G0 X42.500 Y108.000
G1 Z-5.000 F70.00
G0 Z3.000
G0 X60.000 Y124.000
G0 Z3.000
(G81 X60.000 Y124.000 Z-5.000 F70.000 R3.000)
G0 X60.000 Y124.000
G1 Z-5.000 F70.00
G0 Z3.000
G0 X60.000 Y140.000
G0 Z3.000
(G81 X60.000 Y140.000 Z-5.000 F70.000 R3.000)
[... 743 lines cut ...]
[MSG:INFO:     Neg Limit gpio.36]
[MSG:INFO: Axis Y (0.000,100000.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.5 Dir:I2SO.6:low Disable:NO_PIN]
[MSG:INFO:     Neg Limit gpio.35]
[MSG:INFO: Axis Z (0.000,10000.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.3 Dir:I2SO.4 Disable:NO_PIN]
[MSG:INFO:     Neg Limit gpio.34]
[MSG:INFO: feed_hold_pin gpio.18:low:pu]
[MSG:INFO: cycle_start_pin gpio.23:low:pu]
[MSG:INFO: macro0_pin gpio.33:low:pu]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: PWM Spindle Ena:gpio.4 Out:gpio.26 Dir:NO_PIN Freq:5000Hz Period:8191]
[MSG:INFO: Using spindle PWM]
[MSG:INFO: Probe Pin: gpio.22]
[MSG:INFO: Connecting to STA SSID:iiNet68B43D_EXT]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connected - IP is 10.1.1.251]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]

bdring, 2022-09-29:
This should be fixed in v3.6.2.
If not, please start a new issue.

closed 2022-09-29 by bdring