bdring/FluidNC #641: Y-axis not working after update v3.6.2

arlimbad222, 2022-09-30:
### Controller Board

Custom board

### Help From Board Vendor

- [ ] Yes
- [ ] No
- [X] Not Applicable

### Machine Description

4-axis cnc milling machine

### Configuration file

```yaml
name: "4-axis MACHINE"
board: "4-axis MACHINE"

stepping:
  engine: RMT
  idle_ms: 250
  dir_delay_us: 1
  pulse_us: 2
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: NO_PIN
  
  x:
    steps_per_mm: 100
    max_rate_mm_per_min: 10000
    acceleration_mm_per_sec2: 300
    max_travel_mm: 317
    soft_limits: true
    homing:
      cycle: 2
      allow_single_axis: true
      mpos_mm: -2
      positive_direction: true
      feed_mm_per_min: 200
      seek_mm_per_min: 1500
      settle_ms: 250
      seek_scaler: 1
      feed_scaler: 1
    
    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: gpio.16:low:pu
      hard_limits: false
      pulloff_mm: 2.000
      stepstick:
        direction_pin: gpio.25
        step_pin: gpio.18
    motor1:
      null_motor:

  y:
    steps_per_mm: 100
    max_rate_mm_per_min: 10000
    acceleration_mm_per_sec2: 300
    max_travel_mm: 250
    soft_limits: true
    homing:
      cycle: 2
      allow_single_axis: true
      mpos_mm: -2
      positive_direction: true
      feed_mm_per_min: 200
      seek_mm_per_min: 1500
      settle_ms: 250
      seek_scaler: 1
      feed_scaler: 1

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: gpio.17:low:pu
      hard_limits: false
      pulloff_mm: 2.000
      stepstick:
        direction_pin: gpio.15
        step_pin: gpio.26
    motor1:
      null_motor:

  z:
    steps_per_mm: 100
    max_rate_mm_per_min: 6000
    acceleration_mm_per_sec2: 300
    max_travel_mm: 140
    soft_limits: true
    homing:
      cycle: 1
      allow_single_axis: true
      mpos_mm: -2
      positive_direction: true
      feed_mm_per_min: 200
      seek_mm_per_min: 1500
      settle_ms: 250
      seek_scaler: 1
      feed_scaler: 1
      

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: gpio.4:low:pu
      hard_limits: false
      pulloff_mm: 2.000
      stepstick:
        direction_pin: gpio.33
        step_pin: gpio.27
    motor1:
      null_motor:

  a:
    steps_per_mm: 8.889
    max_rate_mm_per_min: 30000
    acceleration_mm_per_sec2: 20000
    max_travel_mm: 0

    motor0:
      limit_all_pin: NO_PIN
      stepstick:
        direction_pin: gpio.13
        step_pin: gpio.12
    motor1:
      null_motor:        

coolant:
  flood_pin: gpio.21
  mist_pin:  gpio.23

        
probe:
  pin: gpio.32:low:pu

control:
  safety_door_pin: NO_PIN
  reset_pin: NO_PIN
  feed_hold_pin: gpio.5:low:pu
  cycle_start_pin: gpio.19:low:pu
  macro0_pin: NO_PIN
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN

PWM:
  pwm_hz: 5000
  output_pin: gpio.2
  enable_pin: gpio.22
  direction_pin: NO_PIN
  disable_with_s0: false
  s0_with_disable: true
  spinup_ms: 0
  spindown_ms: 0
  tool_num: 0
  speed_map: 0=0% 18000=100%
```

### Startup Messages

```text
Grbl 3.6 [FluidNC v3.6.2 (bt) '$' for help]
[MSG:INFO: '$H'|'$X' to unlock]
$SS
[MSG:INFO: FluidNC v3.6.2]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
[MSG:INFO: Local filesystem type is spiffs]
[MSG:INFO: Configuration file:SHRIKE_V1.YAML]
[MSG:WARN: max_travel_mm value 0.000 constrained to range (0.100,10000000.000)]
[MSG:INFO: Machine 4-axis MACHINE]
[MSG:INFO: Board 4-axis MACHINE]
[MSG:INFO: SPI not defined]
[MSG:INFO: No SD Card CS Pin]
[MSG:INFO: See http://wiki.fluidnc.com/en/config/sd_card#sdfallbackcs-access-sd-without-a-config-file]
[MSG:INFO: Stepping:RMT Pulse:2us Dsbl Delay:0us Dir Delay:1us Idle Delay:250ms]
[MSG:INFO: Axis count 4]
[MSG:INFO: Axis X (-319.000,-2.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:gpio.18 Dir:gpio.25 Disable:NO_PIN]
[MSG:INFO:  X All Limit gpio.16:low:pu]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Y (-252.000,-2.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:gpio.26 Dir:gpio.15 Disable:NO_PIN]
[MSG:INFO:  Y All Limit gpio.17:low:pu]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Z (-142.000,-2.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:gpio.27 Dir:gpio.33 Disable:NO_PIN]
[MSG:INFO:  Z All Limit gpio.4:low:pu]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis A (-0.100,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:gpio.12 Dir:gpio.13 Disable:NO_PIN]
[MSG:INFO:   Motor1]
[MSG:INFO: feed_hold_pin gpio.5:low:pu]
[MSG:INFO: cycle_start_pin gpio.19:low:pu]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: PWM Spindle Ena:gpio.22 Out:gpio.2 Dir:NO_PIN Freq:5000Hz Period:8191]
[MSG:INFO: Using spindle PWM]
[MSG:INFO: Flood coolant gpio.21]
[MSG:INFO: Mist coolant gpio.23]
[MSG:INFO: Probe Pin: gpio.32:low:pu]
[MSG:INFO: BT Started with FluidNC]
ok
```

### User Interface Software

UGS

### What happened?

I just updated the #fluidnc version from 3.6.1 to 3.6.2 because I face some resume problems and suddenly the y-axis stop moving I just erase esp and went back to 3.6.1, it's not about hardware because everything running well with V3.6.1, I am hardware guy I can't write code that's why I am here.

thank you for such a nice firmware.

MitchBradley, 2022-09-30:
What happens if you set idle_ms to 255?  Does the y motor lock? What operation are you using to test the y motor? Jogging? Explicit gcode commands?

arlimbad222, 2022-10-01:
I have one dedicated Machine for testing the codes. when I get the problem I just switch pulse direction wires to x and when I move x the y axis moves so it is not motor or driver , because everything is fine on x axis pulse direction wire.

I also checked Y-axis direction pin by multimeter and it works fine when I jog positive it goes to 0v when I jog negative it goes to 5v so now it all about pulse pin.

I tried to jog , home , program run but all the things failed in V3.6.2 so I switch to V3.6.1 and all the axis works fine.

MitchBradley, 2022-10-01:
Try defining the Y limit pin as NO_PIN.  That will tell us whether the problem is related to limits.  Also send `$message/level=debug` and reboot, looking for messages about limits.

arlimbad222, 2022-10-01:
I will try it by tomorrow

For the limit errors i design controller and i use pc817 optocouplers with each  input and every input has own small led so if there is error in input pins the led was blink but i don't see any blinking led on my controller.

It's just for an information i will test the steps you suggest tomorrow and update here.

Thankyou for quick response.

MitchBradley, 2022-10-01:
One thing I noticed when studying your config file - you have the _scaler values set to 1.  They should be larger than one, giving a little extra room to find switches during homing.  The default value of 1.1 is usually okay.

This does not explain your Y axis problem, but I am just mentioning it while it is fresh in my mind.

arlimbad222, 2022-10-02:
Hello @MitchBradley 
I try few tests today i explain step by step here 

1st - test  -  (I Set Y axis - limit_all_pin: NO_PIN )
Result  -  (now the Y-axis moving correctly but Z-axis stop moving & I also send $message/level=debug here is the message
$ss
[MSG:INFO: FluidNC v3.6.2]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
[MSG:INFO: Local filesystem type is spiffs]
[MSG:INFO: Configuration file:SHRIKE_V1.YAML]
[MSG:WARN: max_travel_mm value 0.000 constrained to range (0.100,10000000.000)]
[MSG:DBG: Running after-parse tasks]
[MSG:DBG: Checking configuration]
[MSG:INFO: Machine SHRIKE MACHINE]
[MSG:INFO: Board SHRIKE MACHINE]
[MSG:INFO: SPI not defined]
[MSG:INFO: No SD Card CS Pin]
[MSG:INFO: See http://wiki.fluidnc.com/en/config/sd_card#sdfallbackcs-access-sd-without-a-config-file]
[MSG:INFO: Stepping:RMT Pulse:2us Dsbl Delay:0us Dir Delay:1us Idle Delay:250ms]
[MSG:INFO: Axis count 4]
[MSG:INFO: Axis X (-317.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:gpio.18 Dir:gpio.25 Disable:NO_PIN]
[MSG:INFO:  X All Limit gpio.16:low:pu]
[MSG:DBG: Updating  X All Limit]
[MSG:DBG:  X All Limit 0]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Y (-250.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:gpio.26 Dir:gpio.15 Disable:NO_PIN]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Z (-140.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:gpio.27 Dir:gpio.33 Disable:NO_PIN]
[MSG:INFO:  Z All Limit gpio.4:low:pu]
[MSG:DBG: Updating  Z All Limit]
[MSG:DBG:  Z All Limit 0]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis A (-0.100,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:gpio.12 Dir:gpio.13 Disable:NO_PIN]
[MSG:INFO:   Motor1]
[MSG:INFO: feed_hold_pin gpio.5:low:pu]
[MSG:INFO: cycle_start_pin gpio.19:low:pu]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: PWM Spindle Ena:gpio.22 Out:gpio.2 Dir:NO_PIN Freq:5000Hz Period:8191]
[MSG:INFO: Using spindle PWM]
[MSG:INFO: Flood coolant gpio.21]
[MSG:INFO: Mist coolant gpio.23]
[MSG:INFO: Probe Pin: gpio.32:low:pu]
[MSG:DBG: Begin Bluetooth setup]
[MSG:INFO: BT Started with FluidNC] )

2nd test  -  (I Set X,Y,Z axis - limit_all_pin: NO_PIN )
Result  -  ( all axis working machine run perfectly and I also try $ss but there is no [MSG:DBG:] I don't send $h because there is no limit switch )

3rd test  -  (set feed and speed scaler to 1.5 for all the axis and set [Xlimit_all_pin: gpio.16:low:pu],[Ylimit_all_pin: gpio.17:low:pu]
[Zlimit_all_pin: gpio.4:low:pu] )
Result   -  (same as before X and Z axis Working Good but Y axis not Moving I send $ss and here is the message 
[MSG:INFO: FluidNC v3.6.2]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
[MSG:INFO: Local filesystem type is spiffs]
[MSG:INFO: Configuration file:SHRIKE_V1.YAML]
[MSG:WARN: max_travel_mm value 0.000 constrained to range (0.100,10000000.000)]
[MSG:DBG: Running after-parse tasks]
[MSG:DBG: Checking configuration]
[MSG:INFO: Machine SHRIKE MACHINE]
[MSG:INFO: Board SHRIKE MACHINE]
[MSG:INFO: SPI not defined]
[MSG:INFO: No SD Card CS Pin]
[MSG:INFO: See http://wiki.fluidnc.com/en/config/sd_card#sdfallbackcs-access-sd-without-a-config-file]
[MSG:INFO: Stepping:RMT Pulse:2us Dsbl Delay:0us Dir Delay:1us Idle Delay:250ms]
[MSG:INFO: Axis count 4]
[MSG:INFO: Axis X (-317.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:gpio.18 Dir:gpio.25 Disable:NO_PIN]
[MSG:INFO:  X All Limit gpio.16:low:pu]
[MSG:DBG: Updating  X All Limit]
[MSG:DBG:  X All Limit 0]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Y (-250.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:gpio.26 Dir:gpio.15 Disable:NO_PIN]
[MSG:INFO:  Y All Limit gpio.17:low:pu]
[MSG:DBG: Updating  Y All Limit]
[MSG:DBG:  Y All Limit 0]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Z (-140.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:gpio.27 Dir:gpio.33 Disable:NO_PIN]
[MSG:INFO:  Z All Limit gpio.4:low:pu]
[MSG:DBG: Updating  Z All Limit]
[MSG:DBG:  Z All Limit 0]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis A (-0.100,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:gpio.12 Dir:gpio.13 Disable:NO_PIN]
[MSG:INFO:   Motor1]
[MSG:INFO: feed_hold_pin gpio.5:low:pu]
[MSG:INFO: cycle_start_pin gpio.19:low:pu]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: PWM Spindle Ena:gpio.22 Out:gpio.2 Dir:NO_PIN Freq:5000Hz Period:8191]
[MSG:INFO: Using spindle PWM]
[MSG:INFO: Flood coolant gpio.21]
[MSG:INFO: Mist coolant gpio.23]
[MSG:INFO: Probe Pin: gpio.32:low:pu]
[MSG:DBG: Begin Bluetooth setup]
[MSG:INFO: BT Started with FluidNC]
ok )

4th test - (remove all the changes in YAML file and back to the default that I mention on top of the issue, change the version to 3.6.0 ) 
Result  -  (everything working nicely and work as expected no errors machine home perfectly and run smoothly)

One more question - With standard grbl with Arduino I can move the Machine to hard limit switch but With fluidnc i can't move machine in a positive direction for example (switch pull of distance is 2mm when i send $h machine touch the switch and move 2mm for releasing the switch so now I have 2mm more space to move the machine in a positive direction but with fluidnc i cant move machine in a positive direction after homing) if is there any change in .yaml file let me know .

Thank you for your quick response and helping hand .

MitchBradley, 2022-10-02:
Some is obviously wrong with limit switches in 3.6.2.  I have been trying several solutions to solving various problems that have show up with different kinds of limit circuitry.  Obviously I have not succeeded yet.

The reason you cannot move more in the positive direction is because of soft limits.  When soft limits is true, positive_direction is true, max_travel_mm is 250, and mpos_mm is -2, the allowed motion after homing is within the range -252 .. -2 .  Any attempt to move outside that range will not be allowed.

arlimbad222, 2022-10-02:
Hi @MitchBradley 

Thankyou for your response, I'll wait for new update and test it again and update here.

And i also understand that there is no way to cover the distence to limit switch.

Have a good day 😊

MitchBradley, 2022-10-02:
Note the mpos_mm is the position after the pulloff, i.e. the pulloff / limit switch travel allowance is outside the travel envelope.  Typically we set mpos_mm to 0, so the allowable range would be -250 .. 0.  That way the pulloff amount is only interesting for homing and the run-time limits do not depend on the choice of pulloff distance.

arlimbad222, 2022-10-07:
Hi @MitchBradley 

Today I tried "Test Limits pre-release" & I tried all the homing commands like ($hx,$hy,$hz,$h, G28XYZ0) all the commands are working for me with UGS and open build controller.

Thank you For Your Support, Now You Can Close The issue.

MitchBradley, 2022-10-16:
Please try this https://github.com/bdring/FluidNC/releases/tag/v3.6.4-pre1

closed 2022-10-16 by MitchBradley