bdring/FluidNC #642: sdoppiare l'asse y per non surriscaldare drv8825

distebia, 2022-10-01:
### Controller Board

mks dlc32 v2.1

### Help From Board Vendor

- [ ] Yes
- [ ] No
- [X] Not Applicable

### Machine Description

progetto per laser 450nm portatile

### Configuration file

```yaml
board: MKS DLC32 2.1
name: MKS DLC32 XYZ
meta: (08/07/2022) Biagio Diodo Portabile

arc_tolerance_mm: 0.002
junction_deviation_mm: 0.010
verbose_errors: false
report_inches: false
enable_parking_override_control: false
use_line_numbers: false
planner_blocks: 16

stepping:
  engine: I2S_STATIC
  idle_ms: 255
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: i2so.0
  x:
    steps_per_mm: 79.973
    max_rate_mm_per_min: 20000.000
    acceleration_mm_per_sec2: 5000.000
    max_travel_mm: 1226.000
    soft_limits: true
    homing:
      cycle: 0
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 1000.000
      seek_mm_per_min: 5000.000
      settle_ms: 250.000
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm:1.500
      stepstick:
        step_pin:  i2so.1
        direction_pin: I2SO.2

  y:
    steps_per_mm: 80.194
    max_rate_mm_per_min: 10000.000
    acceleration_mm_per_sec2: 200.000
    max_travel_mm: 898.000
    soft_limits: true
    homing:
      cycle: 0
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 1000.000
      seek_mm_per_min: 5000.000
      settle_ms: 250.000
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm:1.500
      stepstick:
        step_pin: I2SO.5
        direction_pin: I2SO.6

    motor1:	
      motor:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm:1.000
      stepstick:
        step_pin: I2SO.3
        direction_pin: I2SO.4
  

    

i2so:
  bck_pin: gpio.16
  data_pin: gpio.21
  ws_pin: gpio.17

spi:
  miso_pin: gpio.12
  mosi_pin: gpio.13
  sck_pin: gpio.14

sdcard:
  card_detect_pin: NO_PIN
  cs_pin: gpio.15

coolant:
  flood_pin: gpio.4
  mist_pin: NO_PIN
  delay_ms: 0

probe:
  pin: gpio.2
  check_mode_start: false

start:
  must_home: false

Laser:
  pwm_hz: 15000
  output_pin: gpio.32
  enable_pin: NO_PIN
  disable_with_s0: false
  s0_with_disable: true
  speed_map: 0=0.000% 0=12.000% 1000=100.000%
```

### Startup Messages

```text
da realizzare struttura macchina quindi tutto da provare
```

### User Interface Software

lightburn, lasergrbl

### What happened?

Salve, non è un problema in quanto ancora è tutto in fase di progettazione, sto realizzando una macchina a diodo portatile e utilizzo 2 motori 17hs8401 per l'asse Y e un 17hs8401 per l'asse X, non ho la Z, niente finecorsa e niente homing. La mia idea era quella di inserire nella scheda 3 drv8825, uno per l'asse x, uno per l'asse y e l'altro nello slot Z ma configurato per essere Y. Tutto questo per non affaticare in corrente il driver drv8825 (per evitare che un drv8825 inserito su Y comandi 2 17hs8401 per l'asse Y. Il file è solo un esempio, da testare, ho solo copiato la configurazione di Z/motor0 su Y/motor1.
E' fattibile tutto questo o devo necessariamente comandare con Y un drv8825 che gestisce 2 17hs8401?
Grazie

MitchBradley, 2022-10-01:
Yes you are on the right track.  Many people use FluidNC with 2 motors and 2 drivers on one axis.

closed 2022-10-01 by MitchBradley

distebia, 2022-10-01:
Grazie