bdring/FluidNC #643: Problem: FluidNC not installing, fluidTerm not connecting

SimonbJohnson, 2022-10-03:
### Controller Board

6-pack

### Help From Board Vendor

- [ ] Yes
- [ ] No
- [X] Not Applicable

### Machine Description

Fabric cutting machine with rotary wheel and laser cutter

### Configuration file

```yaml
Config file not loaded yet
```

### Startup Messages

```text
--- Available ports:
---  1: /dev/cu.Bluetooth-Incoming-Port 'n/a'
---  2: /dev/cu.usbserial-0001 'CP2102 USB to UART Bridge Controller'
---  3: /dev/cu.wlan-debug   'n/a'
--- Enter port index or full name: 2
--- Fluidterm on /dev/cu.usbserial-0001  115200,8,N,1 ---
--- Quit: Ctrl+] or Ctrl+Q | Menu: Ctrl+T | Help: Ctrl+T followed by 'H' ---
 Speed      : 40MHz
I (77) boot: SPI Mode       : DIU����
I (113) boot: ## Label            Usage          Type ST Offset   Length
I (136) boot:  0 phy_init         RF data          01 01 0000f000 00001000
I (159) boot:  1 otadata          OTA data         01 00 00010000 00002000
I (183) boot:  2 nvs              WiFi data        01 02 00012000 0000e000
I (206) boot:  3 at_customize     unknown          40 00 00020000 000e0000
I (229) boot:  4 ota_0            OTA app          00 10 00100000 00180000
I (252) boot:  5 ota_1            OTA app          00 11 00280000 00180000
I (276) boot: End of partition table
I (289) boot: Disabling RNG early entropy source...
I (306) boot: Loading app partition at offset 00100000
I (1481) boot: segment 0: paddr=0x00100018 vaddr=0x00000000 size=0x0ffe8 ( 65512) 
I (1482) boot: segment 1: paddr=0x00110008 vaddr=0x3f400010 size=0x1c5f0 (116208) map
I (1498) boot: segment 2: paddr=0x0012c600 vaddr=0x3ffb0000 size=0x0215c (  8540) load
I (1529) boot: segment 3: paddr=0x0012e764 vaddr=0x40080000 size=0x00400 (  1024) load
I (1552) boot: segment 4: paddr=0x0012eb6c vaddr=0x40080400 size=0x1b028 (110632) load
I (1631) boot: segment 5: paddr=0x00149b9c vaddr=0x400c0000 size=0x00034 (    52) load
I (1632) boot: segment 6: paddr=0x00149bd8 vaddr=0x00000000 size=0x06430 ( 25648) 
I (1648) boot: segment 7: paddr=0x00150010 vaddr=0x400d0018 size=0x7a56c (501100) map
I (1676) heap_alloc_caps: Initializing. RAM available for dynamic allocation:
I (1698) heap_alloc_caps: At 3FFBA6B8 len 00025948 (150 KiB): DRAM
I (1719) heap_alloc_caps: At 3FFE8000 len 00018000 (96 KiB): D/IRAM
I (1741) heap_alloc_caps: At 4009B428 len 00004BD8 (18 KiB): IRAM
I (1761) cpu_start: Pro cpu up.
I (1773) cpu_start: Single core mode
I (1786) cpu_start: Pro cpu start user code
I (1847) cpu_start: Starting scheduler on PRO CPU.
I (2051) uart: queue free spaces: 10
Bin version:0.10.0
I (2053) wifi: wifi firmware version: c604573
I (2053) wifi: config NVS flash: enabled
I (2054) wifi: config nano formating: disabled
I (2062) wifi: Init dynamic tx buffer num: 32
I (2062) wifi: wifi driver task: 3ffc4f34, prio:23, stack:3584
I (2068) wifi: Init static rx buffer num: 10
I (2072) wifi: Init dynamic rx buffer num: 0
I (2076) wifi: Init rx ampdu len mblock:7
I (2079) wifi: Init lldesc rx ampdu entry mblock:4
I (2084) wifi: wifi power manager task: 0x3ffca2dc prio: 21 stack: 2560
I (2091) wifi: wifi timer task: 3ffcb35c, prio:22, stack:3584
E (2096) phy_init: PHY data partition validated
I (2132) phy: phy_version: 329, Feb 22 2017, 15:58:07, 0, 0
I (2132) wifi: mode : softAP (78:e3:6d:10:1c:55)
I (2135) wifi: mode : sta (78:e3:6d:10:1c:54) + softAP (78:e3:6d:10:1c:55)
I (2139) wifi: mode : softAP (78:e3:6d:10:1c:55)
```

### User Interface Software

FluidTerm

### What happened?

I installed fluidNC via these instructions https://github.com/bdring/FluidNC/wiki/FluidNC-Compiling

I had to press the boot button while installing as per the FAQ suggestions.

When trying to connect via FluidTerm the above message is given and it is not responsive.

Using Espressif ESP32-WROOM-32D

bdring, 2022-10-03:
Is that a bare ESP32 module or connected to a controller?

SimonbJohnson, 2022-10-03:
Bare ESP32 module

bdring, 2022-10-03:
Try running intall-wifi from the release again and reply with the text that is generated.

It may be having trouble choosing the serial port.

SimonbJohnson, 2022-10-03:
That has worked. This time it was writing for a lot longer time compared to before

```
esptool.py --chip esp32 --baud 230400 dump_mem 0x3ff5a018 4 SecurityFuses.bin
esptool.py v3.1
Found 3 serial ports
Serial port /dev/cu.wlan-debug
/dev/cu.wlan-debug failed to connect: [Errno 16] could not open port /dev/cu.wlan-debug: [Errno 16] Resource busy: '/dev/cu.wlan-debug'
Serial port /dev/cu.usbserial-0001
Connecting........_____....._____.....__
Chip is ESP32-D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 78:e3:6d:0a:94:90
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 230400
Changed.
Read 4 bytes
Done!
Hard resetting via RTS pin...
esptool.py --chip esp32 --baud 230400 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size detect 0x1000 common/bootloader_dio_80m.bin 0xe000 common/boot_app0.bin 0x10000 wifi/firmware.bin 0x8000 wifi/partitions.bin
esptool.py v3.1
Found 3 serial ports
Serial port /dev/cu.wlan-debug
/dev/cu.wlan-debug failed to connect: [Errno 16] could not open port /dev/cu.wlan-debug: [Errno 16] Resource busy: '/dev/cu.wlan-debug'
Serial port /dev/cu.usbserial-0001
Connecting........_____.
Chip is ESP32-D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 78:e3:6d:0a:94:90
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 230400
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Flash will be erased from 0x00001000 to 0x00005fff...
Flash will be erased from 0x0000e000 to 0x0000ffff...
Flash will be erased from 0x00010000 to 0x0016bfff...
Flash will be erased from 0x00008000 to 0x00008fff...
Compressed 17120 bytes to 11841...
Wrote 17120 bytes (11841 compressed) at 0x00001000 in 0.7 seconds (effective 185.3 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 47...
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.1 seconds (effective 796.8 kbit/s)...
Hash of data verified.
Compressed 1422224 bytes to 871012...
Wrote 1422224 bytes (871012 compressed) at 0x00010000 in 39.2 seconds (effective 290.4 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 129...
Wrote 3072 bytes (129 compressed) at 0x00008000 in 0.1 seconds (effective 310.5 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```

bdring, 2022-10-03:
It is possible it was having trouble with the serial port. It could not figure out which one to use or something else was holding it open. Some people have to disable all other serial ports to get it to program. 

The install program is not something we wrote. We just access standard ESP tools with our batch files.

The initial text you showed is typical of an ESP32 that was never programmed.

closed 2022-10-03 by bdring