bdring/FluidNC #644: Problem: Cannot move X motor with 6-pack

arhuaco, 2022-10-03:
### Controller Board

6-pack

### Help From Board Vendor

- [ ] Yes
- [ ] No
- [X] Not Applicable

### Machine Description

3040T chinese CNC. DM542T drivers, 3  3-23HS30-2804S 2.8A Stepperonline motors. 500W spindle, controlled with external PWM.

I'll use mechanical switches and normal open for the moment. I'll later switch to normal closed.

### Configuration file

```yaml
board: 6 Pack
name: 6 Pack External XYYZ No Square
meta: Based on 2022-09-19 B. Dring for Dean Embrey

stepping:
  engine: I2S_STREAM
  idle_ms: 250
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: NO_PIN
  x:
    steps_per_mm: 100
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 280
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: true
      mpos_mm: 0.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      standard_stepper:
        disable_pin: I2SO.0:low
        step_pin: I2SO.2
        direction_pin: I2SO.1

i2so:
  bck_pin: gpio.22
  data_pin: gpio.21
  ws_pin: gpio.17

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  card_detect_pin: NO_PIN
[... 519 lines cut ...]
verbose_errors: true
report_inches: false
enable_parking_override_control: false
use_line_numbers: false
planner_blocks: 16
NoSpindle:

ok
[verbose] Other: ok
*** Fetching device state
[verbose] > $GCode/Modes
[verbose] Other: [GC:G0 G54 G17 G21 G90 G94 M5 M9 T0 F0 S0]
[verbose] [GC:G0 G54 G17 G21 G90 G94 M5 M9 T0 F0 S0]
ok
[verbose] Other: ok
*** Connected to FluidNC 3.6.2
```

### User Interface Software

UGS

### What happened?

I'm just starting and I have been trying with an example that was posted recently for this board. I haven't been able to make the motors run thus I'm testing with a single one, the one labeled 'x'. I managed to step it by hand but not with the board. Using "disable_pin: I2SO.0:low" now, but I tried 'high' as well.

MitchBradley, 2022-10-03:
Can you use FluidTerm for initial tests?  UGS is good for when things are working, but when things are going wrong there is just too much magic happening to make sense of the messages.

arhuaco, 2022-10-03:
Logs. I tried to configure WiFi but not trying to use it now.

```
[MSG:INFO: FluidNC v3.6.2]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
[MSG:INFO: Local filesystem type is spiffs]
[MSG:INFO: Configuration file:config.yaml]
[MSG:INFO: Machine 6 Pack External XYYZ No Square]
[MSG:INFO: Board 6 Pack]
[MSG:INFO: I2SO BCK:gpio.22 WS:gpio.17 DATA:gpio.21]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN]
[MSG:INFO: Stepping:I2S_stream Pulse:4us Dsbl Delay:0us Dir Delay:1us Idle Delay:250ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (-280.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.2 Dir:I2SO.1 Disable:I2SO.0:low]
[MSG:INFO: Axis Y (-1000.000,0.000)]
[MSG:INFO: Axis Z (-1000.000,0.000)]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Using spindle NoSpindle]
[MSG:INFO: Connecting to STA SSID:3_primo]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: No SSID]
[MSG:INFO: AP SSID FluidNC IP 192.168.0.1 mask 255.255.255.0 channel 1]
[MSG:INFO: AP started]
[MSG:INFO: WiFi on]
[MSG:INFO: Captive Portal Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
```

bdring, 2022-10-03:
Show a photo of the controller and connections to the motor driver.

MitchBradley, 2022-10-03:
Remove the :low on the disable pin.  The logic sense of that pin is not what you think.

arhuaco, 2022-10-03:
Removed :low -- new message:
```[MSG:INFO: Axis X (-280.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.2 Dir:I2SO.1 Disable:I2SO.0]
```
Pictures follow. Contacts work and there are no shorts. I verified when installing.

[image]

[image]

[image]

Motor.

[image]

MitchBradley, 2022-10-03:
Set idle_ms to 255.  Then when the system starts the motor should lock

arhuaco, 2022-10-03:
Oh oh, they are always locked when the driver is powered. I tried with two drivers and it's the same with both. I managed to step the drivers by hand though, driving control pins with another cable and a 4.5v PSU.

MitchBradley, 2022-10-03:
Try sending  `G1 X100 F400` to move the X axis.

arhuaco, 2022-10-03:
```
G1 X100 F400
ok
```

arhuaco, 2022-10-03:
```
G1 X100 F400
ok
```
Just tried. Not moving.

MitchBradley, 2022-10-03:
Are there any messages after the Grbl 3.6 message when you reset with CTRL-R ?

bdring, 2022-10-03:
Do you have the main power connected and a green LED lit on the controller?

bdring, 2022-10-03:
It does not appear to be on in your picture.

MitchBradley, 2022-10-03:
I will turn it on for you by remote control 
[image]

arhuaco, 2022-10-03:
Oh, it is. 12V 1A.

I had it off when I took the photos for the connection pics :-) I have that PSU with the same switch that turns on the stepper's so that I don't forget.

[image]

bdring, 2022-10-03:
Do you have a meter that can measure voltages? If so add a coolant section to your config file and reload it. 

```yaml
coolant:
  mist: i2so.3
```

Measure the voltage on that header next to the stepper connector from the pin labeled `i_3` to `gnd`

Measure the voltage, send M7, measure the voltage, send M9, measure the voltage.

arhuaco, 2022-10-03:
Oh oh. 0.0 in all cases. I measured x(EN) and it was set (~4.6).

```
coolant:
  mist_pin: i2so.3
```

```
[MSG:INFO: Mist coolant I2SO.3]
```

arhuaco, 2022-10-03:
> Are there any messages after the Grbl 3.6 message when you reset with CTRL-R ?

Missed this one, I always check those. I disconnected the EN pins as suggested. I always reset and check the messages. A warning just helped me in the previous step.

arhuaco, 2022-10-03:
Got a new ESP32. After that I managed to move X! Happy. Weird thing is that this ESP32 was new, straight to the board.

- Measure the voltage: 0v
- Send M7: 5v
- Send M9: 0v

Next time I have issues I'll try this test first.  Thanks a lot to both of you. Looking forward to have everything running.

WiFi worked as well.

closed 2022-10-03 by bdring