bdring/FluidNC #645: Feature: ble

lucapgt, 2022-10-06:
### Machine Context

Building cnc cutting machine that works with android/ios app with ble

### Feature Description

Need ble support for ios

### Other Approaches

Without ble we can't use the app. The app needs to be internet connected and send commands on ble

### How I Can Help

Add ble support

MitchBradley, 2022-10-06:
This would require a lot of development and testing for a limited audience.  How much would you be willing to pay to fund suchn development?

lucapgt, 2022-10-06:
How the people that develope ios app can use this? Because bt classic don't work on ios and with wifi ap internet connection is unavailable. About the payment, i have no idea 😅

bdring, 2022-10-06:
It might be better to use WIFI if you are developing a new app.

MitchBradley, 2022-10-06:
We only use AP mode for initial setup. After that it is better to use STA mode connected to an external router. ESP32's AP mode does not work well for production use.  It requires too many resources and has memory leaks

MitchBradley, 2022-10-06:
With STA mode, iOS can connect to the internet snd the ESP32.

lucapgt, 2022-10-06:
Yes, STA mode can solve 50%. If there is a wifi network i can add the device on it. If wifi network is not present i need to find a solution. But 50% is better then 0%

MitchBradley, 2022-10-06:
Bring a router.  Old slow ones are fine, and can be had for next to nothing.  Many people have several old ones in the junk pile.

MitchBradley, 2022-10-06:
A portable hotspot would probably work too.  They cost $10 and are not much larger than a USB flash drive.

closed 2022-10-09 by MitchBradley