bdring/FluidNC #647: Problem: X & Y Driver test failed. Check motor power after upgrade to 3.6.2

MarkJB, 2022-10-08:
### Controller Board

My own custom [ESP32 Plotter Controller](https://github.com/theworkisthework/ESP32_Plotter_Controller)

I've been testing FluidNC on this board for a while now on a Wallbot plotter and it has been running reasonably reliably with 3.5.1 and TMC2130 in SPI mode. 

The biggest annoyance was I found the TMC2130s would fail to initialize on a reset (reset button) and only a power cycle would re-initialize them correctly, so I wondered if It was worth trying a newer version but after upgrading to 3.6.2, the TMC2130s fail the startup check every time.

Downgrading to 3.5.1 and the drivers work again.
I've not seen these problems on the same board with GrblEsp.

### Help From Board Vendor

- [ ] Yes
- [ ] No
- [X] Not Applicable

### Machine Description

Wallbot with servo and 2 x TMC2130 spi mode stepper drivers.

### Configuration file

```yaml
name: ESP32_Plotter_Controller_RevB_Emerald_Dingo
board: Emerald Dingo (RevB)

kinematics:
  WallPlotter:
    left_axis: 0
    left_anchor_x: -100.000
    left_anchor_y: 100.000
    right_axis: 1
    right_anchor_x: 100.000
    right_anchor_y: 100.000
    segment_length: 10.000

stepping:
  engine: RMT
  idle_ms: 250
  dir_delay_us: 0
  pulse_us: 3
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: gpio.2
  x:
    steps_per_mm: 100.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 200.000
    max_travel_mm: 300.000
    soft_limits: false
    homing:
      cycle: 2
      mpos_mm: 0.000
      positive_direction: true
      settle_ms: 250.000
      seek_mm_per_min: 2000.000
      feed_mm_per_min: 200.000
      seek_scaler: 1.100
      feed_scaler: 5.000
    motor0:
      limit_all_pin: NO_PIN
      hard_limits: false
      tmc_2130:
        direction_pin: gpio.26
        step_pin: gpio.25
        run_amps: 0.350
        hold_amps: 0.125
        microsteps: 16
        stallguard: 16
        stallguard_debug: false
        run_mode: StealthChop
        homing_mode: StealthChop
        use_enable: false
        r_sense_ohms: 0.110
        cs_pin: gpio.17
    motor1:
      null_motor:
  y:
    steps_per_mm: 100.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 200.000
    max_travel_mm: 300.000
    soft_limits: false
    homing:
      cycle: 2
      mpos_mm: 0.000
      positive_direction: false
      settle_ms: 250.000
      seek_mm_per_min: 2000.000
      feed_mm_per_min: 200.000
      seek_scaler: 1.100
      feed_scaler: 5.000
    motor0:
      limit_all_pin: NO_PIN
      hard_limits: false
      tmc_2130:
        direction_pin: gpio.33
        step_pin: gpio.32
        run_amps: 0.350
        hold_amps: 0.125
        microsteps: 16
        stallguard: 16
        stallguard_debug: false
        run_mode: StealthChop
        homing_mode: StealthChop
        use_enable: false
        r_sense_ohms: 0.110
        cs_pin: gpio.16
    motor1:
      null_motor:
  z:
    steps_per_mm: 100.000
    max_rate_mm_per_min: 1000.000
    acceleration_mm_per_sec2: 200.000
    max_travel_mm: 5.000
    soft_limits: false
    homing:
      cycle: 1
      mpos_mm: 5.000
      positive_direction: false
      settle_ms: 250.000
      seek_mm_per_min: 2000.000
      feed_mm_per_min: 200.000
      seek_scaler: 1.100
      feed_scaler: 5.000
    motor0:
      null_motor:
    motor1:
      null_motor:

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  cs_pin: gpio.5

control:
  reset_pin: gpio.0:low:pu

coolant:
  delay_ms: 1000.000

probe:
  check_mode_start: false

macros:
  startup_line0: 
  startup_line1: 
  macro0: 
  macro1: 
  macro2: 
  macro3: 

start:
  must_home: true
  check_limits: true
  deactivate_parking: false

user_outputs:

relay:
  tool_num: 0
  speed_map: 0=0.0% 1000=100.0%
  spinup_ms: 0
  spindown_ms: 0
  output_pin: gpio.27
  disable_with_s0: false
  s0_with_disable: false
  spinup_ms: 0
  spindown_ms: 0
arc_tolerance_mm: 0.002
junction_deviation_mm: 0.010
verbose_errors: false
report_inches: false
enable_parking_override_control: false
use_line_numbers: false
```

### Startup Messages

```text
[MSG:INFO: FluidNC v3.6.2]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
[MSG:INFO: Local filesystem type is spiffs]
[MSG:INFO: Configuration file:config.yaml]
[MSG:INFO: Machine ESP32_Plotter_Controller_RevB_Emerald_Dingo]
[MSG:INFO: Board Emerald Dingo (RevB)]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN]
[MSG:INFO: Stepping:RMT Pulse:3us Dsbl Delay:0us Dir Delay:0us Idle Delay:250ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Shared stepper disable gpio.2]
[MSG:INFO: Axis X (-300.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_2130 Step:gpio.25 Dir:gpio.26 CS:gpio.17 Disable:NO_PIN Index:-1 R:0.110]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Y (0.000,300.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_2130 Step:gpio.32 Dir:gpio.33 CS:gpio.16 Disable:NO_PIN Index:-1 R:0.110]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Z (5.000,10.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:   Motor1]
[MSG:ERR: X Axis driver test failed. Check motor power]
[MSG:ERR: Y Axis driver test failed. Check motor power]
[MSG:INFO: reset_pin gpio.0:low:pu]
[MSG:INFO: Kinematic system: WallPlotter]
[MSG:INFO: Relay Spindle Ena:NO_PIN Out:gpio.27 Dir:NO_PIN]
[MSG:INFO: Using spindle Relay]
[MSG:INFO: Connecting to STA SSID:TP-Link_FBC8]
[MSG:INFO: Connecting.]
[MSG:INFO: Connected - IP is 192.168.2.108]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]

Grbl 3.6 [FluidNC v3.6.2 (wifi) '$' for help]
[MSG:INFO: '$H'|'$X' to unlock]
```

### User Interface Software

WebUI, FluidTerm

### What happened?

After upgrading to 3.6.2 stepper driver test failed.
Was working on 3.5.1.

bdring, 2022-10-08:
We were forced to rewrite the SPI driver to solve that problem. There is a branch of the code that has the fix called TMCSPI. 

If you want to compile it, you can test it. It is passing all our tests. We might tweak a few minor things, but it looks like it will be released soon.

MarkJB, 2022-10-08:
Thanks for the heads up. Driver test is passing on that branch. I'll give that a go.

`[MSG:INFO: FluidNC v3.5.1-pre-FS (TMCSPI-ab6742d2)]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
[MSG:INFO: Local filesystem type is spiffs]
[MSG:INFO: Configuration file:config.yaml]
[MSG:INFO: Machine ESP32_Plotter_Controller_RevB_Emerald_Dingo]
[MSG:INFO: Board Emerald Dingo (RevB)]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN]
[MSG:INFO: Stepping:RMT Pulse:3us Dsbl Delay:0us Dir Delay:0us Idle Delay:250ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Shared stepper disable gpio.2]
[MSG:INFO: Axis X (-300.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_2130 Step:gpio.25 Dir:gpio.26 CS:gpio.17 Disable:NO_PIN Index:-1 R:0.110]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Y (0.000,300.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_2130 Step:gpio.32 Dir:gpio.33 CS:gpio.16 Disable:NO_PIN Index:-1 R:0.110]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Z (5.000,10.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:   Motor1]
[MSG:INFO: X Axis driver test passed]
[MSG:INFO: Y Axis driver test passed]
[MSG:INFO: reset_pin gpio.0:low:pu]
[MSG:INFO: Kinematic system: WallPlotter]
[MSG:INFO: Relay Spindle Ena:NO_PIN Out:gpio.27 Dir:NO_PIN]
[MSG:INFO: Using spindle Relay]
[MSG:INFO: Connecting to STA SSID:TP-Link_FB8C]
[MSG:INFO: Connecting.]
[MSG:INFO: Connected - IP is 192.168.2.108]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]

Grbl 3.5 [FluidNC v3.5.1-pre-FS (TMCSPI-ab6742d2) (wifi) '$' for help]
[MSG:INFO: '$H'|'$X' to unlock]`

closed 2022-10-08 by MarkJB