bdring/FluidNC #648: Feature: Support Grbl 1.1 $30

buster3, 2022-10-08:
### Machine Context

I build a laser engraver and want to use FluidNC together with LaserGRBL. 

### Feature Description

LaserGRBL tries to query $30 to get the max laser power value.

https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration#30---max-spindle-speed-rpm

Can FluidNC support this command? 

### Other Approaches

Fix it at LaserGRBL side.
https://github.com/arkypita/LaserGRBL/issues/1885

### How I Can Help

I'm a C++ developer. I can work on the implementation but need some hints where to look in the code.

MitchBradley, 2022-10-08:
Start at this line in ProcessSettings.cpp`
```c++
    new UserCommand("32", "FakeLaserMode", fakeLaserMode, notIdleOrAlarm);
```
The information you want is in the "speed" member of the last entry of the vector spindle->_speeds .

bdring, 2022-10-09:
Merged PR

closed 2022-10-09 by bdring