bdring/FluidNC #654: When Jogging, Motor Runs Backwards then Forwards

reynolds087, 2022-10-09:
### Controller Board

6-pack External Stepper v2

### Help From Board Vendor

- [ ] Yes
- [ ] No
- [X] Not Applicable

### Machine Description

Gantry router with external steppers. X motor uses single closed loop stepper set to 800 steps per revolution. Lead screw is 8mm lead. Software is set to 100 steps per mm. This used to work with my home-brewed controller, but recently swapped out for the 6 pack. I can't put the old controller on to test because the wires to the steppers are different (old system I jumpered positive across each control pin to 5v, and output pins switched to ground)

### Configuration file

```yaml
board: 6 Pack
name: 2022_07-03_6-Pack_PWM
stepping:
  engine: I2S_STREAM
  idle_ms: 250
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: NO_PIN
  x:
    steps_per_mm: 100.000
    max_rate_mm_per_min: 2500.000
    acceleration_mm_per_sec2: 10.000
    max_travel_mm: 380.000
    soft_limits: true
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 0
      feed_mm_per_min: 10.000
      seek_mm_per_min: 2000.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.27:low:pu
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 1.250
      standard_stepper:
        step_pin: I2SO.13
        direction_pin: I2SO.12
        disable_pin: I2SO.15

  y:
    steps_per_mm: 100.000
    max_rate_mm_per_min: 2500.000
    acceleration_mm_per_sec2: 10.000
    max_travel_mm: 605.000
    soft_limits: true
    homing:
      cycle: 3
      positive_direction: true
      mpos_mm: 0.000
      feed_mm_per_min: 10.000
      seek_mm_per_min: 2000.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: gpio.16:high:pd
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 1.250
      standard_stepper:
        step_pin: I2SO.10
        direction_pin: I2SO.9:low
        disable_pin: I2SO.8

    motor1:
      limit_neg_pin: NO_PIN
      limit_pos_pin: gpio.26:high:pd
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 1.250
      standard_stepper:
        step_pin: I2SO.5
        direction_pin: I2SO.4:low
        disable_pin: I2SO.7

  z:
    steps_per_mm: 1000.000
    max_rate_mm_per_min: 1000.000
    acceleration_mm_per_sec2: 10.000
    max_travel_mm: 300.000
    soft_limits: true
    homing:
      cycle: 1
      positive_direction: true
      mpos_mm: 0.000
      feed_mm_per_min: 10.000
      seek_mm_per_min: 1000.000
      settle_ms: 10
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: gpio.4:low:pu
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 1.250
      standard_stepper:
        step_pin: I2SO.2
        direction_pin: I2SO.1
        disable_pin: I2SO.0

i2so:
  bck_pin: gpio.22
  data_pin: gpio.21
  ws_pin: gpio.17

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  card_detect_pin: NO_PIN
  cs_pin: gpio.5

control:
  safety_door_pin: NO_PIN
  reset_pin: gpio.12:high:pd
  feed_hold_pin: gpio.15:high
  cycle_start_pin: gpio.14:high
  macro0_pin: NO_PIN
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN

coolant:
  flood_pin: NO_PIN
  mist_pin: NO_PIN
  delay_ms: 0

probe:
  pin: gpio.13:high:pd
  check_mode_start: true

macros:
  startup_line0:
  startup_line1:
  macro0:
  macro1:
  macro2:
  macro3:

user_outputs:
  analog0_pin: NO_PIN
  analog1_pin: NO_PIN
  analog2_pin: NO_PIN
  analog3_pin: NO_PIN
  analog0_hz: 5000
  analog1_hz: 5000
  analog2_hz: 5000
  analog3_hz: 5000
  digital0_pin: NO_PIN
  digital1_pin: NO_PIN
  digital2_pin: NO_PIN
  digital3_pin: NO_PIN

start:
  must_home: false

pwm:
  pwm_hz: 5000
  direction_pin: NO_PIN
  output_pin: gpio.25
  enable_pin: I2SO.24:high
  disable_with_s0: false
  s0_with_disable: true
  spinup_ms: 0
  spindown_ms: 0
  tool_num: 0
  speed_map: 0=26% 5000=75%
```

### Startup Messages

```text
[MSG:INFO: FluidNC v3.6.2]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
[MSG:INFO: Local filesystem type is spiffs]
[MSG:INFO: Configuration file:2022_07-03_6-Pack_PWM.txt]
[MSG:INFO: Machine 2022_07-03_6-Pack_PWM]
[MSG:INFO: Board 6 Pack]
[MSG:INFO: I2SO BCK:gpio.22 WS:gpio.17 DATA:gpio.21]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN]
[MSG:INFO: Stepping:I2S_stream Pulse:4us Dsbl Delay:0us Dir Delay:1us Idle Delay:250ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (0.000,380.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.13 Dir:I2SO.12 Disable:I2SO.15]
[MSG:INFO:  X Neg Limit gpio.27:low:pu]
[MSG:INFO: Axis Y (-605.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.10 Dir:I2SO.9:low Disable:I2SO.8]
[MSG:INFO:  Y Pos Limit gpio.16:pd]
[MSG:INFO:   Motor1]
[MSG:INFO:     standard_stepper Step:I2SO.5 Dir:I2SO.4:low Disable:I2SO.7]
[MSG:INFO:  Y2 Pos Limit gpio.26:pd]
[MSG:INFO: Axis Z (-300.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.2 Dir:I2SO.1 Disable:I2SO.0]
[MSG:INFO:  Z Pos Limit gpio.4:low:pu]
[MSG:INFO: reset_pin gpio.12:pd]
[MSG:INFO: feed_hold_pin gpio.15]
[MSG:INFO: cycle_start_pin gpio.14]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: PWM Spindle Ena:I2SO.24 Out:gpio.25 Dir:NO_PIN Freq:5000Hz Period:8191]
[MSG:INFO: Using spindle PWM]
[MSG:INFO: Probe Pin: gpio.13:pd]
[MSG:INFO: STA SSID is not set]
[MSG:INFO: AP SSID FluidNC IP 192.168.0.1 mask 255.255.255.0 channel 1]
[MSG:INFO: AP started]
[MSG:INFO: WiFi on]
[MSG:INFO: Captive Portal Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
```

### User Interface Software

I've tried a few. FluidTerm and UGS yield same results

### What happened?

I send 1mm jog commands (various feed rates do not seem to affect outcome), and two problems occur:

The lead screw rotates more than expected. (10 jogs does not equal exactly one rotation, it exceeds it by approximately 10%)

The lead screw first rotates backwards very slightly, then rotates forwards.

reynolds087, 2022-10-09:
https://user-images.githubusercontent.com/32722984/194782467-b9653189-6806-4a08-850b-604ca6b0986b.mp4

reynolds087, 2022-10-10:
I disconnected the disable pin and it does not spin backwards anymore. But it still goes farther than expected based on the lead of the screw and the steps per mm that are defined. This is happening on both the x and the y axis. I don't have an easy way to test the Z at the moment.

MitchBradley, 2022-10-10:
Try with I2S_STATIC

reynolds087, 2022-10-10:
@MitchBradley , I tried that just now, but it's still not traveling the expected distance. It's like there is a step per mm mismatch, but I don't see how there can be. My X axis lead screw has 8mm lead, the driver is set to 800 steps per rev, and the file is set to 100 steps per mm. Am I missing anything?

MitchBradley, 2022-10-10:
I don't know.  I will have to try and duplicate your setup and measure things.  I have some other stuff I have to do now so it will be awhile before I get to that.

reynolds087, 2022-10-10:
Ok, thanks. Let me know

MitchBradley, 2022-10-10:
Okay I found it.  Your problem is a manifestation of #287 which we just fixed.  I was able to reproduce your problem with v3.6.2, made it go away by changing pwm/output_pin: to a gpio.  Then switched output_pin back to I2SO, upgraded the firmare to v.3.6.3, and still good.

I don't know if this will address the backwards motion; if not, perhaps change idle_ms to 255 to keep the motors locked so gantry stresses do not result in inaccurate positions.

reynolds087, 2022-10-10:
Thanks, @MitchBradley . Can I just upgrade to v3.6.3 to fix it, or do I need to change the configuration of certain pins?

MitchBradley, 2022-10-10:
Just upgrade

reynolds087, 2022-10-11:
[image]
[image]

https://user-images.githubusercontent.com/32722984/194969935-a37e9d79-2609-479d-9aad-134b5ee8139d.mp4

reynolds087, 2022-10-11:
Thought I was still having an issue, but I think it's good now.

reynolds087, 2022-10-11:
It's a lot smoother. I just need to check the distance.

reynolds087, 2022-10-11:
It still runs backwards when the disable pin is used. I never had that problem with GRBL_Esp32. It's strange that I do now without any significant mechanical changes to the machine. Setting idle_ms to 255 works, but then the motors use a lot more energy and run warmer. I haven't stress tested them, but I can imagine they might get hot.

MitchBradley, 2022-10-11:
What command are you using to jog?

reynolds087, 2022-10-11:
I'm not sure what UGS uses, but in fluidterm, I've been testing with $J=g91x10f100

MitchBradley, 2022-10-11:
A close look at the marking on the driver make me think it is a closed loop setup.  I wish you have told us that before.  It changes my thought processes.

MitchBradley, 2022-10-11:
Closed loop steppers automatically adjust the current to maintain the position, so they run cool at standstill.  The only time you would need to disable them is if you needed to move manually.

reynolds087, 2022-10-11:
Ok, then I will just try running them for a while without the disable pin. Do you have any idea why it would behave differently between GRBL_Esp32 vs FluidNC? I suppose it is possible I introduced some error that I can't detect when switching the controllers, but I can't think of what it might be.

MitchBradley, 2022-10-11:
Well you did rewire them.  And I have no idea what configuration you were using for Grbl_Esp32

reynolds087, 2022-10-11:
Alright, well I will give it a try without the disable signal.

closed 2024-12-12 by bdring