bdring/FluidNC #655: Feature: Add Support for the W5500 SPI Ethernet Module

gxdeange, 2022-10-10:
### Machine Context

I am building a custome board that supports 6 channel (I2S) and would like to see if the firmware can support SPI Ethernet (W5500)

### Feature Description

Spi Ertyhernet module support

### Other Approaches

I have not tried any other approach as I do not know how to code

### How I Can Help

I could provide a donation to purchase the ethernet module however they go for a few dollars on AliExpress

MitchBradley, 2022-10-10:
I have used W5500 on ESP8266 so I have a basic idea of how to connect it to the ESP framework. I'm not sure that we want to support it for FluidNC because there are few if any other potential users.  Adding it would require documentation, testing, and support.

gxdeange, 2022-10-10:
> Some of the issues that will come up for supporting W5500 include:
>
>    - Managing MAC addresses
>    - Handling PHY connection/disconnection events
>    - Sharing the SPI bus between the W5500 and the existing SPI use by SD
>    Card and TMC chips, making sure that the usage does not collide when
>    packets are received at arbitrary times. Right now, FluidNC knows exactly
>    when it is using the SPI bus, but that changes if SPI is used for network
>    packet reception.

Ok. No problem.

Thanks again for looking at this request and responding.

Cheers and have a great day

closed 2022-12-02 by MitchBradley