bdring/FluidNC #658: Problem:

doowneergtam, 2022-10-10:
### Controller Board

6 Pack Universal CNC Controller (Bart's V1p5)

Using 
- (2x) 4x Input CNC I/O Modules (Sockets #1 and #2) Limits and Probe
- 5V Output Module (Socket #4) Laser Enable and PWM
- Relay CNC I/O Module (Socket #5) Air Assist Pump

### Help From Board Vendor

- [ ] Yes
- [ ] No
- [ ] Not Applicable

### Machine Description

600mm X 1200mm Custom Built 130W Co2 Laser

### Configuration file

```yaml
board: 6 Pack
name: 6 Pack External XYZA Laser
meta:
stepping:
  engine: I2S_STATIC
  idle_ms: 255
  pulse_us: 3
  dir_delay_us: 3
  disable_delay_us: 0
  segments: 16

axes:
  shared_stepper_disable_pin: NO_PIN
  x:
    steps_per_mm: 160.423
    max_rate_mm_per_min: 18000.000
    acceleration_mm_per_sec2: 800.000
    max_travel_mm: 1260.000
    hard_limits: true
    soft_limits: true
    homing:
      cycle: 3
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 500.000
      seek_mm_per_min: 4000.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.33:low
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      pulloff_mm: 6.000
      standard_stepper:
        step_pin: I2SO.2
        direction_pin: I2SO.1
        disable_pin: I2SO.0

  y:
    steps_per_mm: 99.945
    max_rate_mm_per_min: 30000.000
    acceleration_mm_per_sec2: 8000.000
    max_travel_mm: 604.000
    hard_limits: true
    soft_limits: true
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 500.000
      seek_mm_per_min: 4000.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.32:low
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      pulloff_mm: 6.000
      standard_stepper:
        step_pin: I2SO.5
        direction_pin: I2SO.4
        disable_pin: I2SO.7

  z:
    steps_per_mm: 106.750
    max_rate_mm_per_min: 500.000
    acceleration_mm_per_sec2: 200.000
    max_travel_mm: 140.000
    soft_limits: true
    hard_limits: true
    homing:
      cycle: 1
      positive_direction: true
      mpos_mm: 0.000
      feed_mm_per_min: 200.000
      seek_mm_per_min: 500.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100
        
    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: gpio.35:low
      limit_all_pin: NO_PIN
      pulloff_mm: 4.000
      standard_stepper:
        step_pin: I2SO.10
        direction_pin: I2SO.9
        disable_pin: I2SO.8
        
  a:
    steps_per_mm: 35.545
    max_rate_mm_per_min: 3000.000
    acceleration_mm_per_sec2: 2000.000
    max_travel_mm: 1000.000
    soft_limits: false
    hard_limits: false
    homing:
      cycle: 0
      positive_direction: true
      mpos_mm: 0.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100
        
   motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm:1.000
      standard_stepper:
        step_pin: I2SO.13
        direction_pin: I2SO.12
        disable_pin: I2SO.15

probe:
  pin:  gpio.34
  check_mode_start: true
  
        
i2so:
  bck_pin: gpio.22
  data_pin: gpio.21
  ws_pin: gpio.17

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  card_detect_pin: NO_PIN
  cs_pin: gpio.5

control:
  safety_door_pin: gpio.2
  reset_pin: gpio.25
  feed_hold_pin: gpio.39
  cycle_start_pin: gpio.36
  macro0_pin: NO_PIN
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN

start:
  must_home: true
  
coolant:
  flood_pin: I2SO.24
  mist_pin:  gpio.12
  delay_ms: 0

Laser:
  spinup_ms: 0
  spindown_ms: 0
  tool_num: 0
  pwm_hz: 20000
  output_pin: gpio.14
  enable_pin: gpio.13
  disable_with_s0: false
  s0_with_disable: true
  speed_map: 0=0.000% 1000=100.000%

planner_blocks: 16
```

### Startup Messages

```text
Waiting for connection...
[VER:3.6 FluidNC v3.6.2:]
[OPT:MPHS]
[MSG: Machine: 6 Pack External XYZA Laser]
[MSG: Mode=AP:SSID=FluidNC:IP=192.168.0.1:MAC=A8-03-2A-EB-F0-DD]
ok
Homing
ok
```

### User Interface Software

Lightburn (USB Serial Connection)

### What happened?

I'm having an offset between scanning (fill) and line. It's always in the same direction (along scanning axis) and it widens at higher speeds. It doesn't make a difference if I do the line first or the fill first, the result is identical. When scanning in only one direction the alignment between fill and line is nearly perfect but still off a hair.

I've already ruled out scanning offset, I went through Lightburn's instructions and my scans are starting and ending in the correct timings with good non-wavey lines . My engravings on their own look great at all speeds and if I rerun an engraving it lines up perfectly with the engraving I did first.

I've also ruled out steppers losing steps, machine squareness, and other possible mechanical issues.

I'm at a loss solving this one. Any help would be greatly appreciated.

### Other Information

[image]

MitchBradley, 2022-10-11:
We need a gcode file, as simple as possible, that exhibits the problem.  By simple, I mean really simple.

doowneergtam, 2022-10-11:
Thanks for the quick reply. Here's what you requested. 

Also my laser scans with the Y axis, I've tried swapping the axis so it is scanning with the X axis with the same results.

I'm beginning to suspect the laser's power supply may be causing this problem.  While trying the simple program you requested I noticed the line portion of the program was starting way late. I watched the 6 pack controller's output card and the output and pwm lights were lit up in correct timing and the laser itself wasnt firing. You can see what I mean in the images. It happened several times while trying this for you.

; LightBurn 0.9.24
; GRBL device profile, absolute coords
; Bounds: X504 Y99.56 to X508 Y193
G00 G17 G40 G21 G54
G90
M4
; Scan+Cut @ 300 mm/sec, 55% power
M9
G0X504.05Y189.56
G0Z0
; Layer 100
G91
G1Y-30F18000S0
G1Y-30S550
G1Y-30S0
G1X0.1Y3.44S0
G1Y30S0
G1Y30S550
G1Y30S0
G1X0.1Y-3.44S0
G1Y-30S0
G1Y-30S550
G1Y-30S0
G1X0.1Y3.44S0
G1Y30S0
G1Y30S550
G1Y30S0
G1X0.1Y-3.44S0
G1Y-30S0
G1Y-30S550
G1Y-30S0
G1X0.1Y3.44S0
G1Y30S0
G1Y30S550
G1Y30S0
G1X0.1Y-3.44S0
G1Y-30S0
G1Y-30S550
G1Y-30S0
G1X0.1Y3.44S0
G1Y30S0
G1Y30S550
G1Y30S0
G1X0.1Y-3.44S0
G1Y-30S0
G1Y-30S550
G1Y-30S0
G1X0.1Y3.44S0
G1Y30S0
G1Y30S550
G1Y30S0
G1X0.1Y-3.44S0
G1Y-30S0
G1Y-30S550
G1Y-30S0
G1X0.1Y3.44S0
G1Y30S0
G1Y30S550
G1Y30S0
G1X0.1Y-3.44S0
G1Y-30S0
G1Y-30S550
G1Y-30S0
G1X0.1Y3.44S0
G1Y30S0
G1Y30S550
G1Y30S0
G1X0.1Y-3.44S0
G1Y-30S0
G1Y-30S550
G1Y-30S0
G1X0.1Y3.44S0
G1Y30S0
G1Y30S550
G1Y30S0
G1X0.1Y-3.44S0
G1Y-30S0
G1Y-30S550
G1Y-30S0
G1X0.1Y3.44S0
G1Y30S0
G1Y30S550
G1Y30S0
G1X0.1Y-3.44S0
G1Y-30S0
G1Y-30S550
G1Y-30S0
G1X0.1Y3.44S0
G1Y30S0
G1Y30S550
G1Y30S0
G1X0.1Y-3.44S0
G1Y-30S0
G1Y-30S550
G1Y-30S0
G1X0.1Y3.44S0
G1Y30S0
G1Y30S550
G1Y30S0
G1X0.1Y-3.44S0
G1Y-30S0
G1Y-30S550
G1Y-30S0
G1X0.1Y3.44S0
G1Y30S0
G1Y30S550
G1Y30S0
G1X0.1Y-3.44S0
G1Y-30S0
G1Y-30S550
G1Y-30S0
G1X0.1Y3.44S0
G1Y30S0
G1Y30S550
G1Y30S0
G1X0.1Y-3.44S0
G1Y-30S0
G1Y-30S550
G1Y-30S0
G1X0.1Y3.44S0
G1Y30S0
G1Y30S550
G1Y30S0
G1X0.1Y-3.44S0
G1Y-30S0
G1Y-30S550
G1Y-30S0
G1X0.1Y3.44S0
G1Y30S0
G1Y30S550
G1Y30S0
G1X0.1Y-3.44S0
G1Y-30S0
G1Y-30S550
G1Y-30S0
G1X0.1Y3.44S0
G1Y30S0
G1Y30S550
G1Y30S0
G1X0.1Y-3.44S0
G1Y-30S0
G1Y-30S550
G1Y-30S0
G1X0.1Y3.44S0
G1Y30S0
G1Y30S550
G1Y30S0
G1X0.1Y-3.44S0
G1Y-30S0
G1Y-30S550
G1Y-30S0
G1X0.1Y3.44S0
G1Y30S0
G1Y30S550
G1Y30S0
G1X0.1Y-3.44S0
G1Y-30S0
G1Y-30S550
G1Y-30S0
G1X0.1Y3.44S0
G1Y30S0
G1Y30S550
G1Y30S0
G1X0.1Y-3.44S0
G1Y-30S0
G1Y-30S550
G1Y-30S0
G1X0.1Y3.44S0
G1Y30S0
G1Y30S550
G1Y30S0
G90
G0X504Y133
; Layer 100
G1Y163S150F600
G1X508
G1Y133
G1X504
M9
G1S0
M5
G90
; return to user-defined finish pos
G0 X0 Y0
M2

 
[image]

doowneergtam, 2022-10-11:
I have another brand new laser power supply I can get tomorrow after work to rule it out.

MitchBradley, 2022-10-11:
If you look carefully at the GCode, you will see that it has baked into it an offset between negative-going and positive-going scans.  Here is an excerpt from your file that is annotated to show it:
; Scan+Cut @ 300 mm/sec, 55% power
```
M9
G0X504.05Y189.56 (Y pos is 189.56)
G0Z0
; Layer 100
G91
G1Y-30F18000S0  (Ypos is 159.56; laser turns on for next segment)
G1Y-30S550 (laser turns off at Ypos 129.56)
G1Y-30S0  (Ypos is 99.56)
G1X0.1Y3.44S0 (Ypos is 103)
G1Y30S0 (Ypos is 133; laser turns on for next segment)
G1Y30S550 (laser turns off at Ypos 163)
G1Y30S0
```
So for negative-going, the laser is ostensibly on between (129.56, 159.56), while positive-going is (133, 163).

This is probably some sort of Lightburn compensation for laser turn-on time.

When you are drawing the box, that sort of compensation would not work the same way because the short sides of the box are drawn as simple movements in X, not overscanned raster lines in Y with time compensation.

I am not a Lightburn expert so I'm not sure what to do about it.

doowneergtam, 2022-10-11:
Thank you, I will try when I get home from work. 

I did have to set a scanning offset in LightBurn to compensate for the firing delay (doubled engraving, one for each scan direction), I’m assuming that is what you’re seeing. 

For the speed map suggestion, you’re saying to change the 0% in the speed map to just below where the tube actually fires.

If the laser doesn’t actually fire until 7% maybe try setting the speed map to 

speed_map: 65=0.000% 1000=100.000%

MitchBradley, 2022-10-11:
Please read the wiki entry on speed map.  I spent a long time writing it and making examples.

doowneergtam, 2022-10-12:
Okay thanks, after reading the speed map wiki I set mine to 

speed_map: 0=5% 1000=100%

Now when manually firing the laser tube begins arcing at 0.02%. 

Okay now the offset, when I run with Lightburn’s “Scanning offset adjustment” off I’m getting “doubled” engravings but the line “rectangle” after seems to be centered correctly between the two. 

It seems like the scanning offset adjustment in LightBurn is shifting the engraving towards the Y+ Direction consistently and the line is in the correct place. I’ve been using LightBurn and co2 lasers for a long time, I use scanning offset adjustment on most machines. This is the first time I’ve seen this issue. 

When I re-enable the scanning offset I’m still getting the original result, an offset between lines and scans. 

I’ve also since replaced my laser power supply with a brand new one and replaced the shielded IO cable from the 6 pack Controller to the laser’s power supply with no change. 

I saved the gcode and an image for both offset off and on. The first is offset on the second is off. You’ll also notice in the image I’m getting late firing on the line starts. These results were after making the speed map changes I mentioned. 

[image]

This is a OneDrive link to the two gcode files. 

https://1drv.ms/f/s!ArzLEbLAKBIghu5f-xSPxZTElPYnrw

Thanks

MitchBradley, 2022-10-12:
Try replacing S0 with S1 in the gcoce files

doowneergtam, 2022-10-12:
My lasers power supply operates at 20khz. I’ve tried different frequencies with no change. I’m going to test the pwm tonight. It seems like there’s a definite delay somewhere that’s causing the late line starts and the offset. 

I’m going to try attaching a diode laser and test at the same speeds and see if the issue is present. At least then I would know if the issue is limited to co2 lasers. 

Thank you for all of your help.

MitchBradley, 2022-12-02:
Any update on this?

closed 2024-12-12 by bdring