bdring/FluidNC #659: Problem: SD Problem Latest version 3.6.3

sotchrys, 2022-10-11:
### Controller Board

Custom
each new version of the source code and a new surprise! This time in version 3.6.3 after creating the firmware, the card reader does not work. When I reinstalled the stable version 3.4.4 ... I always do the tests on a card that only has the esp32, OLED and SD Card reader nothing else...

### Help From Board Vendor

- [ ] Yes
- [ ] No
- [X] Not Applicable

### Machine Description

Pen Laser TMC2208 

### Configuration file

```yaml
board: sotchrys
name: PEN_LASER
meta: 
stepping:
  engine: RMT
  idle_ms: 250
  pulse_us: 3
  dir_delay_us: 0
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: gpio.13
  shared_stepper_reset_pin: NO_PIN
  x:
    steps_per_mm: 80.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 1000.000
    max_travel_mm: 300.000
    soft_limits: false
    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: gpio.15:low:pu
      hard_limits: false
      pulloff_mm: 1.000
      stepstick:
        step_pin: gpio.12
        direction_pin: gpio.26
        disable_pin: NO_PIN
        ms1_pin: NO_PIN
        ms2_pin: NO_PIN
        ms3_pin: NO_PIN
        reset_pin: NO_PIN

  y:
    steps_per_mm: 80.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 1000.000
    max_travel_mm: 300.000
    soft_limits: false
    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: gpio.4:low:pu
      hard_limits: false
      pulloff_mm: 1.000
      stepstick:
        step_pin: gpio.14
        direction_pin: gpio.25
        disable_pin: NO_PIN
        ms1_pin: NO_PIN
        ms2_pin: NO_PIN
        ms3_pin: NO_PIN
        reset_pin: NO_PIN

  z:
    steps_per_mm: 100.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 1000.000
    max_travel_mm: 25
    homing:
      cycle: 1
      mpos_mm: 0
      positive_direction: false
    motor0:
      rc_servo:
        pwm_hz: 50
        output_pin: gpio.27
        min_pulse_us: 2000
        max_pulse_us: 1000

kinematics:
  Cartesian:

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  cs_pin: gpio.5
  card_detect_pin: NO_PIN

control:
  safety_door_pin: NO_PIN
  reset_pin: NO_PIN
  feed_hold_pin: NO_PIN
  cycle_start_pin: NO_PIN
  macro0_pin: NO_PIN
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN

coolant:
  flood_pin: NO_PIN
  mist_pin: NO_PIN
  delay_ms: 1000

probe:
  pin: NO_PIN
  check_mode_start: false 

start:
  must_home: false
  deactivate_parking: false
  check_limits: false
  
Laser:
  pwm_hz: 5000
  output_pin: gpio.17
  enable_pin: NO_PIN
  disable_with_s0: false
  s0_with_disable: true
  tool_num: 100
  speed_map: 0=0.000% 255=100.000%
```

### Startup Messages

```text
board: sotchrys
name: PEN_LASER
meta: 
stepping:
  engine: RMT
  idle_ms: 250
  pulse_us: 3
  dir_delay_us: 0
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: gpio.13
  shared_stepper_reset_pin: NO_PIN
  x:
    steps_per_mm: 80.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 1000.000
    max_travel_mm: 300.000
    soft_limits: false
    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: gpio.15:low:pu
      hard_limits: false
      pulloff_mm: 1.000
      stepstick:
        step_pin: gpio.12
        direction_pin: gpio.26
        disable_pin: NO_PIN
        ms1_pin: NO_PIN
        ms2_pin: NO_PIN
        ms3_pin: NO_PIN
        reset_pin: NO_PIN

  y:
    steps_per_mm: 80.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 1000.000
    max_travel_mm: 300.000
    soft_limits: false
    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: gpio.4:low:pu
      hard_limits: false
      pulloff_mm: 1.000
      stepstick:
        step_pin: gpio.14
        direction_pin: gpio.25
        disable_pin: NO_PIN
        ms1_pin: NO_PIN
        ms2_pin: NO_PIN
        ms3_pin: NO_PIN
        reset_pin: NO_PIN

  z:
    steps_per_mm: 100.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 1000.000
    max_travel_mm: 25
    homing:
      cycle: 1
      mpos_mm: 0
      positive_direction: false
    motor0:
      rc_servo:
        pwm_hz: 50
        output_pin: gpio.27
        min_pulse_us: 2000
        max_pulse_us: 1000

kinematics:
  Cartesian:

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  cs_pin: gpio.5
  card_detect_pin: NO_PIN

control:
  safety_door_pin: NO_PIN
  reset_pin: NO_PIN
  feed_hold_pin: NO_PIN
  cycle_start_pin: NO_PIN
  macro0_pin: NO_PIN
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN

coolant:
  flood_pin: NO_PIN
  mist_pin: NO_PIN
  delay_ms: 1000

probe:
  pin: NO_PIN
  check_mode_start: false 

start:
  must_home: false
  deactivate_parking: false
  check_limits: false
  
Laser:
  pwm_hz: 5000
  output_pin: gpio.17
  enable_pin: NO_PIN
  disable_with_s0: false
  s0_with_disable: true
  tool_num: 100
  speed_map: 0=0.000% 255=100.000%
```

### What happened?

.

bdring, 2022-10-11:
What controller is it? Attach a photo.

MitchBradley, 2022-10-11:
> each new version of the source code and a new surprise!

Each user and a different hardware design that we know nothing about.  Makes testing essentially impossible.

bdring, 2022-10-11:
Pen Laser TMC2208 makes it sound like custom hardware. 

I suspect there are pins interfering with the SPI circuitry.

MitchBradley, 2022-10-11:
I just tested on a controller that "only has the esp32, OLED and SD Card reader".  The OLED is of the "tiny" variety.  It is a [GeckoBlaster](http://wiki.fluidnc.com/en/hardware/existing_hardware#gecko-blaster) that is not connected to the Gecko drive box, and the rotary knob is not installed, so it really is "only esp32, oled, sd card".

I fetched the latest main into my git repo, pulled the tags, and verified that the version was at tag v3.6.3.  I then built with
```
pio run -e wifi-oled-tiny -t upload
```
Connected with FluidTerm and
```
ok
$sd/list
[FILE: t.gc|SIZE:20]
[DIR:localfs]
[FILE:  gr3pos.flnc|SIZE:1434]
[FILE:  gr3_2k_20.flnc|SIZE:1236]
[FILE:  gr3.flnc|SIZE:1432]
[FILE:  gr3gecko.yaml|SIZE:925]
etc
```

DEST1981, 2022-10-15:
I also have a problem with the memory card.

```
 rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
 configsip: 0, SPIWP:0xee
 clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
 mode:DIO, clock div:1
 load:0x3fff0030,len:1184
 load:0x40078000,len:12812
 load:0x40080400,len:3032
 entry 0x400805e4
 
 [MSG:INFO: FluidNC v3.6.3]
 [MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
 [MSG:INFO: Local filesystem type is spiffs]
 [MSG:INFO: Configuration file:config3ax.yaml]
 [MSG:DBG: Running after-parse tasks]
 [MSG:DBG: Checking configuration]
 [MSG:INFO: Machine ESP32 Dev Controller V4]
 [MSG:INFO: Board ESP32 Dev Controller V4]
 [MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
 [MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN]
 .........................................................................................................
 Grbl 3.6 [FluidNC v3.6.3 (wifi) '$' for help]
 [MSG:ERR: mount_to_vfs failed code 0xffffffff]
```
I did not use the board, the ESP32 is directly connected to the SD card.

MitchBradley, 2022-10-15:
@DEST1981 Please do not hijack this issue.  Start your own.

MitchBradley, 2022-10-18:
@sotchrys - we are still waiting for the requested information - @bdring requested a photo, and the Startup Messages box contains the config file, not the startup messages.  Startup messages can be obtained by sending $SS .  Another thing that would be good to know is the size and format of the SD card that you are using.

Kenedy-Hanes, 2022-10-23:
Yo tengo el mismo problema y lo solucioné con un adaptador de memoria MICRO SD.

EL controlador que estoy usando es una Placa de desarrollo ESP32 Wemos/Espduino
[image]

El módulo con el que tuve este problema es el siguiente.
[image]

Para solucionar el problema se necesita de un adaptador de memorias MicroSD y unos pines que iran soldados.
[image]
[image]
Una vez soldados los pines me quedaron asi.
[image]
El cableado para el controlador es el siguiente. Les pido encarecidamente que tengan cuidado al momento de alimentar la memoria Micro SD para funcionar solo necesita 3.3v . Si lo conectan a 5V la memoria MicroSD se dañará.
[image]
[image]
Los conectores de los costados del adapatador no se usan.

Esa es mi configuración.

name: "ESP32"
board: "Wemos_CNCshield"
meta: "KyS CNC"

stepping:
  engine: RMT
  idle_ms: 250
  dir_delay_us: 1
  pulse_us: 2
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: NO_PIN

  x:
    steps_per_mm: 40.650
    max_rate_mm_per_min: 400
    acceleration_mm_per_sec2: 2
    max_travel_mm: 100
    homing:
      cycle: 2
      mpos_mm: -3.000
      feed_mm_per_min: 400
      positive_direction: false
    
    motor0:
      limit_all_pin: gpio.36:low
      hard_limits: false
      pulloff_mm: 3.000
      stepstick:
        direction_pin: gpio.16
        step_pin: gpio.26

  y:
    steps_per_mm: 40.650
    max_rate_mm_per_min: 400
    acceleration_mm_per_sec2: 2
    max_travel_mm: 100
    homing:
      cycle: 2
      mpos_mm: -3
      feed_mm_per_min: 400
      positive_direction: false
    
    motor0:
      limit_all_pin: gpio.34:low
      hard_limits: false
      pulloff_mm: 3
      stepstick:
        direction_pin: gpio.27
        step_pin: gpio.25

    motor1:
      limit_all_pin: gpio.39:low
      hard_limits: false
      pulloff_mm: 3
      stepstick:
        direction_pin: gpio.32
        step_pin: gpio.33

  z:
    steps_per_mm: 40.650
    max_rate_mm_per_min: 400
    acceleration_mm_per_sec2: 2
    max_travel_mm: 10
    homing:
      cycle: 1
      mpos_mm: 3
      feed_mm_per_min: 400
      positive_direction: true
    
    motor0:
      limit_all_pin: gpio.35:low
      hard_limits: false
      pulloff_mm: 3
      stepstick:
        direction_pin: gpio.14
        step_pin: gpio.17

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  cs_pin: gpio.5
  card_detect_pin: NO_PIN

coolant:
  flood_pin: gpio.15

probe:
  pin: gpio.13:low:pu

user_outputs:
  digital0_pin: gpio.12

PWM:
  pwm_hz: 5000
  output_pin: gpio.2
  enable_pin: gpio.4
  disable_with_s0: false
  s0_with_disable: true
  speed_map: 0=0% 10000=100%

Espero les sirva, saludos amigos

MitchBradley, 2022-10-23:
@Kenedy-Hanes that's very clever.  Thanks for sharing it with us.

sotchrys, 2022-11-10:
after version 3.6.1, any external card that works with 5 volts does not work. The solution is if you have this card, make this modification by short-circuiting the input with the output. I did it with 2 pins and a jamber

https://user-images.githubusercontent.com/3457857/201054545-5565afd1-c464-4997-af4c-b8a45cfe7883.mp4

Ruslan13rus, 2022-11-24:
> после версии 3.6.1 не работает любая внешняя карта, работающая с 5 вольтами. Решение: если у вас есть эта карта, сделайте эту модификацию, замкнув вход с выходом. Я сделал это с 2 пинами и перемычкой или купил простую карту без каких-либо электронных компонентов только для работы блока питания на 3,3 вольта. Я попробовал оба, и они отлично работают.
> 
>  sd_mod.mp4

Video doesn't work

MitchBradley, 2022-11-24:
Add a jumper as shown, or just put a blob of solder across the two pins.

[image]

Ruslan13rus, 2022-11-24:
> Add a jumper as shown, or just put a blob of solder across the two pins.
> 
> [image]

It doesn't help

MitchBradley, 2022-11-24:
You will just have to debug it yourself, unless some other user is willing to help you.  The developers do not have the time to debug custom hardware.

DEST1981, 2022-11-24:
> > после версии 3.6.1 не работает любая внешняя карта, работающая с 5 вольтами. Решение: если у вас есть эта карта, сделайте эту модификацию, замкнув вход с выходом. Я сделал это с 2 пинами и перемычкой или купил простую карту без каких-либо электронных компонентов только для работы блока питания на 3,3 вольта. Я попробовал оба, и они отлично работают.
> > sd_mod.mp4
> 
> Video doesn't work. How to connect

Руслан, с преобразователем уровня работать не будет, ногоногая микросхемка. То что там замыкают это линейный регулятор напряжения, от него не холодно не жарко. Можете его не трогать, если подключаете модуль 5ти вольтам. Если используете 3 вольта от ESP, тогда его лучше из схемы исключить.
Я бы просто перемычек напаял от разъема прямо к картоприёмнику. По взрослому, надо либо другой модуль, без преобразователя, либо убрать из схемы преобразователь.
А еще попробуйте другую карту памяти. У меня была проблема с картой 2гб. Взял на 16 и о чудо всё завелось. Причем на старых прошивках содержимое отображалось корректно, но запись приводила к ошибкам.

MitchBradley, 2022-11-24:
http://wiki.fluidnc.com/en/config/sd_card#supported-sd-cards

Ruslan13rus, 2022-11-24:
[image]
[image]
Файлы для работы карты отличаются, в новых версиях меньше строк. Кто нибудь попробуйте скомпелировать версию 3.6.4 заменив перед этим файлы SDCard.h и SDCard.cpp из версии 3.6.1(я просто не знаю как скомпелировать (

sotchrys, 2022-11-25:
После версии 3.6.1 не работают карты, имеющие напряжение питания 5 вольт.
У меня была проблема, и я решил ее, как показано на фото выше.

closed 2022-11-27 by sotchrys