bdring/FluidNC #660: Problem: YL620 VFD and RS485 communication

HenrykXXX, 2022-10-11:
### Controller Board

Custom Dev Board

### Help From Board Vendor

- [ ] Yes
- [ ] No
- [X] Not Applicable

### Machine Description

Gantry router with 3x TB6600 motor drivers + Chinesse water cooled 2.2kw spindle with YL620 VFD

### Configuration file

```yaml
name: "6050 3-axes"
board: "ESP32 Dev Controller V4"

start:
  must_home: false
  deactivate_parking: false
  check_limits: false

stepping:
  engine: RMT
  idle_ms: 255
  dir_delay_us: 1
  pulse_us: 4
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: gpio.13:high
  
  x:
    steps_per_mm: 160
    max_rate_mm_per_min: 5000
    acceleration_mm_per_sec2: 200
    max_travel_mm: 1000
    homing:
      cycle: 0
    
    motor0:
      limit_all_pin: NO_PIN
      stepstick:
        direction_pin: gpio.32:high
        step_pin: gpio.33:high

  y:
    steps_per_mm: 200
    max_rate_mm_per_min: 5000
    acceleration_mm_per_sec2: 200
    max_travel_mm: 1000
    homing:
      cycle: 0

    motor0:
      limit_all_pin: NO_PIN
      stepstick:
        direction_pin: gpio.25:high
        step_pin: gpio.26:high

  z:
    steps_per_mm: 200
    max_rate_mm_per_min: 5000
    acceleration_mm_per_sec2: 200
    max_travel_mm: 1000
    homing:
      cycle: 0
    motor0:
      limit_all_pin: NO_PIN
      stepstick:
        direction_pin: gpio.27:low
        step_pin: gpio.14:high

coolant:
  flood_pin: gpio.22
  mist_pin:  gpio.23

        
probe:
  pin: gpio.4:low:pu

YL620:
  uart:
    txd_pin: gpio.16
    rxd_pin: gpio.17
    rts_pin: gpio.5
    baud: 9600
    mode: 8N1
  modbus_id: 1
  tool_num: 0
  speed_map: 0=0% 0=25% 6000=25% 24000=100%
  off_on_alarm: false
```

### Startup Messages

```text
VFD RS-485 Unresponsive
```

### User Interface Software

UGS

### What happened?

Hello!

I am trying to connect FluidNC controller to YL620 VFD via rs-485 and it does not want to work. I am using pins 5, 16 and 17 for UART output and uart to rs-485 converter board. Even with converter board disconnected when i check activity on esp32 pins with my scope i can see that rx and tx pins are staying high and rts pin pulses high every second or so. There is no frames beeing send from controller. VFD is set correctly to work with rs485 input, I have no idea what is happening but there is some issue here, i hope we can figgure it out!

Have a great day!

bdring, 2022-10-11:
Show the schematic and a photo of your custom controller.

HenrykXXX, 2022-10-11:
I separated it there is nothing more then dev board connected to rs adapter powered via usb
[image]

bdring, 2022-10-11:
I hooked up a logic analyzer to my controller. It seems to be sending TX and RTS

I used these pins because they are easier to connect.

```yaml
YL620:
  uart:
    txd_pin: gpio.25
    rxd_pin: gpio.27
    rts_pin: gpio.26
    baud: 9600
    mode: 8N1
  modbus_id: 1
  tool_num: 0
  speed_map: 0=0% 0=25% 6000=25% 24000=100%
  off_on_alarm: false
```
[image]

[image]

HenrykXXX, 2022-10-11:
Well i can see the rts  pin long pulse but there is no data pulses on my tx pin, i will try different pin tomorrow.

bdring, 2022-10-11:
Try it without the RS485 thingy attached.

MitchBradley, 2022-10-11:
Here is some information about how to make the connection at the VFD end.  All the actual connection information that you gave us stops at the little adapter thing, and that is a schematic not a photo, so if you did something obviously wrong we have no way of knowing.  https://forum.linuxcnc.org/24-hal-components/39128-yalang-yl620-vfd-rs485-modbus-communicaiton

HenrykXXX, 2022-10-11:
Ok so i coudln't sleep and i did some tests, when i am using pin 26 for tx it works fine and i can see data transfered while RTS goes high
[image]

but with pin 17 for tx it doenst work. Are You sure that pin multiplexing can use any pin for uart combinations? I have no experience with esp32 but with stm or atmels there are pre deffined combinations to use with uarts/spi/i2c meaby that is the case?

MitchBradley, 2022-10-11:
The ESP32 pin mux is fairly general but I will look closely at the ref manual to see if I can discover anything.

MitchBradley, 2022-10-11:
The other key info we don't have is the parameter settings on your VFD.

bdring, 2022-10-11:
I just loaded your original config file to an ESP32 and it worked fine.

[image]

Maybe you damaged a pin on the ESP32.

BTW: You did not load your full startup messages. Can you add those via edit to your first message.

MitchBradley, 2022-10-11:
Also, we need all the startup messages, not just the one that might seem relevant to you.  There can be important information buried in them that you might not have noticed.

HenrykXXX, 2022-10-11:
VFD Settings:

P00.01	3	
P03.00	3	
P03.01	1	
P03.02	2	

I am pretty sure this is correct, i am testing it with a scope now just to see if mcu sends data via pins so it was unppluged for that. From what i can tell pin 16 and 17 doesnt work for me as tx pin... pin 26 works fine.

I just checked with new dev board that i just unpacked now and there is same issue with pin 16 and 17 so i don't think that is because of hardvere dammage.

I went through ref manual for esp32 quickly and i can't find any restrictions when it comes to pin mux for uarts so i have no idea what is the issue. 

I know that pins 16 and 17 work fine for pwm controll and there is no problem there for me. 

Hm here is exact dev board that i am using:

https://www.tme.eu/pl/details/esp32-devkitc-ve/zestawy-uruchomieniowe-pozostale/espressif/?fbclid=IwAR2hG3rh8FBt-9uTF4gPb1QdamdOvS03kN48pg5Jq-mcCCmOP2p1eTkhzzk

bdring, 2022-10-11:
Those might be PSRAM versions. I'll look for a schematic and datasheet on them.

MitchBradley, 2022-10-11:
[image]

WROVER is indeed PSRAM.  PSRAM uses some of the pins that would otherwise be available for other purposes.  We support only WROOM modules, without PSRAM.

bdring, 2022-10-11:
We don't need the ram and it uses up valuable pins.

[image]

HenrykXXX, 2022-10-11:
Ok that explains my issue. Yes other pins work no problem i just checked. I think i have enough pins for my machine anyway so i will just avoid using this ones.

Thanks for help!  I hope i didn't waste that much of Your time. Meaby this can help someone in the future who will buy this dev board.

closed 2022-10-11 by bdring

qulet, 2024-10-03:
Czy konverter nie powinien być zasilany 3,3V ? [HenrykXXX] pisze że zasila go z usb 5V. Ja walcze z tym samym problemem konwerter mam zasilany 3,3 V  FVD YL620-A  ustawienia VFD:
VFD YL620-A
P00.01 3 Sterowanie RS485
P03.00 3 transmisja 9600
P03.01 1 identyfikator
P03.02 2 8 1 Brak
P03.03 3 Błąd komunikacji Brak zatrzymania
P07.08 5 częstotliwość sterowana przez RS485
podmieniałem A i B
Wrzeciono nie kręci.
Sprawdzałem  programem ModbusMAT 1.1 i przejściówką  USB RTS485  wrzeciono reaguje na wysłane ramki:
Start [01][06][20][00][00][02][03][CB]
stop [01][06][20][00][00][01][43][CA] 
więc problem jest chyba w działaniu konwertera TTL RTS485 
plik configuracji Fluid jest dobrze zdefiniowany.

uart1:
  txd_pin: gpio.25
  rxd_pin: gpio.27
  rts_pin: gpio.26
  baud: 9600
  mode: 8N1

YL620:
  uart_num: 1
  modbus_id: 1
  tool_num: 0
  speed_map: 0=0% 0=25% 6000=25% 24000=100%
  off_on_alarm: false

Nie mam oscyloskopu żeby sprawdzić czy zdefiniowany  Gpio Tx  na ESP  po wydaniu Gcode np M3 S10000 coś wysyła. 
Jak to mogę sprawdzić innym sposobem  czy np dioda Led podłaczona do GND i pinu GPIO 25 powinna zaświecić ?
Dziękuję za wszystkie wskazówki .

Shouldn't the converter be powered by 3.3V? [HenrykXXX] writes that he powers it from a 5V USB. I'm struggling with the same problem, my converter is powered by 3.3V FVD YL620-A VFD settings:
VFD YL620-A
P00.01 3 RS485 control
P03.00 3 transmission 9600
P03.01 1 identifier
P03.02 2 8 1 None
P03.03 3 Communication error No stop
P07.08 5 frequency controlled by RS485
I swapped A and B
The spindle doesn't turn. I checked with the ModbusMAT 1.1 program and the RTS485 USB adapter, the spindle responds to the frames sent:
Start [01][06][20][00][00][02][03][CB]
stop [01][06][20][00][00][01][43][CA]
so the problem is probably in the operation of the RTS485 TTL converter
the Fluid configuration file is well defined.

uart1:
txd_pin: gpio.25
rxd_pin: gpio.27
rts_pin: gpio.26
baud: 9600
mode: 8N1

YL620:
uart_num: 1
modbus_id: 1
tool_num: 0
speed_map: 0=0% 0=25% 6000=25% 24000=100%
off_on_alarm: false

I don't have an oscilloscope to check if the defined Gpio Tx on ESP after issuing Gcode e.g. M3 S10000 sends anything.
How can I check it in another way if e.g. the LED diode connected to GND and GPIO pin 25 should light up?
Thank you for all the tips.