bdring/FluidNC #661: Install FluidNC on Esp32S3

xs23933, 2022-10-11:
### Machine Context

/dev/cu.usbmodem1444301 failed to connect: This chip is ESP32-S3(beta3) not ESP32. Wrong --chip argument?

install-fs stoped.

### Feature Description

i well be try install-fs.sh 
The shell automatically skips this usb device ESP32-S3-Module

/dev/cu.usbmodem1444301 failed to connect: This chip is ESP32-S3(beta3) not ESP32. Wrong --chip argument?

install-fs stoped.

### Other Approaches

none

### How I Can Help

none

bdring, 2022-10-11:
We do not support the S3 chip at this time.

Easix, 2022-10-16:
Can you support esp32s3 ??? @bdring

MitchBradley, 2022-10-16:
We will eventually, but it will not happen instantly.

MitchBradley, 2022-10-16:
It will happen sooner if we get funding for the task.  Supporting a new processor variant is expensive because we need to acquire or build hardware, learn about the quirks of the new chip, answer questions for other people who are learning about the chip and building their own hardware, write documentation, and many other tasks that take a lot of time.

closed 2022-10-16 by MitchBradley

borthtd, 2023-01-11:
Does "closed" mean you now support ESP32S3?
If so, where are the files?

If not, timeline/probability of supporting?

MitchBradley, 2023-01-11:
No, closed means that there is no point in keeping the issue open.  It is something that we hope to do but it will happen when it happens.  This project has many requests for features and lots of support requests - often for custom hardware built by inexperienced people - so the developers are tapped out.  We could use help, especially the kind where the person offering to help actually follows through in a timely fashion.

borthtd, 2023-01-11:
Thanks for answering. I wish I could level up past Noob and actually help!! Thanks to all you DEVs do!!!