bdring/FluidNC #662: hard Limit "Reset to continue" but switch not pressed.

ashokavarthanan, 2022-10-11:
### Controller Board

4_Axis_External_Driver

### Help From Board Vendor

- [ ] Yes
- [ ] No
- [X] Not Applicable

### Machine Description

4 axis , 1.5kw Spindle.

### Configuration file

```yaml
name: "ESP32 Dev Controller V4"
board: "ESP32 Dev Controller V4"

stepping:
  engine: RMT
  idle_ms: 250
  dir_delay_us: 1
  pulse_us: 2
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: gpio.15:high
  
  x:
    steps_per_mm: 240
    max_rate_mm_per_min: 2000
    acceleration_mm_per_sec2: 50
    max_travel_mm: 1000
    homing:
      cycle: 2
      allow_single_axis: true
      mpos_mm: 5.00
      seek_mm_per_min: 500
      feed_mm_per_min: 200
      positive_direction: false
    
    motor0:
      limit_neg_pin: gpio.35:low
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 5.00
      stepstick:
        direction_pin: gpio.13
        step_pin: gpio.12
    motor1:
      null_motor:

  y:
    steps_per_mm: 240
    max_rate_mm_per_min: 2000
    acceleration_mm_per_sec2: 50
    max_travel_mm: 1000
    homing:
      cycle: 2
      allow_single_axis: true
      mpos_mm: 5.00
      seek_mm_per_min: 500
      feed_mm_per_min: 200
      positive_direction: false

    motor0:
      limit_neg_pin: gpio.32:low:pu
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 5.00
      stepstick:
        direction_pin: gpio.27
        step_pin: gpio.14
    motor1:
      null_motor:

  z:
    steps_per_mm: 120
    max_rate_mm_per_min: 1000
    acceleration_mm_per_sec2: 25
    max_travel_mm: 1000

    motor0:
      limit_all_pin: NO_PIN
      stepstick:
        direction_pin: gpio.25:low
        step_pin: gpio.26
    motor1:
      null_motor:

  a:
    steps_per_mm: 240
    max_rate_mm_per_min: 1000
    acceleration_mm_per_sec2: 25
    max_travel_mm: 1000

    motor0:
      limit_all_pin: NO_PIN
      stepstick:
        direction_pin: gpio.17
        step_pin: gpio.33
    motor1:
      null_motor:
spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  cs_pin: gpio.5
  card_detect_pin: NO_PIN

coolant:
  flood_pin: NO_PIN
  mist_pin:  NO_PIN

probe:
  pin: NO_PIN

PWM:
  pwm_hz: 5000
  output_pin: gpio.2
  enable_pin: gpio.22
  direction_pin: NO_PIN
  disable_with_s0: false
  s0_with_disable: true
  spinup_ms: 0
  spindown_ms: 0
  tool_num: 0
  speed_map: 0=0% 10000=100%
```

### Startup Messages

```text
[MSG:INFO: FluidNC v3.6.3]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
[MSG:INFO: Local filesystem type is spiffs]
[MSG:INFO: Configuration file:config.yaml]
[MSG:INFO: Machine ESP32 Dev Controller V4]
[MSG:INFO: Board ESP32 Dev Controller V4]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN]
[MSG:INFO: Stepping:RMT Pulse:2us Dsbl Delay:0us Dir Delay:1us Idle Delay:250ms]
[MSG:INFO: Axis count 4]
[MSG:INFO: Shared stepper disable gpio.15]
[MSG:INFO: Axis X (5.000,1005.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:gpio.12 Dir:gpio.13 Disable:NO_PIN]
[MSG:INFO:  X Neg Limit gpio.35:low]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Y (5.000,1005.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:gpio.14 Dir:gpio.27 Disable:NO_PIN]
[MSG:INFO:  Y Neg Limit gpio.32:low:pu]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Z (-1000.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:gpio.26 Dir:gpio.25:low Disable:NO_PIN]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis A (-1000.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:gpio.33 Dir:gpio.17 Disable:NO_PIN]
[MSG:INFO:   Motor1]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: PWM Spindle Ena:gpio.22 Out:gpio.2 Dir:NO_PIN Freq:5000Hz Period:8191]
[MSG:INFO: Using spindle PWM]
[MSG:INFO: BT Started with FluidNC]
```

### User Interface Software

cncjs

### What happened?

hard limit when g-code execution?
Problem: hard Limit "Reset to continue" but switch not pressed.

ashokavarthanan, 2022-10-11:
[image]
Limit switch diagram .

MitchBradley, 2022-10-11:
We need to know what is on the controller end of the limit switch circuit too.  The config file says "ESP32 Dev Controller V4" which suggest one of Bart's boards, but since that field is just informational, we don't know if that is really the board or if you are using something custom.

ashokavarthanan, 2022-10-11:
yes customized 
z axis Homing removed.
x and y axis limit pins are changed (due to esp32s model gpio36 and gpio39 not worked)
Limit switch works fine when started.
Homing $H works fine without any issue (xy only).

pin changed

MitchBradley, 2022-10-11:
Limit switches need pullup resistors. https://learn.sparkfun.com/tutorials/pull-up-resistors/all

Some of the ESP32 input have internal pullup resistors that you can enable by adding :pu to the gpio spec in the config files, but GPIOs 34-39 do not.  http://wiki.fluidnc.com/en/hardware/esp32_pin_reference

Even for the GPIOs where :pu works, it is better to have 10K external pullups because the internal pullups are very high resistance, often making the limits subject to noise pickup.

ashokavarthanan, 2022-10-11:
> Even for the GPIOs where :pu works, it is better to have 10K external pullups because the internal pullups are very high resistance, often making the limits subject to noise pickup.

ok i will update my controller board using external pullups then try.
Thanks.

MitchBradley, 2022-10-11:
Yes, but you should really read the stuff that is on the internet.  We are not your unpaid personal electronics tutors.