bdring/FluidNC #665: Feature: Improve spindle (laser) spinup & spindown wait time in dynamic power mode

dan21dan, 2022-10-12:
### Machine Context

I have build my own laser engraver based on wemos + CNC shield. It works fine. Now I add a relay that switches on the laser and its cooling. The relay is controlled by "Laser - enable_pin" signal.

### Feature Description

I thought that setting spinup_ms would make the machine wait a given time before using the laser. Unfortunately, when used in dynamic power mode, there is almost no wait (although I set the value to the maximum, i.e. 20000). Therefore, I propose to add another variable that allows to set the time that must elapse before the spindle (laser) starts to be used, regardless of the type of power for the spindle is set or in which mode I want to use the spindle. Could be called e.g. spinstart_ms or warmup_ms.

On the other hand, when the spindown_ms variable is set, the voltage is on the "Laser - enable_pin" for a set amount of time after the job is completed (even in dynamic power mode). Which is correct in my opinion. I would just ask for the possibility to increase the limit of the value that can be set. I suggest to increase the limit to 5 minutes (i.e. 300000). Alternatively, add another variable to the config to set the time that the voltage on "Laser - enable_pin" will be on after the job is finished (i.e. after the M5 command is called). Could be called e.g. spinstop_ms or shutdown_ms.

### Other Approaches

I don't have an alternative solution for "spinstart_ms".

"spinstop_ms" alternative solution: I add an RC circuit between the "Laser - enable_pin" signal and the relay, which will delay the laser power off. The disadvantage is that this will not be adjustable.

### How I Can Help

I can test it during development. I will write the documentation.

bdring, 2022-10-12:
Please describe exactly what you want to the machine to do and why. Show a few lines of your gcode and comment it to say what you want to happen at each line.

You are proposing a solution to a poorly define problem, so we cannot comment it.

MitchBradley, 2022-10-12:
You could put the laser enable on a user digital output and control it with M62 in GCode, using explicit delays in gcode too.

dan21dan, 2022-10-12:
For the sake of completeness, I needed the first command to be executed after some time after the laser was switched on (to make sure that the laser control board is already completely initialized, since its power supply is only triggered by a signal "Laser - enable").
After the engraving was finished, I wanted the laser power supply to be still on for some time to allow the laser to cool down a bit.
I didn't know there was a delay G code (G04). To solve my request, I just need to add G04 with time to the header and footer in the gcode generator. So **command G04 solves this issue**.

I apologize for creating this issue that you had to deal with unnecessarily. Thanks to MitchBradley for his reply, which helped me find the G04 command.

Thank you for the development and support FuildNC.

closed 2022-10-12 by dan21dan