bdring/FluidNC #666: Problem: Fluid NC reports NO SD Card with Error 60

enarges, 2022-10-12:
### Controller Board

6-pack

### Help From Board Vendor

- [ ] Yes
- [ ] No
- [X] Not Applicable

### Machine Description

MP CNC with dual X, Y, and sinlge Z endstop.  Relay & PWM Spindle.

### Configuration file

```yaml
board: 6 Pack
name: 6 Pack TMC2130 XXYYZ
stepping:
  engine: I2S_STREAM
  idle_ms: 250
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: NO_PIN
  x:
    steps_per_mm: 200.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 200.000
    max_travel_mm: 420.000
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 0.0
      feed_mm_per2min: 1500.000
      seek_mm_per_min: 2000.000
      settle_ms: 200
      seek_scaler: 1.100
      feed_scaler: 1.100
    motor0:
      limit_neg_pin: gpio.33:high
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 10.000
      tmc_2130:
        cs_pin: i2so.3
        run_amps: 0.750
        hold_amps: 0.5
        microsteps: 32
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_StealthChop: 3
        run_mode: CoolStep
        homing_mode: CoolStep
        use_enable: false
        step_pin: I2SO.2
        direction_pin: I2SO.1
        disable_pin: I2SO.0

    motor1:
      limit_neg_pin: gpio.32:high
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 10.000
      tmc_2130:
        cs_pin: i2so.6
        run_amps: 0.75
        hold_amps: 0.20
        microsteps: 32
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_StealthChop: 3
        run_mode: CoolStep
        homing_mode: CoolStep
        use_enable: false
        step_pin: I2SO.5
        direction_pin: I2SO.4
        disable_pin: I2SO.7

  y:
    steps_per_mm: 200.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 200.000
    max_travel_mm: 320.000
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 0.0
      feed_mm_per2min: 1500.000
      seek_mm_per_min: 2000.000
      settle_ms: 200
      seek_scaler: 1.100
      feed_scaler: 1.100
    motor0:
      limit_neg_pin: gpio.35:high
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 10.000
      tmc_2130:
        cs_pin: i2so.11
        run_amps: 0.750
        hold_amps: 0.5
        microsteps: 32
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_StealthChop: 3
        run_mode: CoolStep
        homing_mode: CoolStep
        use_enable: false
        step_pin: I2SO.10
        direction_pin: I2SO.9
        disable_pin: I2SO.8

    motor1:
      limit_neg_pin: gpio.34:high
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 10.000
      tmc_2130:
        cs_pin: i2so.14
        run_amps: 0.75
        hold_amps: 0.20
        microsteps: 32
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_StealthChop: 3
        run_mode: CoolStep
        homing_mode: CoolStep
        use_enable: false
        step_pin: I2SO.13
        direction_pin: I2SO.12
        disable_pin: I2SO.15

  z:
    steps_per_mm: 800
    max_rate_mm_per_min: 80.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 85.000
    soft_limits: false
    homing:
      cycle: 1
      positive_direction: true
      mpos_mm: 150.000
      feed_mm_per_min: 200.000
      seek_mm_per_min: 400.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: gpio.25:high
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 3.000
      tmc_2130:
        cs_pin: i2so.19
        run_amps: 0.50
        hold_amps: 0.20
        microsteps: 16
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_StealthChop: 3
        run_mode: CoolStep
        homing_mode: CoolStep
        use_enable: false
        step_pin: I2SO.18
        direction_pin: I2SO.17:low
        disable_pin: I2SO.16

i2so:
  bck_pin: gpio.22
  data_pin: gpio.21
  ws_pin: gpio.17

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  cs_pin: gpio.5
  card_detect_pin: NO_PIN

pwm:
  pwm_hz: 5000
  direction_pin: NO_PIN
  output_pin: gpio.26
  enable_pin: gpio.14
  disable_with_s0: false
  s0_with_disable: true
  spinup_ms: 0
  spindown_ms: 0
  tool_num: 0
  speed_map: 0=0.000% 3000=25% 6000=50% 9000=75% 12000=100.000%

probe:
  pin: NO_PIN
  check_mode_start: true

macros:
  startup_line0:
  startup_line1:
  macro0:
  macro1:
  macro2:
  macro3:

start:
  must_home: false
```

### Startup Messages

```text
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:12812
load:0x40080400,len:3032
entry 0x400805e4

[MSG:INFO: FluidNC v3.6.3]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
[MSG:INFO: Local filesystem type is spiffs]
[MSG:INFO: Configuration file:config.yaml]
[MSG:WARN: Ignored key feed_mm_per2min]
[MSG:WARN: Ignored key feed_mm_per2min]
[MSG:INFO: Machine 6 Pack TMC2130 XXYYZ]
[MSG:INFO: Board 6 Pack]
[MSG:INFO: I2SO BCK:gpio.22 WS:gpio.17 DATA:gpio.21]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN]
[MSG:INFO: Stepping:I2S_stream Pulse:4us Dsbl Delay:0us Dir Delay:1us Idle Delay:250ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (0.000,420.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_2130 Step:I2SO.2 Dir:I2SO.1 CS:I2SO.3 Disable:I2SO.0 Index:-1 R:0.110]
[MSG:INFO:  X Neg Limit gpio.33]
[MSG:INFO:   Motor1]
[MSG:INFO:     tmc_2130 Step:I2SO.5 Dir:I2SO.4 CS:I2SO.6 Disable:I2SO.7 Index:-1 R:0.110]
[MSG:INFO:  X2 Neg Limit gpio.32]
[MSG:INFO: Axis Y (0.000,320.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_2130 Step:I2SO.10 Dir:I2SO.9 CS:I2SO.11 Disable:I2SO.8 Index:-1 R:0.110]
[MSG:INFO:  Y Neg Limit gpio.35]
[MSG:INFO:   Motor1]
[MSG:INFO:     tmc_2130 Step:I2SO.13 Dir:I2SO.12 CS:I2SO.14 Disable:I2SO.15 Index:-1 R:0.110]
[MSG:INFO:  Y2 Neg Limit gpio.34]
[MSG:INFO: Axis Z (65.000,150.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_2130 Step:I2SO.18 Dir:I2SO.17:low CS:I2SO.19 Disable:I2SO.16 Index:-1 R:0.110]
[MSG:INFO:  Z Pos Limit gpio.25]
[MSG:ERR: X Axis TMC driver not detected - expected 0x11 got 0x0]
[MSG:ERR: X2 Axis TMC driver not detected - expected 0x11 got 0x0]
[MSG:ERR: Y Axis TMC driver not detected - expected 0x11 got 0x0]
[MSG:ERR: Y2 Axis TMC driver not detected - expected 0x11 got 0x0]
[MSG:ERR: Z Axis TMC driver not detected - expected 0x11 got 0x0]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: PWM Spindle Ena:gpio.14 Out:gpio.26 Dir:NO_PIN Freq:5000Hz Period:8191]
[MSG:INFO: Using spindle PWM]
[MSG:INFO: Connecting to STA SSID:Sirius]
[MSG:INFO: Connecting.]
[MSG:INFO: Connected - IP is 192.168.86.29]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
```

### What happened?

Trying to use a FAT32 formattted SD-CARD results in:

E (33918) sdmmc_sd: sdmmc_init_sd_if_cond: send_if_cond (1) returned 0x108
[MSG:ERR: sdmmc_card_init failed code 0x108]
No SD card
error:60

MitchBradley, 2022-10-12:
Probably not related to the SD issue, but you might as well fix the typos in the config file:
[MSG:WARN: Ignored key feed_mm_per2min]
[MSG:WARN: Ignored key feed_mm_per2min]

bdring, 2022-10-12:
With power off, try removing the stepper drivers. Then power on and test again. A stepper driver may be interfering.

MitchBradley, 2022-10-12:
1. Do you have external power on so the green light is on ?
2. Is the VCC jumper set to 3.3V ?
3. Are the three jumpers that select for TMC drivers vs stepstick mode select set for TMC?

enarges, 2022-10-12:
With All Stepper Drivers removed, the SD card is detected

bdring, 2022-10-12:
Send a close up photo of a driver.

enarges, 2022-10-12:
[image]
[image]

bdring, 2022-10-12:
They look like the correct type.

Put them all in the controller and send a hi resolution photo of the entire controller with power on.

enarges, 2022-10-12:
[image]

bdring, 2022-10-12:
First observation.

 - The Vcc should be on the 3.3V side.

enarges, 2022-10-12:
Put all drivers back in, and the now the all the SD cards are working.

enarges, 2022-10-12:
Main power was off.  Just connected micro usb to the esp in the board.  I had brought it up out of the shop to debug this issue.

MitchBradley, 2022-10-12:
A misinserted driver can load down the shared signals on the SPI bus and prevent the SD card from working.

enarges, 2022-10-12:
Weird thing is now Y will run for a minute to home, then the Y1 stepper will randomly reverse.  Maybe a bad stepper driver, though it worked fine before I removed them, just had no SD.

MitchBradley, 2022-10-12:
If you have homing problems, test with FluidTerm.  Set $message/level=debug and issue a homing command like $hy .  Show us the messages from the terminal.

enarges, 2022-10-12:
Grbl 3.6 [FluidNC v3.6.3 (wifi) '$' for help]
ok
ok
$10=1
ok
$message/level=debug
ok
$hy
[MSG:DBG: Homing Cycle Y]
[MSG:DBG: Y Axis Coolstep]
[MSG:DBG: Y2 Axis Coolstep]
[MSG:DBG: Homing nextPhase FastApproach]
[MSG:DBG: Starting from 0.000,-252.000,0.000]
[MSG:DBG: Planned move to 0.000,-604.000,0.000 @ 2000.000]
[MSG:DBG: CycleStop FastApproach]
[MSG:DBG: Y Axis Coolstep]
[MSG:DBG: Y2 Axis Coolstep]
<Home|MPos:0.000,-604.000,0.000|FS:0,0|Ov:100,100,100>
ALARM:9

enarges, 2022-10-12:
Meanwhile $hx reports:
[MSG:DBG: Homing Cycle X]
[MSG:DBG: X Axis Coolstep]
[MSG:DBG: X2 Axis Coolstep]
[MSG:DBG: Homing nextPhase FastApproach]
[MSG:DBG: Starting from 0.000,-604.000,0.000]
[MSG:DBG: Planned move to -462.000,-604.000,0.000 @ 2000.000]
[MSG:DBG:  X2 Neg Limit 1]
[MSG:DBG: Homing limited X2]
[MSG:DBG:  X Neg Limit 1]
[MSG:DBG: Homing limited X X2]
[MSG:DBG: Homing nextPhase Pulloff0]
[MSG:DBG: Starting from -10.020,-604.000,0.000]
[MSG:DBG: Planned move to -0.020,-604.000,0.000 @ 50.000]
[MSG:DBG:  X Neg Limit 0]
[MSG:DBG:  X2 Neg Limit 0]
[MSG:DBG: CycleStop Pulloff0]
[MSG:DBG: Homing nextPhase SlowApproach]
[MSG:DBG: Starting from -0.020,-604.000,0.000]
[MSG:DBG: Planned move to -11.020,-604.000,0.000 @ 50.000]
[MSG:DBG:  X2 Neg Limit 1]
[MSG:DBG: Homing limited X2]
[MSG:DBG:  X Neg Limit 1]
[MSG:DBG: Homing limited X X2]
[MSG:DBG: Homing nextPhase Pulloff1]
[MSG:DBG: Starting from -9.400,-604.000,0.000]
[MSG:DBG: Planned move to 0.600,-604.000,0.000 @ 50.000]
[MSG:DBG:  X Neg Limit 0]
[MSG:DBG:  X2 Neg Limit 0]

enarges, 2022-10-12:
When Y homes now it moves about 50-100mm and then one of the Y steppers reverses direction and they start fighting against each other.  This did not happen before.  95% of the time jogging Y is normal, now randomly exhibits the same behavior.

MitchBradley, 2022-10-12:
Did it reach the y limit switches?

enarges, 2022-10-12:
Nope, way a long distance away (100-150mm)

MitchBradley, 2022-10-12:
That seems like a bad connection on the Y2 direction pin.

MitchBradley, 2022-10-12:
All of these problems could be explained by dodgy solder joints where the 6 pack board connects to the header sockets that the drivers plug into.

enarges, 2022-10-12:
Problem appears to be the stepper driver.  I swapped Y1 with Z, and now z randomly reverses and Y an X work normally.  I’ll swap it out.

MitchBradley, 2022-10-12:
The traces and IC pads on those 2130 drivers are extremely tiny.  It is amazing that they ever work.  Can we close this ticket?

enarges, 2022-10-12:
I believe so.

closed 2022-10-12 by MitchBradley