bdring/FluidNC #667: Problem: Y-Axis not homing / moving correctly after upgrade from 3.5.1 to 3.6.3

ModuloFS, 2022-10-12:
### Controller Board

self build

### Help From Board Vendor

- [ ] Yes
- [ ] No
- [X] Not Applicable

### Machine Description

Gantry router Ratrig Killerbee with DM556T stepper driver, dual y motors and limit switches.

### Configuration file

```yaml
name: "ModuloController"
board: "ModuloControl V1"

stepping:
  engine: RMT
  idle_ms: 255
  dir_delay_us: 1
  pulse_us: 5
  disable_delay_us: 5
  
start:
  must_home: true
  deactivate_parking: true
  check_limits: true

axes:
  shared_stepper_disable_pin: gpio.23
  
  x:
    steps_per_mm: 400.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 150.000
    max_travel_mm: 510.000
    soft_limits: true
    
    homing:
      cycle: 2
      mpos_mm: 510.000
      feed_mm_per_min: 200.000
      seek_mm_per_min: 400.000
      positive_direction: true

    motor0:
      limit_pos_pin: gpio.35:low
      hard_limits: true
      pulloff_mm: 5.000
      stepstick:
        direction_pin: gpio.27
        step_pin: gpio.26

  y:
    steps_per_mm: 400.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 150.000
    max_travel_mm: 750.000
    soft_limits: true

    homing:
      cycle: 3
      mpos_mm: 750.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 400.000
      positive_direction: true

    motor0:
      limit_pos_pin: gpio.39:low
      hard_limits: true
      stepstick:
        direction_pin: gpio.32:low
        step_pin: gpio.25

    motor1:
      limit_pos_pin: gpio.36:low
      hard_limits: true
      stepstick:
        direction_pin: gpio.22:low
        step_pin: gpio.21

  z:
    steps_per_mm: 400.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 150.000
    max_travel_mm: 150.000
    soft_limits: true
    
    homing:
      cycle: 1
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 400.000
      positive_direction: true

    motor0:
      limit_pos_pin: gpio.34:low
      hard_limits: true
      stepstick:
        direction_pin: gpio.18
        step_pin: gpio.19

probe:
  pin: gpio.4:high
  
control:
  safety_door_pin: gpio.33:low
  reset_pin: NO_PIN
  feed_hold_pin: NO_PIN

relay:
  direction_pin: NO_PIN
  output_pin: gpio.5
  enable_pin: NO_PIN
  disable_with_s0: false
  s0_with_disable: true
  spinup_ms: 0
  spindown_ms: 0
  tool_num: 0
  speed_map: 0=0.000% 0=100.000% 1=100.000%
```

### Startup Messages

```text
$ss
[MSG:INFO: FluidNC v3.6.3]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
[MSG:INFO: Local filesystem type is spiffs]
[MSG:INFO: Configuration file:config_ratrig_killerbee.yaml]
[MSG:INFO: Machine ModuloController]
[MSG:INFO: Board ModuloControl V1]
[MSG:INFO: SPI not defined]
[MSG:INFO: No SD Card CS Pin]
[MSG:INFO: See http://wiki.fluidnc.com/en/config/sd_card#sdfallbackcs-access-sd-without-a-config-file]
[MSG:INFO: Stepping:RMT Pulse:5us Dsbl Delay:5us Dir Delay:1us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Shared stepper disable gpio.23]
[MSG:INFO: Axis X (0.000,510.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:gpio.26 Dir:gpio.27 Disable:NO_PIN]
[MSG:INFO:  X Pos Limit gpio.35:low]
[MSG:INFO: Axis Y (0.000,750.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:gpio.25 Dir:gpio.32:low Disable:NO_PIN]
[MSG:INFO:  Y Pos Limit gpio.39:low]
[MSG:INFO:   Motor1]
[MSG:INFO:     stepstick Step:gpio.21 Dir:gpio.22:low Disable:NO_PIN]
[MSG:INFO:  Y2 Pos Limit gpio.36:low]
[MSG:INFO: Axis Z (0.000,150.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:gpio.19 Dir:gpio.18 Disable:NO_PIN]
[MSG:INFO:  Z Pos Limit gpio.34:low]
[MSG:INFO: safety_door_pin gpio.33:low]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Relay Spindle Ena:NO_PIN Out:gpio.5 Dir:NO_PIN]
[MSG:INFO: Using spindle Relay]
[MSG:INFO: Probe Pin: gpio.4]
[MSG:INFO: AP SSID FluidNC IP 192.168.178.1 mask 255.255.255.0 channel 1]
[MSG:INFO: AP started]
[MSG:INFO: WiFi on]
[MSG:INFO: Captive Portal Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
ok
```

### User Interface Software

WebUI, gSender

### What happened?

If homing of the y-axis with dual motor is started, the axis goes towards the limit switches, as soon as they get activated, the motors are stopping and not moving any more. In this state, the limit switches stays activated and the motors sounds as the direction pins are changed randomly. This behaviour starts after the upgrade from 3.5.1 to 3.6.3. Before the update everything worked fine.

MitchBradley, 2022-10-12:
Try it with FluidTerm.  Set $message/level=debug and then home with $hy .  Show us the messages from the terminal.

MitchBradley, 2022-10-13:
You do not have pulloff_mm defined for Y so it is defaulting to 1 mm, which is apparently too small for your machine.  The 1 mm pulloff move is not releasing the switch.

ModuloFS, 2022-10-13:
Ok, is there any difference between the "stepstick" definition in compare to the "standard stepper" definition?

MitchBradley, 2022-10-13:
stepstick has configurable mode select and reset pins; standard stepper does not.  standard stepper is basically an external driver where everything is configured with dip switches on the driver itself.

MitchBradley, 2022-10-13:
Can we close this ticket now?

ModuloFS, 2022-10-13:
Give me a second, i try it right now.

ModuloFS, 2022-10-13:
Ok, i changed the pulloff to 5 mm, the machine pulls off but not symetrically. If i start the homing after that again i get the same problem with the stopping stepper and the Alarm 8. So the Problem is not solved...

bdring, 2022-10-13:
Did you change the pull off for both motors on that axis?

ModuloFS, 2022-10-13:
Yes

bdring, 2022-10-13:
Show your current config file.

bdring, 2022-10-13:
Just send the $axes/y command. That is enough

ModuloFS, 2022-10-13:
name: "ModuloController"
board: "ModuloControl V1"

stepping:
  engine: RMT
  idle_ms: 255
  dir_delay_us: 1
  pulse_us: 5
  disable_delay_us: 5
  
start:
  must_home: true
  deactivate_parking: true
  check_limits: true

axes:
  shared_stepper_disable_pin: gpio.23
  
  x:
    steps_per_mm: 400.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 150.000
    max_travel_mm: 510.000
    soft_limits: true
    
    homing:
      cycle: 2
      mpos_mm: 510.000
      feed_mm_per_min: 200.000
      seek_mm_per_min: 400.000
      positive_direction: true

    motor0:
      limit_pos_pin: gpio.35:low
      hard_limits: true
      pulloff_mm: 5.000
      standard_stepper:
        direction_pin: gpio.27
        step_pin: gpio.26

  y:
    steps_per_mm: 400.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 150.000
    max_travel_mm: 750.000
    soft_limits: true

    homing:
      cycle: 3
      mpos_mm: 750.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 400.000
      positive_direction: true

    motor0:
      limit_pos_pin: gpio.39:low
      hard_limits: true
      pulloff_mm: 5.000
      standard_stepper:
        direction_pin: gpio.32:low
        step_pin: gpio.25

    motor1:
      limit_pos_pin: gpio.36:low
      hard_limits: true
      pulloff_mm: 5.000
      standard_stepper:
        direction_pin: gpio.22:low
        step_pin: gpio.21

  z:
    steps_per_mm: 400.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 150.000
    max_travel_mm: 150.000
    soft_limits: true
    
    homing:
      cycle: 1
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 400.000
      positive_direction: true

    motor0:
      limit_pos_pin: gpio.34:low
      hard_limits: true
      standard_stepper:
        direction_pin: gpio.18
        step_pin: gpio.19

probe:
  pin: gpio.4:high
  
control:
  safety_door_pin: gpio.33:low
  reset_pin: NO_PIN
  feed_hold_pin: NO_PIN

relay:
  direction_pin: NO_PIN
  output_pin: gpio.5
  enable_pin: NO_PIN
  disable_with_s0: false
  s0_with_disable: true
  spinup_ms: 0
  spindown_ms: 0
  tool_num: 0
  speed_map: 0=0.000% 0=100.000% 1=100.000%

ModuloFS, 2022-10-13:
I have changed now also the stepstick config to the standard stepper config.

ModuloFS, 2022-10-13:
Ok, now it seams solved, i changed the dir_delay_us to 5. And now it seems to work. Has there changed anything in the timing between the version 3.5.1 and 3.6.3?

ModuloFS, 2022-10-13:
Here is my config now:

name: "ModuloController"
board: "ModuloControl V1"

stepping:
  engine: RMT
  idle_ms: 255
  dir_delay_us: 5
  pulse_us: 5
  disable_delay_us: 5
  
start:
  must_home: true
  deactivate_parking: true
  check_limits: true

axes:
  shared_stepper_disable_pin: gpio.23
  
  x:
    steps_per_mm: 400.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 150.000
    max_travel_mm: 510.000
    soft_limits: true
    
    homing:
      cycle: 2
      mpos_mm: 510.000
      feed_mm_per_min: 200.000
      seek_mm_per_min: 400.000
      positive_direction: true

    motor0:
      limit_pos_pin: gpio.35:low
      hard_limits: true
      pulloff_mm: 5.000
      standard_stepper:
        direction_pin: gpio.27
        step_pin: gpio.26

  y:
    steps_per_mm: 400.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 150.000
    max_travel_mm: 750.000
    soft_limits: true

    homing:
      cycle: 3
      mpos_mm: 750.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 400.000
      positive_direction: true

    motor0:
      limit_pos_pin: gpio.39:low
      hard_limits: true
      pulloff_mm: 5.000
      standard_stepper:
        direction_pin: gpio.32:low
        step_pin: gpio.25

    motor1:
      limit_pos_pin: gpio.36:low
      hard_limits: true
      pulloff_mm: 5.000
      standard_stepper:
        direction_pin: gpio.22:low
        step_pin: gpio.21

  z:
    steps_per_mm: 400.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 150.000
    max_travel_mm: 150.000
    soft_limits: true
    
    homing:
      cycle: 1
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 400.000
      positive_direction: true

    motor0:
      limit_pos_pin: gpio.34:low
      hard_limits: true
      standard_stepper:
        direction_pin: gpio.18
        step_pin: gpio.19

probe:
  pin: gpio.4:high
  
control:
  safety_door_pin: gpio.33:low
  reset_pin: NO_PIN
  feed_hold_pin: NO_PIN

relay:
  direction_pin: NO_PIN
  output_pin: gpio.5
  enable_pin: NO_PIN
  disable_with_s0: false
  s0_with_disable: true
  spinup_ms: 0
  spindown_ms: 0
  tool_num: 0
  speed_map: 0=0.000% 0=100.000% 1=100.000%

bdring, 2022-10-13:
No change in that area. What motor controllers are you using?

ModuloFS, 2022-10-13:
I am using DM556T. Ok side hint, if i change the pulloff back to 1 mm i get the same problem again even with dir_delay_us 5. If i change the pulloff back to 2mm it works again...

bdring, 2022-10-13:
Send a photo of your switches and the controller.

ModuloFS, 2022-10-13:
On the photo you see not very much... i can send the schematic

ModuloFS, 2022-10-13:
[CNC-Controller.pdf](https://github.com/bdring/FluidNC/files/9780351/CNC-Controller.pdf)

bdring, 2022-10-13:
I think your switches have a little more hysteresis than 1mm.

I tested on my test rig and the dual axis homing is working as designed. I tested with multiple pulloffs. Symmetrical and asymmetrical.

bdring, 2022-10-13:
Send a picture of your machine and the switches.

ModuloFS, 2022-10-13:
[image]
[image]
[image]

MitchBradley, 2022-10-13:
Show us the complete message log from a fail under the new circumstances

ModuloFS, 2022-10-15:
Now the problems appears not on the y-axis any more but on the z-axis. The z-axis stops not on every homing run but it has everytime glitches at the first limit switch touch off. I have measured the electrical signals of the strange behaviour, maybe it helps you:

Here are the Measurement positions in my circuit: (the whole schematic of my controller is in the conversation above)

There, i measured my z limit switch signal:

[image]

Here you can see the measurement position of the z step signal:

[image]

In the following picture you can see the oszillogram of the homing run at the first limit switch touchoff. The limit switch signal has clean edges (you can see it not here but i checked before), no spikes. The reason why the signal is not going to GND is because i measured above the LED (accesabiliy of the measurement position)  you see the dropout voltage of the LED. If you look at channel 2 you can see the step signal. Here you see that i get some glitches after the limit switch is activated. You can hear these glitches too! After a few of this single steps the stepps are continuing and making the next an last touchoff. This touchoff works perfectly without any stops.

[image]

There is the Debug Log of the Homing run shown above:

$HZ
[MSG:DBG: Homing Cycle Z]
[MSG:DBG: Homing nextPhase FastApproach]
[MSG:DBG: Starting from 0.000,0.000,150.000]
[MSG:DBG: Planned move to 0.000,0.000,315.000 @ 400.000]
[MSG:DBG:  Z Pos Limit 1]
[MSG:DBG: Homing limited Z]
[MSG:DBG: Homing nextPhase Pulloff0]
[MSG:DBG: Starting from 0.000,0.000,160.598]
[MSG:DBG: Planned move to 0.000,0.000,155.598 @ 100.000]
[MSG:DBG:  Z Pos Limit 0]
[MSG:DBG: CycleStop Pulloff0]
[MSG:DBG: Homing nextPhase SlowApproach]
[MSG:DBG: Starting from 0.000,0.000,155.598]
[MSG:DBG: Planned move to 0.000,0.000,161.098 @ 100.000]
[MSG:DBG:  Z Pos Limit 1]
[MSG:DBG: Homing limited Z]
[MSG:DBG: Homing nextPhase Pulloff1]
[MSG:DBG: Starting from 0.000,0.000,157.673]
[MSG:DBG: Planned move to 0.000,0.000,152.673 @ 100.000]
[MSG:DBG:  Z Pos Limit 0]
[MSG:DBG: CycleStop Pulloff1]
[MSG:DBG: Homing nextPhase Pulloff2]
[MSG:DBG: mpos was 0.000,0.000,152.673]
[MSG:DBG: mpos becomes 0.000,0.000,150.000]
[MSG:DBG: mpos transformed 0.000,0.000,150.000]
[MSG:DBG: Homing done]
ok

Here you can see my used config file:

name: "ModuloController"
board: "ModuloControl V1"

stepping:
  engine: RMT
  idle_ms: 255
  dir_delay_us: 5
  pulse_us: 5
  disable_delay_us: 5
  
start:
  must_home: true
  deactivate_parking: true
  check_limits: true

axes:
  shared_stepper_disable_pin: gpio.23
  
  x:
    steps_per_mm: 400.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 150.000
    max_travel_mm: 510.000
    soft_limits: true
    
    homing:
      cycle: 2
      mpos_mm: 510.000
      feed_mm_per_min: 200.000
      seek_mm_per_min: 400.000
      positive_direction: true

    motor0:
      limit_pos_pin: gpio.35:low
      hard_limits: true
      pulloff_mm: 5.000
      standard_stepper:
        direction_pin: gpio.27
        step_pin: gpio.26

  y:
    steps_per_mm: 400.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 150.000
    max_travel_mm: 750.000
    soft_limits: true

    homing:
      cycle: 3
      mpos_mm: 750.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 400.000
      positive_direction: true

    motor0:
      limit_pos_pin: gpio.39:low
      hard_limits: true
      pulloff_mm: 2.000
      standard_stepper:
        direction_pin: gpio.32:low
        step_pin: gpio.25

    motor1:
      limit_pos_pin: gpio.36:low
      hard_limits: true
      pulloff_mm: 2.000
      standard_stepper:
        direction_pin: gpio.22:low
        step_pin: gpio.21

  z:
    steps_per_mm: 400.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 150.000
    max_travel_mm: 150.000
    soft_limits: true
    
    homing:
      cycle: 1
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 400.000
      positive_direction: true

    motor0:
      limit_pos_pin: gpio.34:low
      hard_limits: true
      pulloff_mm: 5.000
      standard_stepper:
        direction_pin: gpio.18
        step_pin: gpio.19

probe:
  pin: gpio.4:high
  
control:
  safety_door_pin: gpio.33:low
  reset_pin: NO_PIN
  feed_hold_pin: NO_PIN

relay:
  direction_pin: NO_PIN
  output_pin: gpio.5
  enable_pin: NO_PIN
  disable_with_s0: false
  s0_with_disable: true
  spinup_ms: 0
  spindown_ms: 0
  tool_num: 0
  speed_map: 0=0.000% 0=100.000% 1=100.000%

ModuloFS, 2022-10-15:
I figured out a new detail of the problem. The axis, which is not homing correctly has the right idle signal state, the other axes are inverted. With the following configuration file i geht the following signal states if the machine is not moving: X = high, Y1 = high Y2 = high, Z = low.
This stayes the same even after a power cycle.

name: "ModuloController"
board: "ModuloControl V1"

stepping:
  engine: RMT
  idle_ms: 255
  dir_delay_us: 5
  pulse_us: 5
  disable_delay_us: 5
  
start:
  must_home: true
  deactivate_parking: true
  check_limits: true

axes:
  shared_stepper_disable_pin: gpio.23
  
  x:
    steps_per_mm: 400.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 150.000
    max_travel_mm: 510.000
    soft_limits: true
    
    homing:
      cycle: 2
      mpos_mm: 510.000
      feed_mm_per_min: 200.000
      seek_mm_per_min: 400.000
      positive_direction: true

    motor0:
      limit_pos_pin: gpio.35:low
      hard_limits: true
      pulloff_mm: 5.000
      standard_stepper:
        direction_pin: gpio.27
        step_pin: gpio.26:high

  y:
    steps_per_mm: 400.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 150.000
    max_travel_mm: 750.000
    soft_limits: true

    homing:
      cycle: 3
      mpos_mm: 750.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 400.000
      positive_direction: true

    motor0:
      limit_pos_pin: gpio.39:low
      hard_limits: true
      pulloff_mm: 2.000
      standard_stepper:
        direction_pin: gpio.32:low
        step_pin: gpio.25:high

    motor1:
      limit_pos_pin: gpio.36:low
      hard_limits: true
      pulloff_mm: 2.000
      standard_stepper:
        direction_pin: gpio.22:low
        step_pin: gpio.21:high

  z:
    steps_per_mm: 400.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 150.000
    max_travel_mm: 150.000
    soft_limits: true
    
    homing:
      cycle: 1
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 400.000
      positive_direction: true

    motor0:
      limit_pos_pin: gpio.34:low
      hard_limits: true
      pulloff_mm: 5.000
      standard_stepper:
        direction_pin: gpio.18
        step_pin: gpio.19:high

probe:
  pin: gpio.4:high
  
control:
  safety_door_pin: gpio.33:low
  reset_pin: NO_PIN
  feed_hold_pin: NO_PIN

relay:
  direction_pin: NO_PIN
  output_pin: gpio.5
  enable_pin: NO_PIN
  disable_with_s0: false
  s0_with_disable: true
  spinup_ms: 0
  spindown_ms: 0
  tool_num: 0
  speed_map: 0=0.000% 0=100.000% 1=100.000%

bdring, 2022-10-15:
You have a shared pin that controls idle state.

```yaml
axes:
   shared_stepper_disable_pin: gpio.23
```

It is not a problem with the firmware.

ModuloFS, 2022-10-15:
Now i erased the complete ESP and flashed it agian. Now it i do not see any failures with homing and all step signal have the same signal direction / idle state. But the idle state is still inverted, if i do not add :low or :high to the step pin the signal state is HIGH if no stepper is moving and a step signal is done by pulling the pin low for 5 us. If i use :low after my step pin definition the step signal is idle LOW and a step signal pulls the pin to high for 5us. 
Now i have a new problem, if i jog the machine over 3000 mm/min i get short stops of the stepper during the movement.... I see it also in the step signals.
What has the disable pin to do with my step pins?

bdring, 2022-10-15:
You said nothing about step pins...

"The axis, which is not homing correctly has the right idle signal state, the other axes are inverted"

I assumed you meant the disable signal, because that is controlled by the idle time.

There is no "idle state" signal.

Try using STA mode for the Wifi. The ESP32 AP mode is not good.

http://wiki.fluidnc.com/en/features/wifi_bt#wifi-ap-mode

ModuloFS, 2022-10-15:
Ok, with idle state i meant the signal state if no step signal is generated or if the stepper ist not moving. Now am back at the old homing problem. Here is the way i got back: 
1. Upload a new config file with this line removed:    shared_stepper_disable_pin: gpio.23
2. Power cycle the ESP
3. Testing, got a problem that now my steppers are disabled -> this is ok because of my hardware design.
4. Added the line shared_stepper_disable_pin: gpio.23 to the config file again to get the original config file
5. Uploaded the new config file to the ESP
6. Power cycle the ESP
7. And now i get the same homing problems again...

I try STA mode now. If this not solves my problem i am at the end with my latin how we used to say in germany....

ModuloFS, 2022-10-15:
After changing to STA mode and doing a power cycle, the homing worked again, but i get again short stops during a fast jogging movement over 3000mm/min.
Additional after he power cycle the polarity of the step signals agreed with the configuration. Because before they where inverted i changed the config file back and power cycled the ESP. 
After that, the polarity was again worng and the homing is not working any more.....

So the change to the STA mode has changed nothing.... 

I am debugging this stuff the whole day with basically no solution quite disappointing...

closed 2022-10-15 by bdring

ModuloFS, 2022-10-15:
Why you closed the issue?

bdring, 2022-10-15:
I thought you were done. You you want a refund?

ModuloFS, 2022-10-15:
Yes, for today i am done. But maybe someone else has the same issue?

ModuloFS, 2022-10-15:
By the way, when I said quite disappointing, I meant the situation of not to finde the failure in general, not your support or the project!

MitchBradley, 2022-10-15:
The very large amount of noise on your scope traces worries me.  I can think of several possibilities:
1. The scope ground is not connected or is connected to a ground point that is very far away from the probe point.
2. The bypassing is inadequate.  The schematic shows a few 470n capacitors of unknown type.  A more typical value is 100n or even 10n in ceramic.  As the capacitance increases, so does the self inductance, so larger values are worse for higher frequency noise.  The gold standard is to have a variety of capacitors with different values, like 100u to handle low frequency noise, 100n for higher frequencies, and 10n for very high frequencies.  TTL can generate impressive current spikes when it switches.
3. The grounding layout is inadequate.  For grounding to be effective at high frequencies, there must be a path for return current to flow in the ground wires or plane that is physically close to the path that the signal takes, so the loop that is formed by the signal current and its associated return current is small.  The impedance of the ground return must also be low enough (large traces or plane) to avoid IR voltage shifts between the ground nodes of different chips.
4. Noise could be coming in from outside components - steppers, spindles, anything else in the room - due to inadequate power line filtering, ground loops in the overall system, and many other factors. Overall system grounding and plug-in points are important.  I run my spindles from separate outlets than my controllers.  Separate circuits would be better but in my setup, separate outlets work.  Every environment is different.

I do not know which - or any - of these problems pertain to your system.  You showed us your schematic, but we haven't seen your physical layout (PCB or otherwise), nor the overall wiring of the system (you just showed closeups, not the big picture).

ModuloFS, 2022-10-15:
Yes, the noise on the scope comes from bad grounding point of the scope-probe in the circuit (long grounding lead picks up a lot fo noise from the stepper driver). But my behaviour is comming with the change of the firmware from 3.5.1 to 3.6.3 and i measure definetly wrong signal states comming out of the ESP so this has less to do with EMC problems.
I am currently rebuilding the circuit anyway (switching to a new spindle with vfd and 0-10V control) and maybe if i am ready i can do some better measurements. If i get more information i let you know. 

Just a last question, if i change / update the firmware is it correct to run the follwoing scripts on windows to get a completely clean and new installation on the ESP:
1. erase.bat
2. install-fs.bat
3. install-wifi.bat

Or is just the install-wifi.bat script needed?

ModuloFS, 2022-10-20:
I flashed https://github.com/bdring/FluidNC/releases/tag/v3.6.4-pre1 and first tests showed no problems with homing now! The signal levels of the step signals are agree with the config file too!