bdring/FluidNC #669: G38.2 P flag sets incorrect offset

daxliniere, 2022-10-14:
### Controller Board

DLC32

### Help From Board Vendor

- [ ] Yes
- [X] No
- [ ] Not Applicable

### Machine Description

SWOLE-CNC

### Configuration file

```yaml
board: MKS-DLC32
name: SWOLE-CNC
meta: V2.6.2 (2202.10.11) Dax Liniere, originally by Skorpi
 
kinematics:
  Cartesian:
 
stepping:
  engine: I2S_STATIC
  idle_ms: 0
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0
 
axes:
  shared_stepper_disable_pin: I2SO.0
  x:
    steps_per_mm: 800.000
    #was 802.341
    max_rate_mm_per_min: 1600.000
# was 1500.000
    acceleration_mm_per_sec2: 80.000
    max_travel_mm: 302.000
    soft_limits: true
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 20.000
      seek_mm_per_min: 200.000
# was 100 in first SWOLE-CNC iteration running MKS firmware
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100
 
    motor0:
      limit_neg_pin: gpio.36
      limit_pos_pin: gpio.34
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 1.000
      stepstick:
        step_pin: I2SO.1
        direction_pin: I2SO.2:low
        disable_pin: NO_PIN
        ms1_pin: NO_PIN
        ms2_pin: NO_PIN
        ms3_pin: NO_PIN
        reset_pin: NO_PIN
 
  y:
    steps_per_mm: 802.322
    #tested 806.062, features too small
    max_rate_mm_per_min: 1600.000
    acceleration_mm_per_sec2: 80.000
# was 300.000
    max_travel_mm: 186.000
    soft_limits: true
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 20.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100
 
    motor0:
      limit_neg_pin: gpio.35
      limit_pos_pin: gpio.23
      #this pin doesn't have the same input circuitry as the 3 dedicated on-board limit switch inputs
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 1.000
      stepstick:
        step_pin: I2SO.5
        direction_pin: I2SO.6:low
        disable_pin: NO_PIN
        ms1_pin: NO_PIN
        ms2_pin: NO_PIN
        ms3_pin: NO_PIN
        reset_pin: NO_PIN
  z:
    steps_per_mm: 802.497
    max_rate_mm_per_min: 1000.000
    acceleration_mm_per_sec2: 50.000
# was 500.000
    max_travel_mm: 100
#46.000
    soft_limits: true
    homing:
      cycle: 1
      positive_direction: true
      mpos_mm: 0.000
      feed_mm_per_min: 20.000
      seek_mm_per_min: 100.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100
 
    motor0:
      limit_pos_pin: gpio.18
      #this pin doesn't have the same input circuitry as the 3 dedicated on-board limit switch inputs
      hard_limits: true
      pulloff_mm: 1.000
      stepstick:
        step_pin: I2SO.3
        direction_pin: I2SO.4:low
        disable_pin: NO_PIN
        ms1_pin: NO_PIN
        ms2_pin: NO_PIN
        ms3_pin: NO_PIN
        reset_pin: NO_PIN
 
i2so:
  bck_pin: gpio.16
  data_pin: gpio.21
  ws_pin: gpio.17
 
spi:
  miso_pin: gpio.12
  mosi_pin: gpio.13
  sck_pin: gpio.14
 
sdcard:
  cs_pin: gpio.15
  card_detect_pin: NO_PIN
# This could be GPIO.39, but Card Detect has no supported functions in FluidNC
 
control:
  safety_door_pin: gpio.33:low:pu
# Does the start pin need to be wired to the other side of the switch??
  cycle_start_pin: NO_PIN
  feed_hold_pin: NO_PIN
  reset_pin: NO_PIN
  macro0_pin: gpio.19:low:pu
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN
 
macros:
  startup_line0:
  startup_line1:
  macro0: $X&$H
#unlock and home XYZ
  macro1: G38.2 G91 Z-46 F100&G0 Z1&G38.2 G91 Z-2 F50&G10 L20 P1 Z4.985&G10 L20 P2 Z4.985&G10 L20 P3 Z4.985&G10 L20 P4 Z4.985&G10 L20 P5 Z4.985&G10 L20 P6 Z4.985&G53 G0 Z0
#z-probe
  macro2:
  macro3: $SD/Run=lasertest.gcode
 
coolant:
  flood_pin: gpio.0
# continuous air
  mist_pin:  gpio.4
# pulsed air 
  delay_ms: 0
 
probe:
  pin: gpio.22:low
  check_mode_start: true
 
 
10V:
# Spindle
  direction_pin: NO_PIN
  forward_pin: NO_PIN
  reverse_pin: NO_PIN
  output_pin: gpio.32
  enable_pin: gpio.5:low
  pwm_hz: 5000
  disable_with_s0: false
  s0_with_disable: false
  off_on_alarm: true
#switch off spindle when in alarm state
  spinup_ms: 10000
  spindown_ms: 12000
  tool_num: 0
  speed_map: 0=0% 0=25% 5868=25% 23237=99% 24000=100%
#DLC32 Spindle TTL max output voltage is 4.89v, so it can never reach 24000rpm/400Hz
 
 
#pwm:
#  direction_pin: NO_PIN
#  output_pin: gpio.32
#  enable_pin: gpio.5:low
#  pwm_hz: 5000
#  disable_with_s0: false
#  s0_with_disable: true
#  spinup_ms: 10000
#  spindown_ms: 12000
#  tool_num: 0
#  speed_map: 0=0% 0=25% 5868=25% 23237=99% 24000=100%
 
 
user_outputs:
  analog0_pin: NO_PIN
  analog1_pin: NO_PIN
  analog2_pin: NO_PIN
  analog3_pin: NO_PIN
  analog0_hz: 5000
  analog1_hz: 5000
  analog2_hz: 5000
  analog3_hz: 5000
  digital0_pin: NO_PIN
  digital1_pin: NO_PIN
  digital2_pin: NO_PIN
  digital3_pin: NO_PIN
 
start:
  must_home: false
# Pins that could be used: 0(SDA), 4(SCL), 5, 18, 19, 22, 23, 25, 26, 27, 32 (TTL spindle control), 33, 39, I2SO.7
# Pins not currently in use: 25 (TMC_CS_Y), 26 (TMC_CS_X), 27 (TMC_CS_Z), (39 SD card detect), I2SO.7
```

### Startup Messages

```text
...
```

### User Interface Software

UGS dev 2022-08-25

### What happened?

Issued **G38.2 G91 X-2 F10 P3.979** and once completed and probe offset is applied, my DRO reads 3.960 instead of 3.979

It happened a couple times when I was using a different P value, but I thought I was just going nuts or had already been at an offset or something strange.
But it's happened again with a different P value, so I think there's a bug.

[image]

bdring, 2022-10-15:
It records the location of the touch and then decelerates to a stop. In the case of a Z down probe. The DRO should read a tiny bit lower than the touch point.

The probe has always worked this way since the dawn of Grbl.

bdring, 2022-10-15:
I was going to add an extra note to the Wiki, but it was already there.

[image]

bdring, 2022-10-15:
http://wiki.fluidnc.com/en/config/probe

closed 2022-10-15 by bdring

daxliniere, 2022-10-16:
Okay, thanks Bart. It's cool that you calculate that stuff.
I'm thinking though, if I'm traveling at 10mm/min, surely this shouldn't be producing 17-19 microns overshoot, right?
Should I be setting acceleration faster inside my probing macro? (macro 1 in my config.)

MitchBradley, 2022-10-16:
10 mm/min  /  60 sec/min / 50 mm/sec2 = .0033 sec   dist = 1/2 * a * t^2 = 1/2 * 50 mm/sec2 * (.0033)^2 sec2 = .00028 mm

daxliniere, 2022-10-16:
> 10 mm/min / 60 sec/min / 50 mm/sec2 = .0033 sec dist = 1/2 * a * t^2 = 1/2 * 50 mm/sec2 * (.0033)^2 sec2 = .00028 mm

[image]

MitchBradley, 2022-10-16:
Basic physics.  Any high school textbook.

MitchBradley, 2022-10-16:
Decelerating from a velocity of 10 mm/min = 0.167 mm/sec , at acceleration 50 mm/min, should take 0.0033 seconds, during which time the distance traveled is 0.00028 mm

daxliniere, 2022-10-16:
So that would confirm that my suggestion of 0.017-0.019mm to be inaccurate?

MitchBradley, 2022-10-16:
That assumes a perfectly stiff machine.  I don't have one that stiff.

MitchBradley, 2022-10-16:
You have idle_ms at 0.  To keep the motors enabled all the time, the appropriate setting is 255.  If the motors are disabled, they can no long hold the precise position and the mechanicals will relax.

MitchBradley, 2022-10-16:
But if all this is coming from the DRO reading, it should be determined by what the controller thinks, not where the machine actually is.

daxliniere, 2022-10-16:
> That assumes a perfectly stiff machine. I don't have one that stiff.

Neither me. I think only [Piotr Fox](https://www.youtube.com/watch?v=zmnuG0q2Atg) does...

bdring, 2022-10-16:
What is your mm per step? Was the touch exactly on a step?

daxliniere, 2022-10-16:
> But if all this is coming from the DRO reading, it should be determined by what the controller thinks, not where the machine actually is.

It would be cool if FNC supported DRO inputs, but since it doesn't, wouldn't that reinforce my point that ~18 microns is an incorrect assumption/calculation from FNC? It is significantly different from the result of your calculation and FNC can't know how rigid a machine is to be able to estimate the probable stop position (outside of deceleration, of course).

daxliniere, 2022-10-16:
> What is your mm per step? Was the touch exactly on a step?

That's a very good point. Roughly 800 steps/mm.