bdring/FluidNC #670: Problem: Direction pins can't be shared anymore

TjeerdvanDijk, 2022-10-15:
### Controller Board

ESP32-dev

### Help From Board Vendor

- [ ] Yes
- [ ] No
- [X] Not Applicable

### Machine Description

My own mill

### Configuration file

```yaml
--- FluidNC ---
y:
  motor0:
    standard_stepper:
      direction_pin: gpio.33
      step_pin: gpio.27
  motor1:
    standard_stepper:
      direction_pin: gpio.33
      step_pin: gpio.12

--- Grbl_ESP32 ---

#define DEFAULT_HOMING_SQUARED_AXES (bit(Y_AXIS))

#define Y_DIRECTION_PIN         GPIO_NUM_33
#define Y2_DIRECTION_PIN        Y_DIRECTION_PIN
```

### Startup Messages

```text
[MSG:ERR: Configuration error at /machine/axes/y/motor1/standard_stepper: ERR: gpio.33 - Pin is already used.]
```

### What happened?

So i'm trying to give FluidNC a shot, moving away from Grbl_ESP32. 

I used to square my Y axis using a single limit switch and a single direction pin, which works just great. It was especially great because there is a limited amount of pins at our disposal. 

Now I feel like this feature is removed in FluidNC and that is a shame, it simply disables me from upgrading because my custom PCB does not provide more pins, frankly my setup already uses every single available pin of the ESP32. 

I hope this feature can be added again

bdring, 2022-10-15:
Please upload your full config file.

TjeerdvanDijk, 2022-10-15:
my config is incomplete because I am still setting up my machine, but this is where I got stuck 
```name: ESP32_V4
board: unknown

kinematics:
  Cartesian:

stepping:
  engine: RMT
  idle_ms: 250
  dir_delay_us: 0
  pulse_us: 3
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: gpio.23
  x:
    steps_per_mm: 100.000
    max_rate_mm_per_min: 1000.000
    acceleration_mm_per_sec2: 200.000
    max_travel_mm: 300.000
    soft_limits: true
    homing:
      cycle: 2
      mpos_mm: 0.000
      positive_direction: true
      settle_ms: 250.000
      seek_mm_per_min: 2000.000
      feed_mm_per_min: 200.000
      seek_scaler: 1.100
      feed_scaler: 5.000
    motor0:
      limit_all_pin: gpio.16:low
      hard_limits: false
      standard_stepper:
        direction_pin: gpio.25
        step_pin: gpio.13
  y:
    steps_per_mm: 100.000
    max_rate_mm_per_min: 1000.000
    acceleration_mm_per_sec2: 200.000
    max_travel_mm: 300.000
    soft_limits: true
    homing:
      cycle: 2
      mpos_mm: 0.000
      positive_direction: true
      settle_ms: 250.000
      seek_mm_per_min: 2000.000
      feed_mm_per_min: 200.000
      seek_scaler: 1.100
      feed_scaler: 5.000
    motor0:
      limit_all_pin: gpio.17:low
      hard_limits: false
      standard_stepper:
        direction_pin: gpio.33
        step_pin: gpio.27
    motor1:
      standard_stepper:
        direction_pin: gpio.33
        step_pin: gpio.12
  z:
    steps_per_mm: 100.000
    max_rate_mm_per_min: 1000.000
    acceleration_mm_per_sec2: 200.000
    max_travel_mm: 300.000
    soft_limits: true
    homing:
      cycle: 1
      mpos_mm: 0.000
      positive_direction: false
      settle_ms: 250.000
      seek_mm_per_min: 2000.000
      feed_mm_per_min: 200.000
      seek_scaler: 1.100
      feed_scaler: 5.000
    motor0:
      limit_all_pin: gpio.5:low
      hard_limits: false
      standard_stepper:
        direction_pin: gpio.32
        step_pin: gpio.26

coolant:
  flood_pin: gpio.22
  mist_pin: gpio.21
  delay_ms: 1000.000

probe:
  pin: gpio.4:pu
  check_mode_start: false

macros:
  startup_line0: 
  startup_line1: 
  macro0: 
  macro1: 
  macro2: 
  macro3: 

start:
  must_home: true
  check_limits: true
  deactivate_parking: false

user_outputs:

PWM:
  tool_num: 0
  speed_map: 0=0.0% 1000=100.0%
  spinup_ms: 0
  spindown_ms: 0
  output_pin: gpio.18
  enable_pin: gpio.19
  disable_with_s0: false
  s0_with_disable: false
  pwm_hz: 5000
arc_tolerance_mm: 0.002
junction_deviation_mm: 0.010
verbose_errors: false
report_inches: false
enable_parking_override_control: false
use_line_numbers: false ```

TjeerdvanDijk, 2022-10-15:
please read 'What happened?'

bdring, 2022-10-15:
FluidNC does not let you assign a pin to more than one function.

If you change the direction pin to NO_PIN. It will work.

bdring, 2022-10-15:
```yaml
    motor1:
      standard_stepper:
        direction_pin: NO_PIN
        step_pin: gpio.12
        
```

TjeerdvanDijk, 2022-10-15:
Oh allright, but i will still be able to square my axis ? correct?

bdring, 2022-10-15:
FluidNC requires a switch for each motor for auto squaring. We use a stress free method now that does not twist the axis during homing dual motor per axis machines.

I am not sure about the single direction pin for auto squaring.

TjeerdvanDijk, 2022-10-15:
Okay, may I request this as a new feature? Twisting is not a problem depending on how a machine is built. So it is a decision the creator of the machine has to make, in my opinion the software shouldn't assume that twisting is a problem

bdring, 2022-10-15:
We added a lot of features to the homing in FluidNC. The code is quite complex now. Homing with 1 switch requires extra cycles. We do not want increase the complexity at this time. It is getting to the point where we can no longer test all the features of homing. I cannot afford all that hardware and desk space.

TjeerdvanDijk, 2022-10-15:
Got it, ill just have to live with Grbl_esp32 until i create a new pcb. I appreciate the effort though, been using this software for 2 years or so

closed 2022-10-15 by TjeerdvanDijk